ptrkrysik/gr-gsm #21: how to install the gr-gsm on the raspberrypi manually without pybomb

helxsz, 2015-01-15:
I have installed gr-gsm on my ubuntu, it works well. now I want to put it into my raspberrypi. I know using pybomb to install it is a easy way, but pybomb needs to download and recompile everything, which it is too slow for me. I have installed compiled gnu-radio into the PI,  which is much quicker installation. In my previous gr-gsm installation, I have also installed a gnu-radio on my ubuntu, but pybomb still downloaded and recompiled. 

My question, if I have installed some precompiled package, how do I bypass these installed package through pybomb, or I can just install all the needed package myself manually and install gr-gsm at last

ptrkrysik, 2015-01-15:
Hi @helxsz. I've changed installation description to the one with use of pybombs for easiness for the user. Maybe I will add more general installation again then. If you want to install gr-gsm you might follow instructions from the older readme:
https://github.com/ptrkrysik/gr-gsm/blob/72f22649686b77ca4fb877e66809283731e95c2a/README.md
Basically you have to install all required software (mainly gnuradio), then download gr-gsm:
git clone https://github.com/ptrkrysik/gr-gsm.git
and compile and install it with following commands:
cd gr-gsm
mkdir build
cd build
cmake ..
make
sudo make install

helxsz, 2015-01-19:
Thanks, @ptrkrysik , it works well for me.

closed 2015-01-19 by helxsz