ptrkrysik/gr-gsm #28: No module named grgsm

themullet, 2015-02-13:
Hi there, just tried installing using:

sudo ./pybombs install gr-gsm

When I go to run airprobe_rtlsdr.py getting the error: 

Traceback (most recent call last):
  File "/home/steveswallow/target/bin/airprobe_rtlsdr.py", line 19, in <module>
    import grgsm
ImportError: No module named grgsm

Same thing with airprobe_file as well. 

My python version is Python 2.7.6. 

Cheers

ptrkrysik, 2015-02-15:
Do you have /usr/local/lib/python2.7/dist-packages/grgsm directory?
try also:
sudo ldconfig

themullet, 2015-02-17:
perfect, got it, moved package to that directory and also moved libgnuradio-gsm.so and worked perfectly. Thanks for the help.

closed 2015-02-17 by themullet

henrycaga, 2015-04-07:
hi, im also getting this kind of error.. i dont have a "grgsm" folder in /usr/local/lib/python2.7/dist-packages

need help. thank you

ptrkrysik, 2015-04-07:
If you don't have that directory either you didn't install gr-gsm or made a mistake when configuring pyboms and set installation path into directory other than /usr/local

henrycaga, 2015-04-07:
hi sir, thanks for the reply. i forgot to update this. i already installed the gr-gsm by using the /usr/local prefix at pybombs config. its already solved. im now running the gr-gsm on odroid :) thanks a lot sir

henrycaga, 2015-04-07:
more power on this code. keep it up. very useful thumbs up to airprobe / gr-gsm

phocean, 2015-05-19:
Hi guys,

I also had this issue on Fedora, but the solutions I have seen so far in this thread are not optimal.
If the libs are not found on your system, then you need to adjust your environment.

Concerning the libs, make sure that ldconfig works against the local folder.
Just add the path to /etc/ldconfig.conf, or, better, create a simple file like this:
```shell
% cat /etc/ld.so.conf.d/gr-gsm.conf 
/usr/local/lib64
```

Similarly for Python, add the following line in your $HOME/.bashrc or $HOME/.zshrc:
```shell
PYTHONPATH=$PYTHONPATH:/usr/local/lib64/python2.7/site-packages/
export PYTHONPATH
```

I hope this helps :-)

tiger762, 2016-04-04:
I'm also having this problem, but none of the suggestions has worked for me. I have PYTHONPATH defined as /usr/local/lib64/python2.7/site-packages/

I did the manual build, as PyBOMBS is always a nightmare for me for some reason. I did this:

git clone https://github.com/ptrkrysik/gr-gsm
cd gr-gsm
mkdir build
cd build
cmake ..
make
make install

It appeared to build fine, but I am wondering if "swig" did not get built, because under $PYTHONPATH there is nothing resembling SWIG.

attina, 2017-01-05:
I fixed same problem by following command on Fedora 25, hope this can help you guys.

```
export PYTHONPATH=$PYTHONPATH:/usr/local/lib64/python2.7/site-packages/grgsm:/usr/local/lib64/python2.7/site-packages
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib64/
```
output:

> linux; GNU C++ version 6.2.1 20160916 (Red Hat 6.2.1-2); Boost_106000; UHD_003.010.001.000-0-unknown
>ARFCN:   21, Freq:  939.2M, CID: 58453, LAC:  4544, MCC: 460, MNC:   0, Pwr: -29
>ARFCN:   33, Freq:  941.6M, CID:  8393, LAC:  4530, MCC: 460, MNC:   0, Pwr: -27
>ARFCN:   36, Freq:  942.2M, CID: 57566, LAC:  4530, MCC: 460, MNC:   0, Pwr: -49
>ARFCN:   38, Freq:  942.6M, CID:     0, LAC:  4530, MCC: 460, MNC:   0, Pwr: -39
>ARFCN:  122, Freq:  959.4M, CID: 45883, LAC:  4309, MCC: 460, MNC:   1, Pwr: -28

Lohith12, 2019-12-25:
ModuleNotFoundError: No module named 'grgsm' . Please help

catalinalb, 2019-12-27:
> ModuleNotFoundError: No module named 'grgsm' . Please help

Same on ubuntu 18.04 installation with pybombs

velichkov, 2019-12-27:
Hi @Lohith12, @catalinalb,

> ModuleNotFoundError: No module named 'grgsm' . Please help

In case you are using gnuradio 3.8 with python 3 try adding `/usr/local/lib/python3/dist-packages/` to `PYTHONPATH` and then from the same terminal run the grgsm commands.

```
export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH
./grgsm_livemon
```

Lohith12, 2019-12-29:
> Hi @Lohith12, @catalinalb,
>
>
> In case you are using gnuradio 3.8 with python 3 try adding `/usr/local/lib/python3/dist-packages/` to `PYTHONPATH` and then from the same terminal run the grgsm [...]

Thanks bro it worked

catalinalb, 2020-01-05:
How can I add this line ? where should I find PYTHONPATH ?

velichkov, 2020-01-05:
@catalinalb,

> How can I add this line ? 

What line?

>  where should I find PYTHONPATH ?

`PYTHONPATH` is an environment variable. Open a terminal end execute `echo $PYTHONPATH`

batica81, 2023-09-12:
Form me on Debian 11, this path actually worked:

export PYTHONPATH=/usr/local/lib/python3.9/site-packages/:$PYTHONPATH