ptrkrysik/gr-gsm #29: Does not work for me.

pwns4cash, 2015-02-26:
I have followed the directions, yet when I try to run airprobe_rtlsdr.py  the following is returned

Traceback (most recent call last):
  File "/usr/local/bin/airprobe_rtlsdr.py", line 19, in <module>
    import osmosdr
  File "/usr/lib/python2.7/dist-packages/osmosdr/__init__.py", line 45, in <module>
    from osmosdr_swig import *
  File "/usr/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 26, in <module>
    _osmosdr_swig = swig_import_helper()
  File "/usr/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 22, in swig_import_helper
    _mod = imp.load_module('_osmosdr_swig', fp, pathname, description)
ImportError: /usr/lib/x86_64-linux-gnu/libgnuradio-osmosdr-0.0.3git.so.0.0.0: undefined symbol: _ZN3uhd6device4findERKNS_13device_addr_tE

gnuradio 3.6.5.1-1kali0

I have tried to force install as well... just not happening... suggestions?

ptrkrysik, 2015-02-26:
gnuradio 3.6.5 is way too old. At least 3.7.3 is required.

Did you use pybombs?

pwns4cash, 2015-02-28:
yes I used pybombs

ptrkrysik, 2015-04-03:
Did you install gnuradio with pybombs? If not - try to do that.

pwns4cash, 2015-04-07:
I did.

ptrkrysik, 2015-04-07:
Run:

gnuradio-companion --version

You should obtain 3.7.x version.
If not - remove other installations of gnuradio (but only if you don't
need them for some other program to work).

ptrkrysik, 2015-04-21:
I'm closing this issue for now. If you still have problem, please provide some specific information that will help me too debug the problem. I will reopen it then.

closed 2015-04-21 by ptrkrysik

Rajnish357, 2015-05-06:
@pwns4cash did you found any solution ?