ptrkrysik/gr-gsm #31: cannot get airprobe to work

epgreen, 2015-03-10:
Hello, whenever I start airprobe_rtlsdr.py it opens, but the spectrum doesn't run, and it says it cannot find any compatible devices, due to gnuradio error code 528. I did not install this via pybombs.

ptrkrysik, 2015-03-10:
Do you have an RTL-SDR device (or any other supported by Osmosdr source)
connected to your computer when you start the program? If yes: did you
check that it works as SDR device in gnuradio?

epgreen, 2015-03-10:
Yes I have an rtlsdr, I was trying to get it to work with that. I also have a hackrf that I haven't tried (didn't think it was supported). The rtlsdr works with gqrx and rtl_power, but I will be making a quick wbfm program on grc to see if that works.

epgreen, 2015-04-02:
Have you been able to figure this out? Still won't work

dmaynor, 2015-04-02:
What OS/Version/Gnuradio Version are you using?

epgreen, 2015-04-02:
Ubuntu 14.10 primarily, with Gnuradio version 3.7.3.

ptrkrysik, 2015-04-03:
@epgreen  try to run a simple flowgraph in gnruadio with RTL-SDR source and check if it works.

ptrkrysik, 2015-04-21:
I'm closing this issue for now. I will reopen it if you provide morel info for so I can find source of the problem. Without it I can only guess what exactly problem you might have.

closed 2015-04-21 by ptrkrysik