ptrkrysik/gr-gsm #33: ctrl channel demapper - Issue with multiple instances

romankh, 2015-04-09:
I am using a customized variant of airprobe_file with a recorded file of a bts with non-combined channel config. 

In GRC, i have two ctrl channel demapper + decoder, one for TS0 and one for TS1.

The demapper config of the TS0 demapper is 
timeslot_nr: 0
starts_fn_mod51: [2,6,12,16,22,26,32,36,42,46]
channel_types: [BCCH,CCCH,CCCH,CCCH,CCCH,CCCH,CCCH,CCCH,CCCH,CCCH]

and the config for TS
timeslot_nr: 1
starts_fn_mod51: [0,4,8,12,16,20,24,28,32,36,40,44]
channel_types: [SDCCH, SDCCH, SDCCH, SDCCH, SDCCH, SDCCH, SDCCH, SDCCH,SACCH,SACCH,SACCH,SACCH]

where BCCH = 1, CCCH = 2, SDCCH = 8 and SACCH = 136

If both demappers are enabled, i have the following issue:
Some bursts of TS1 are assigned the gsmtap sub_type (Channel type in wireshark) CCCH instead of SDCCH or SACCH for some reason. If the demapper for TS 0 is disabled, the bursts are assigned the correct channel type.

It looks like those bursts were handled by the wrong demapper despite the fact, that the gsmtap header is assigned the correct timeslot.

romankh, 2015-04-10:
I can provide the grc and a small capture if necessary to reproduce

ptrkrysik, 2015-04-10:
Hi @romankh, it would greatly speed up debugging, if you could provide a
sample capture and two flowgraphs where you see difference.

romankh, 2015-04-10:
Hi @ptrkrysik , here is an archive containing a short cfile and two grc files:

http://www.file-upload.net/download-10531018/grgsm-issue_33.grc.tar.gz.html

How i get the differences:

- Open the first flow graph, where the demapper for ts0 is disabled. 
- Run the flowgraph with the provided cfile
- Capture in wireshark and set filter gsmtap.ts == 1
- Here TS1 is decoded correctly, the Sys Info Type 5 messages are reasonable
- Open the second flow graph, which is the same as the first one, but with demapper/decoder enabled for ts0 also
- Run again
- Capture in wireshark, same filter

Now, what happens for me is, that in the second run, some messages have the timeslot in the gsmtap header set correctly to 1, but the channel is set to CCCH and messages are decoded incorrectly. 

The incorrect messages differ from run to run, you may have to run the second flow graph multiple times.

romankh, 2015-04-11:
Ok, i kept overseeing that the channel is assigned before the timeslot condition in universal_ctrl_chans_demapper_impl.cc

I will test tomorrow and commit the modification

ptrkrysik, 2015-04-13:
Many thanks for fixing this bug @romankh ! I'm closing this issue for now, as your contributions should fix this problem.

closed 2015-04-13 by ptrkrysik