ptrkrysik/gr-gsm #34: Issue with installation

sv1, 2015-04-09:
Hello, 

I have followed the installation instructions but it seems that something is wrong. This is the first time I deal with pybombs and I am not sure if everything is correct:

$sudo ./pybombs install gr-gsm
Settled on prefix: /usr/local
Initializing environmental variables...
no existing inventory found, creating an empty one...
value updated
False
None
Loading recipes ...
/usr/local/
Installing packages:
* uhd
* wxpython
* pycairo
* gnuradio
* rtl-sdr
* osmo-sdr
* hackrf
* libosmo-dsp
* gr-iqbal
* bladeRF
* gr-osmosdr
* gr-gsm
Installing from source: uhd
no existing inventory found, creating an empty one...
Cloning into 'uhd'...
remote: Counting objects: 48900, done.
remote: Compressing objects: 100% (54/54), done.
remote: Total 48900 (delta 23), reused 0 (delta 0), pack-reused 48846
Receiving objects: 100% (48900/48900), 94.38 MiB | 844.00 KiB/s, done.
Resolving deltas: 100% (37037/37037), done.
Checking connectivity... done.
no existing inventory found, creating an empty one...
 
    CC=gcc CXX=g++ cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=/usr/local/ 

Configuring: (100%) [==========================================================]
Building:    (100%) [==========================================================]

    make install

Installing: (100%) [===========================================================]
installation ok via: src
Installing from source: wxpython
--2015-04-09 18:15:18--  http://prdownloads.sourceforge.net/wxpython/wxPython-src-2.8.11.0.tar.bz2
Resolving prdownloads.sourceforge.net (prdownloads.sourceforge.net)... 216.34.181.59
Connecting to prdownloads.sourceforge.net (prdownloads.sourceforge.net)|216.34.181.59|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://downloads.sourceforge.net/project/wxpython/wxPython/2.8.11.0/wxPython-src-2.8.11.0.tar.bz2 [following]
--2015-04-09 18:15:19--  http://downloads.sourceforge.net/project/wxpython/wxPython/2.8.11.0/wxPython-src-2.8.11.0.tar.bz2
Resolving downloads.sourceforge.net (downloads.sourceforge.net)... 216.34.181.59
Connecting to downloads.sourceforge.net (downloads.sourceforge.net)|216.34.181.59|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: http://kent.dl.sourceforge.net/project/wxpython/wxPython/2.8.11.0/wxPython-src-2.8.11.0.tar.bz2 [following]
--2015-04-09 18:15:19--  http://kent.dl.sourceforge.net/project/wxpython/wxPython/2.8.11.0/wxPython-src-2.8.11.0.tar.bz2
Resolving kent.dl.sourceforge.net (kent.dl.sourceforge.net)... 212.219.56.185
Connecting to kent.dl.sourceforge.net (kent.dl.sourceforge.net)|212.219.56.185|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 30604503 (29M) [application/octet-stream]
Saving to: ‘wxPython-src-2.8.11.0.tar.bz2’

wxPython-src-2.8.11.0.tar.bz2 100%[===================================================>]  29,19M   434KB/s   in 58s    

2015-04-09 18:16:18 (515 KB/s) - ‘wxPython-src-2.8.11.0.tar.bz2’ saved [30604503/30604503]

Extract wxPython-src-2.8.11.0.tar.bz2
 
    ./configure --prefix=/usr/local/  --disable-unicode  CC=gcc CXX=g++

Configuring: (100%) [==========================================================]
Building:    (100%) [==========================================================]

    make install && \
    make -C contrib/src/gizmos install && \
    make -C contrib/src/stc install && \
    cd wxPython && python setup.py install --prefix=/usr/local/ WXPORT=gtk2 UNICODE=0 BUILD_GLCANVAS=0

Installing: (100%) [===========================================================]

/gr-gsm/apps$ python airprobe_rtlsdr.py 
Traceback (most recent call last):
  File "airprobe_rtlsdr.py", line 18, in <module>
    import grgsm
ImportError: No module named grgsm

ptrkrysik, 2015-04-10:
Check if you have:
/usr/local/lib/python2.7/dist-packages/grgsm/
directory after installation.

sv1, 2015-04-11:
Hello Piotr,

pybombs really bombed my system!!! I reversed all installations of pybombs, and followed the good old cmake.

$git clone https://github.com/ptrkrysik/gr-gsm.git
$cd gr-gsm
$mkdir build
$cd build
$cmake ..
$make
$sudo checkinstall make install
$sudo ldconfig

And everything is working perfectly!!!! thanks!!!

ptrkrysik, 2015-04-11:
I've added description of manual compilation the the README.md

closed 2015-04-11 by ptrkrysik