ptrkrysik/gr-gsm #35: gr-gsm is so slow to open the airprobe GUI :(

simonepsp, 2015-04-10:
Hi,
I'm trying to run your gr-gsm python script but it takes a lot to start.

My terminal shows this output for about 15 minutes, then the airprobe GUI appears.

<b>└──╼ $airprobe_rtlsdr.py 
linux; GNU C++ version 4.7.2; Boost_105300; UHD_003.008.002-160-g3f0b770a

<b>"gr-osmosdr v0.1.4-16-g61184a19 (0.1.5git) gnuradio v3.7.6.1-327-g1f25ff92
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace"</b>

Can I do something to fix this delay?

Thank you

ptrkrysik, 2015-04-10:
I have no idea why it might take such long time to start, especially
that I don't observe problem like this on my side. Can you give
information about:
-GNU/Linux distribution that you use,
-GNU Radio version,

Does a simple GNU Radio flowgraph with RTL-SDR source and QT Freq sink
runs ok for you?

simonepsp, 2015-04-10:
I'm using Parrot OS (debian based) with GNUradio 3.6.5.1.

I tried to run a simple GNUradio flowchart with RTL-SDR,QT freq sink,options,parameter and ranges.
It still run slow... :( 

Thanks for your help

ptrkrysik, 2015-04-10:
If gnuradio blocks only blocks runs slow on your machine it's clearly
not gr-gsm's fault that it takes very long time for airprobe_rtlsdr
application to start on you machine.

Probably there is some more general issue with running new GNU Radio on
your system.

simonepsp, 2015-04-10:
yes, you're right.
I'll investigate on that :)

Thank you for your help ptrksysik. Your gr-gsm is wonderful :)
Bye.

ptrkrysik, 2015-04-10:
You can also try installing this docker:

https://github.com/marcelmaatkamp/docker-gnuradio-gr-gsm-ptrkrysik

prepared by @marcelmaatkamp . I want to add this as alternative method
of installation once I figure out what docker really is ;).

closed 2015-04-11 by ptrkrysik