ptrkrysik/gr-gsm #37: Contributions

romankh, 2015-04-12:
Hi @ptrkrysik, 

i would like to contribute to gr-gsm, and have a few questions:

* Do you prefer having an issue for every pull request ? 
* Do you prefer one pull request for every bug fix / feature / ... or an aggreation into one pull request 
* Have any limitations / rules / style guides ? 
* Do you support issue assignments to users ? For example, i recently implemented a demapper for TCH/F, and will implement the corresponding decoder in the next weeks. So if someone else is working on this feature, i would spend the time one something else.

ptrkrysik, 2015-04-13:
Hi @romankh 

First I would like to thank you for your contributions. I want to keep you as contributor so if you have any questions regarding the code - just ask:

* I prefer to have issue per bug before pull request is filled.
* I prefer to have pull requests that address the same problem/feature.
* As for the style I'm trying to keep to gnuradio's coding guide
https://gnuradio.org/redmine/projects/gnuradio/wiki/Coding_guide
not always it followed, but maybe it's time to start following it more strictly.
* You are the first person who is implementing something significant that is also on my list of things to be done. I think that the method proposed by you - that user working on something first files an issue with proposition to do something and I'll assign it is good. There are other things on my list of things to be done like:
   * Adding to a TCH demapper block ability to work with channel hopping (this is something dependent on your work for example).
   * Implementing GSM network monitor application.
   * Changing directory structure so addition of new blocks with gr_modtool would be easy again, while everything wouldn't be in the same directory. It's very hard for me after some time to figure out which file belongs to which block if they are not put into some cathegories. Probably it's even harder for new users.

It's also very good news that you are working on decoding of TCH/F. I've adapted once decoding of TCH/F in airprobe from OpenBTS - it was done in quick and very dirty way, but it worked, so it might be good place to start. Generally OpenBTS and osmocom.BB are very good reference implementations of many GSM internals.

romankh, 2015-04-13:
Thanks for the clarifications. 
I will open issues for the TCH/F demapper and decoder later this afternoon.

closed 2015-04-13 by romankh