ptrkrysik/gr-gsm #38: TCH/F demapper block

romankh, 2015-04-13:
Hi @ptrkrysik, 

as mentioned in an other conversation, i started an implementation of a demapper block for TCH/F. 
I orientated the implementation very much on your universal_ctrl_channel_demapper.

Here some of the cornerstones of my approach:
* The demapper is not as general as universal_ctrl_channel_demapper, i dont built any configuration options in. So one cannot configure the channel types in the 26-multiframe, there are just 2*3*4 bursts plus 1 SACCH/TF burst, which are mapped according to the specification. I didnt see any use for it in case of TCH. However, i didnt consider so far how the behaviour is specified for hopping channels. 
* The demapper has two different output ports, one for TCH bursts, and another one for ACCH, so we can send the bursts the right decoder only. 
* Currently i check if the stealing bits are set for TCH bursts, and if so i consider them as FACCH, and the bursts are sent to the ACCH port. The check is rather simple, if one frame has stealing

I have only tested this approach for correct ACCH messages (which looked fine so far), i wanted to test it with an TCH decoder as well, but this will take me some more time. 

So, what do you think ?

ptrkrysik, 2015-04-13:
I think it is right direction - first to place focus on TCH/F channels
decoding and only after that to think about something else, like channel
hopping or decoding of other traffic channels.

Separation of traffic channels flow and control channels to separate
outputs is exactly how I thought I would have done that.

The decoding of FACCH and SACCH can probably be done by current control
channels decoder. So this might be good milestone in implementing the
demapper - to have working control channels decoding. Or you already
have it working?

You will probably need some test data - are you able to collect it or do
you want me to send you some samples?

P.S. Regarding "there are just 234 bursts" - you probably intended to
write "there are just 24 bursts"

romankh, 2015-04-13:
Yes, you are right, 24 bursts.

I prepared myself a capture using OpenBTS, so for the moment this should be sufficient, thank you. 

FACCH and SACCH are already working, the ctrl channels decoder is decoding the bursts properly. Only traffic is untested so far. 

The code needs some clean up, afterwards i could push it as a first implementation.

ptrkrysik, 2015-04-16:
The SACCH decoding with use of demapper coded by you works. I've checked it using test data that I prepared for airprobe years ago:
https://svn.berlin.ccc.de/projects/airprobe/raw-attachment/wiki/DeModulation/piotr_downlink_second.tar.bz2

and this flowgraph:

https://app.box.com/s/sd5ckfkq2ptno2fdpbaxr23b151yqib5

I'm not able yet to get Wireshark to interpret the data correctly.

ptrkrysik, 2015-04-16:
When I set sub_type to 0x86 (header->sub_type = 0x86;) messages get interpreted correctly.
However I don't know where and by who such type was defined.

romankh, 2015-04-16:
I tested your flowgraph with your data, i just added a burstprinter to the tch-output, and ran it from the companion. 
Looks good for me, i have system information type 5, 5ter and 6. 
In system information type 6 i see the cell identity and location area id as described in second.description

romankh, 2015-04-16:
Ok, i had a similar issue in wireshark with the correct channel types.
0x86 should be SACCH of an SDCCH Channel, not for TCH. At least i believe so. 

This is how it looks for me: 

[image]

However i did a while ago an update of wireshark, i currently have version 1.12.1
Also, a time ago i stumbled over the gsmtap.h in either libosmocore or in the wireshark dissector, and there were a lot of additions compared to the version we use here.

romankh, 2015-04-16:
Sorry, to answer your question:
I believe 0x86 is `GSMTAP_CHANNEL_ACCH|GSMTAP_CHANNEL_SDCCH` and 0x89 `GSMTAP_CHANNEL_ACCH|GSMTAP_CHANNEL_TCH_F`
I saw this useage also either in the code of libosmocore or the wireshark dissector

ptrkrysik, 2015-04-16:
Sorry, looks like false alarm on my side. I turned back the sub_type to
the one you set and everything works ok.

However gsmtap.h header need to be updated if there is some newer
version available.

romankh, 2015-04-27:
Hi @ptrkrysik,

just for your information, i am still after that feature, but i am kind of busy until 08.05., thats why there is not much happening at the moment. I think i can finish the demapper until 08.05., after that i will have more time for the decoder.

ptrkrysik, 2015-05-04:
Hi @romankh ,
Thanks for keeping me informed. I the meantime I did some experiments with receiving uplink data and I got first decoded messages by only slightly changing the receiver block.

romankh, 2015-05-22:
I close this issue, block is implemented

closed 2015-05-22 by romankh