ptrkrysik/gr-gsm #39: GR-GSM usage

amlwwalker, 2015-04-13:
Hi I have the bladeRF and I am trying to collect some GSM data.
I have run `airprobe_bladerf.grc` and `airprobe_rtlsdr.grc` and I get the gui and I can see the 'wave' appear so I know my bladeRF is responding.
I don't get a stream of data appearing in my terminal however, and I dont see anything in wireshark. I think I have a base frequency at 960Mhz, but I was wondering if there is a clear way of knowing that it has found a frequency or not, so it is easier to know whether something is not working as expected, rather than just 'its not getting any data'

Thanks

romankh, 2015-04-14:
Hi, 

my suggestions: 

* try to find a frequency where you _know_ that there is a C0 of a BTS. 
For example, on some Samsung phones you can use the USSD code `*#0011#` to display a service menu with e.g. BcchFrq where your phone is camping on. Then you take that ARFCN number and feed it to an ARFCN calculator to get the right frequency. 

* Tune to that frequency and try to find the right ppm and gain values for your setup. You can use the sliders of the GUI for example

* If the message printer in your  .grc file is not disabled, and the flow graph is able to decode messages, the printer will print them either in gnuradio companion or the command line.

ptrkrysik, 2015-04-14:
Hi @amlwwalker ,

Generally if there is synchronisation with a BTS you should see immediately messages printed on the standard output (as @romankh has written). If you still have problem you can send few seconds of raw signal to me and I will check if there are no issues with it. Some people reported high DC offset on bladerf. The solution might be to gather signal on immediate frequency or to use a DC blocker that will estimate and remove DC offset.

ptrkrysik, 2015-05-30:
I'm closing this issue as I didn't get any reply for over month. If you @amlwwalker send additional data - I will reopen it.

closed 2015-05-30 by ptrkrysik