ptrkrysik/gr-gsm #42: Enhancement proposal: Addition of blocks for storing/readin received bursts to/from a file

ptrkrysik, 2015-04-21:
Offer for someone who wants to get taste of gr-gsm development but don't want to start with something overly complicated:

Currently storing of data is done by saving/reading raw signal to/from a file with use of gnuradio's file sink/source blocks. It is good method for testing the receiver, but it takes too much space during normal operation. Also there are many situations in which someone might want to process bursts multiple times but focussing on different part of the data each time. There is no need then to process the signal on each run to get bursts.

The solution is to add burst sink and source. Sink will have one message input (messages containing bursts) and will store bursts to a file (it's name will be given as a parameter). Source will read bursts from files and will send them through a message output.

romankh, 2015-05-22:
I haven't look very deep into gnuradio, but as far as I have seen pmt support serialization, so maybe we can just serialize the bursts and write them to the output file, and vice versa

ptrkrysik, 2015-05-22:
This would be great. I was looking (maybe not very carefully) for a gnuradio block that does that but I didn't find any. If pmt supports serialization - there should be such block in gnuradio definitely.

romankh, 2015-05-22:
I also just had a brief research, saw it in some pmt documentation. 
Will try it out, then give you feedback

romankh, 2015-05-26:
Serialization of pmt isn't a big deal, however i am not sure if it would be possible to have a sync_block have pmt message port in? Also, the Gnuradio BlocksCodingGuide says:

> The sync block allows users to write blocks that consume and produce an equal number of items per port. A sync block may have any number of inputs or outputs. When a sync block has zero inputs, its called a source. When a sync block has zero outputs, its called a sink.

So, do we have any drawbacks if we just stay with gr::block in this case ?

ptrkrysik, 2015-05-26:
Gnuradio takes care only of sample inputs and outputs to be synchronous
in case of gr::sync_block. So gr::sync_block in case of message only
blocks is no different than gr::block.

ptrkrysik, 2015-05-30:
I think we can close this issue. The blocks implemented by you @romankh  can be probably used for writing/reading any pmt objects.

closed 2015-05-30 by ptrkrysik