ptrkrysik/gr-gsm #43: OS X comile

csvendsen, 2015-04-25:
Could not compile under OSX without  changing
decryption_impl.cc
- bool uplink_burst = (be16toh(header->arfcn) & 0x4000) ? true : false;
 +bool uplink_burst = ((header->arfcn) & 0x4000) ? true : false;
bursts_printer_impl.cc
 - frame_nr = be32toh(header->frame_number);
  + frame_nr =  (header->frame_number);

ptrkrysik, 2015-04-25:
Hi @csvendsen ,

Thank you for the information.

After making the change mentioned by you everything compiles on OSX?

To fix the problem the change will be a little bit more complicated than proposed.
The function be16toh plays important role and it should be replaced by a portable version.

csvendsen, 2015-04-26:
Yeah after making the changes above it would compile but i cannot confirm if it worked as i had some other issues with gnuradio. Ill post back a report once i get everything running and ill try to compile again with the update.
Thanks Chris

ptrkrysik, 2015-04-30:
The problem should have been fixed by martinjlowm patches. I can't test it as I don't have OS X. Please keep me posted if it compiles for you.

Also - how you compile gr-gsm under OS X?

dmaynor, 2015-05-04:
Im compiling it under OSX now. @csvendsen  do you use Mac Ports, Homebrew or Xcode?

csvendsen, 2015-05-06:
Im using homebrew but still having issues with gnuradio and multiple python versions to be honest i haven't had a free moment to work on it yet. Put ill post an update when i can. Thanks

ptrkrysik, 2015-05-06:
Hi David,

Can you tell how you compile/install gr-gsm (and requiered software) on OSX?

dmaynor, 2015-05-06:
I am running the latest 10.10.3 on a Macbook pro.
$ uname -a
Darwin Davids-MBP.hsd1.ga.comcast.net 14.3.0 Darwin Kernel Version 14.3.0: Mon Mar 23 11:59:05 PDT 2015; root:xnu-2782.20.48~5/RELEASE_X86_64 x86_64

I have Macports installed. This required Xcode and the Apple X11 server (https://xquartz.macosforge.org/landing/) to be installed.
$ port
[Users/dmaynor] > version
Version: 2.3.3

I used macports to install gnuradio, uhd, bladerf, airspy, and Gqrx (Macports handles the dependencies). 
I cloned gr-gsm, entered the directory and created a build directory. I entered that directory and did a cmake.

```
$ git clone https://github.com/ptrkrysik/gr-gsm.git
$ cd gr-gsm
$ mkdir build
$ cd build
$ cmake ..
$ make
```

Cmake produces the following:
```
-- The CXX compiler identification is AppleClang 6.1.0.6020049
-- The C compiler identification is AppleClang 6.1.0.6020049
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- Found PkgConfig: /opt/local/bin/pkg-config (found version "0.28") 
Checking for GNU Radio Module: RUNTIME
-- checking for module 'gnuradio-runtime'
--   found gnuradio-runtime, version 3.7.6.1
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
-- Found GNURADIO_RUNTIME: /opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib  
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: PMT
-- checking for module 'gnuradio-runtime'
--   found gnuradio-runtime, version 3.7.6.1
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
-- Found GNURADIO_PMT: /opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib  
GNURADIO_PMT_FOUND = TRUE
-- checking for module 'volk'
--   found volk, version 0.1
-- Found VOLK: /opt/local/lib/libvolk.dylib  
-- checking for module 'cppunit'
--   found cppunit, version 1.13.2
-- Found CPPUNIT: /opt/local/lib/libcppunit.dylib  
-- Found Doxygen: /opt/local/bin/doxygen (found version "1.8.9.1") 
-- Found SWIG: /opt/local/bin/swig (found version "3.0.5") 
-- Found PythonLibs: /usr/lib/libpython2.7.dylib (found suitable version "2.7.6", minimum required is "2") 
-- Found PythonInterp: /opt/local/bin/python2.7 (found suitable version "2.7.9", minimum required is "2") 
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE - Failed
-- Configuring done
[... 120 lines cut ...]
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
Warning: Tag `XML_SCHEMA' at line 1134 of file `/Users/dmaynor/code/gr-gsm/build/docs/doxygen/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
Warning: Tag `XML_DTD' at line 1140 of file `/Users/dmaynor/code/gr-gsm/build/docs/doxygen/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
Warning: doxygen no longer ships with the FreeSans font.
You may want to clear or change DOT_FONTNAME.
Otherwise you run the risk that the wrong font is being used for dot generated graphs.
/Users/dmaynor/code/gr-gsm/lib/receiver/receiver_impl.h:90: warning: The following parameters of gr::gsm::receiver_impl::find_fcch_burst(const gr_complex *input, const int nitems, double &computed_freq_offset) are not documented:
  parameter 'computed_freq_offset'
/Users/dmaynor/code/gr-gsm/lib/receiver/receiver_impl.h:99: warning: argument 'computed_freq_offset' of command @param is not found in the argument list of gr::gsm::receiver_impl::compute_freq_offset(const gr_complex *input, unsigned first_sample, unsigned last_sample)
/Users/dmaynor/code/gr-gsm/lib/receiver/receiver_impl.h:151: warning: argument 'input_signal' of command @param is not found in the argument list of gr::gsm::receiver_impl::correlate_sequence(const gr_complex *sequence, int length, const gr_complex *input)
/Users/dmaynor/code/gr-gsm/lib/receiver/receiver_impl.h:151: warning: The following parameters of gr::gsm::receiver_impl::correlate_sequence(const gr_complex *sequence, int length, const gr_complex *input) are not documented:
  parameter 'input'
/Users/dmaynor/code/gr-gsm/lib/receiver/receiver_impl.h:178: warning: argument 'search_range' of command @param is not found in the argument list of gr::gsm::receiver_impl::get_norm_chan_imp_resp(const gr_complex *input, gr_complex *chan_imp_resp, float *corr_max, int bcc)
[... cut ...]
/Users/dmaynor/code/gr-gsm/lib/receiver/receiver_impl.h:193: warning: Found unknown command `\b_type'
/Users/dmaynor/code/gr-gsm/lib/receiver/receiver_impl.h:189: warning: The following parameters of gr::gsm::receiver_impl::send_burst(burst_counter burst_nr, const unsigned char *burst_binary, uint8_t burst_type, unsigned int input_nr) are not documented:
  parameter 'burst_type'
  parameter 'input_nr'
[100%] Built target doxygen_target
```

After this you can make install. The standalone airprobe works or I can open the flowgraphs in grc.

dmaynor, 2015-05-06:
@csvendsen are you using virtualenv?

ptrkrysik, 2015-07-02:
I copied the description to the gr-gsm mailing-list:
https://groups.google.com/forum/#!topic/gr-gsm/RNVVYpmmPzs

closed 2015-07-02 by ptrkrysik

dslokva, 2015-07-02:
Hi!!!

In my case (updated mac os 10.10, macports) - I can do cmake, make, make install with another warnings, but without errors.
```
MacBook-Pro:build dimas$ cmake ..
-- Build type not specified: defaulting to release.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: PMT
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
GNURADIO_PMT_FOUND = TRUE
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.5.
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   gnuradio-grgsm

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/dimas/gr-gsm/build
```

make output:

```
Scanning dependencies of target gnuradio-grgsm
[  2%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/receiver/receiver_impl.cc.o
[  4%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/receiver/clock_offset_control_impl.cc.o
/Users/dimas/gr-gsm/lib/receiver/clock_offset_control_impl.cc:106:16: warning: using integer absolute value function 'abs' when argument is of floating
      point type [-Wabsolute-value]
            if(abs(ppm) > 100) //safeguard against flawed measurements
               ^
/Users/dimas/gr-gsm/lib/receiver/clock_offset_control_impl.cc:106:16: note: use function 'std::abs' instead
            if(abs(ppm) > 100) //safeguard against flawed measurements
               ^~~
               std::abs
/Users/dimas/gr-gsm/lib/receiver/clock_offset_control_impl.cc:135:24: warning: using integer absolute value function 'abs' when argument is of floating
      point type [-Wabsolute-value]
                    if(abs(d_last_ppm_estimate-d_ppm_estimate) > 0.1)
                       ^
/Users/dimas/gr-gsm/lib/receiver/clock_offset_control_impl.cc:135:24: note: use function 'std::abs' instead
                    if(abs(d_last_ppm_estimate-d_ppm_estimate) > 0.1)
                       ^~~
                       std::abs
2 warnings generated.
[  6%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/receiver/cx_channel_hopper_impl.cc.o
[  9%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/misc_utils/bursts_printer_impl.cc.o
[ 11%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/misc_utils/extract_system_info_impl.cc.o
[ 13%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/demapping/universal_ctrl_chans_demapper_impl.cc.o
[ 16%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/demapping/tch_f_chans_demapper_impl.cc.o
[ 18%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/decoding/control_channels_decoder_impl.cc.o
[ 20%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/decoding/tch_f_decoder_impl.cc.o
In file included from /Users/dimas/gr-gsm/lib/decoding/tch_f_decoder_impl.cc:30:
/Users/dimas/gr-gsm/lib/decoding/tch_f_decoder_impl.h:76:53: warning: in-class initialization of non-static data member is a C++11 extension
[... 676 lines cut ...]
===========                      =======
CoreServices                       2516K
Dispatch continuations             16.0M
Kernel Alloc Once                     8K
MALLOC                            154.9M
MALLOC (admin)                       32K
STACK GUARD                        56.0M
Stack                              10.1M
VM_ALLOCATE                         348K
__DATA                             29.2M
__IMAGE                             528K
__LINKEDIT                        124.0M
__TEXT                            304.6M
__UNICODE                           552K
mapped file                        86.7M
shared memory                         4K
===========                      =======
TOTAL                             785.6M

```

All packages except PyQt4 was installed via macports. PyQt4 I installed from this instruction:
http://www.noktec.be/python/how-to-install-pyqt4-on-osx

Any ideas to make it working?

KirilShishkoski, 2017-12-15:
@dmaynor How did you manage to compile on macOS? I'm running macOS and I'm currently stuck on compiling libosmocore dependency which after running the 'cmake ..' command it throws an error 'No package libosmocore found'