ptrkrysik/gr-gsm #45: Error in Parrot Security OS.

tankaiha, 2015-04-30:
I just installed Parrot Security OS, which based on Debian, and use MATE desktop environment.
After install gr-gsm using pybombs, and run airprobe_rtlsdr.py, it says:

┌─[root@parrotsec]─[/usr/local/bin]
└──╼ #./airprobe_rtlsdr.py
linux; GNU C++ version 4.9.2; Boost_105500; UHD_003.009.git-144-g407e3086

gr-osmosdr v0.1.4-21-gbe9af0fe (0.1.5git) gnuradio 3.7.8git-129-gefd2116f
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf rfspace 

Then nothing happend. No windows display.
After waiting some minutes, I press Ctrl+C, it interrupt:

^CUsing device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
Traceback (most recent call last):
  File "./airprobe_rtlsdr.py", line 341, in <module>
    tb = airprobe_rtlsdr(fc=options.fc, gain=options.gain, samp_rate=options.samp_rate, ppm=options.ppm)
  File "./airprobe_rtlsdr.py", line 133, in __init__
    self.rtlsdr_source_0 = osmosdr.source( args="numchan=" + str(1) + " " + "" )
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 954, in make
    return _osmosdr_swig.source_make(args)
KeyboardInterrupt

What's the problem? Is it because the MATE desktop? Can it be solved?
Thank's.

ptrkrysik, 2015-05-04:
Looks like issue #35. This is probably general problem with new GNU Radio and Parrot OS which hangs up for few minutes before starting GNU Radio application. I don't have solution for it yet. I think the best place to ask is GNU Radio's mailing list (gnuradio-discuss https://gnuradio.org/redmine/projects/gnuradio/wiki/MailingLists).

ptrkrysik, 2015-05-30:
Hi @tankaiha ,

Do you have any update about grgsm on Parrot OS?

dikinugraha, 2015-06-12:
I have previously been installed gr-gsm, I did not find any problems. but I tried to update to the latest version UHD, when UHD successfully updated, I get problems when installing gr-gsm, it says : 
                                                                                                                                                           Installing from source: gr-gsm
Building:    (100%) [==========================================================]
Build failed. Re-trying with reduced makewidth and higher verbosity.
[  0%] [  5%] Built target grgsm_swig_swig_doc
Built target pygen_python_8662b
[ 11%] Built target pygen_apps_23458
[ 13%] Built target _grgsm_swig_swig_tag
[ 19%] Built target grgsm_swig_swig_2d0df
[ 25%] Built target pygen_swig_3240d
[ 27%] Building CXX object lib/CMakeFiles/gnuradio-gsm.dir/decoding/tch_f_decoder_impl.cc.o
In file included from /home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.cc:30:0:
/home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.h:76:55: error: a brace-enclosed initializer is not allowed here before ‘{’ token
/home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.h:76:92: error: ISO C++ forbids initialization of member ‘amr_nb_magic’ [-fpermissive]
/home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.h:76:92: error: making ‘amr_nb_magic’ static [-fpermissive]
/home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.h:76:92: error: invalid in-class initialization of static data member of non-integral type ‘const unsigned char [6]’
/home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.cc: In constructor ‘gr::gsm::tch_f_decoder_impl::tch_f_decoder_impl(gr::gsm::tch_mode, const string&)’:
/home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.cc:74:20: error: ‘amr_nb_magic’ was not declared in this scope
make[2]: *** [lib/CMakeFiles/gnuradio-gsm.dir/decoding/tch_f_decoder_impl.cc.o] Error 1
make[1]: *** [lib/CMakeFiles/gnuradio-gsm.dir/all] Error 2
make: *** [all] Error 2
Build failed. See output above for error messages.                       

What is the problem? please give solution
tanks's

dikinugraha, 2015-06-15:
hai @ptrkrysik 

I have previously been installed gr-gsm, I did not find any problems. but I tried to update to the latest version UHD, when UHD successfully updated, I get problems when installing gr-gsm, it says :
Installing from source: gr-gsm
Building: (100%) [==========================================================]
Build failed. Re-trying with reduced makewidth and higher verbosity.
[ 0%] [ 5%] Built target grgsm_swig_swig_doc
Built target pygen_python_8662b
[ 11%] Built target pygen_apps_23458
[ 13%] Built target _grgsm_swig_swig_tag
[ 19%] Built target grgsm_swig_swig_2d0df
[ 25%] Built target pygen_swig_3240d
[ 27%] Building CXX object lib/CMakeFiles/gnuradio-gsm.dir/decoding/tch_f_decoder_impl.cc.o
In file included from /home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.cc:30:0:
/home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.h:76:55: error: a brace-enclosed initializer is not allowed here before ‘{’ token
/home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.h:76:92: error: ISO C++ forbids initialization of member ‘amr_nb_magic’ [-fpermissive]
/home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.h:76:92: error: making ‘amr_nb_magic’ static [-fpermissive]
/home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.h:76:92: error: invalid in-class initialization of static data member of non-integral type ‘const unsigned char [6]’
/home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.cc: In constructor ‘gr::gsm::tch_f_decoder_impl::tch_f_decoder_impl(gr::gsm::tch_mode, const string&)’:
/home/diki/pybombs/src/gr-gsm/lib/decoding/tch_f_decoder_impl.cc:74:20: error: ‘amr_nb_magic’ was not declared in this scope
make[2]: *** [lib/CMakeFiles/gnuradio-gsm.dir/decoding/tch_f_decoder_impl.cc.o] Error 1
make[1]: *** [lib/CMakeFiles/gnuradio-gsm.dir/all] Error 2
make: *** [all] Error 2
Build failed. See output above for error messages.

What is the problem? please give solution
thank's

romankh, 2015-06-15:
@dikinugraha, 

the referenced commit should fix the problem. Did you try it ?

closed 2015-06-20 by ptrkrysik