ptrkrysik/gr-gsm #46: Error during install: fatal: reference repository

Rajnish357, 2015-05-03:
Hello Everyone,
I am first time working with pybombs and facing this error while gr gsm installation:---
Using gitcache at /usr/local/
fatal: reference repository '/usr/local/' is not a local directory.
fatal: reference repository '/usr/local/' is not a local directory.
fatal: reference repository '/usr/local/' is not a local directory.
______
This is what my pybombs setting looks like:
Settled on prefix: /usr/local
Initializing environmental variables...
False
None
Loading recipes ...
gituser = root
gitcache = /usr/local/
gitoptions = --single-branch
prefix = /usr/local/
satisfy_order = deb,src
forcepkgs = 
forcebuild = gnuradio,uhd,gr-air-modes,gr-osmosdr,gr-iqbal,gr-fcdproplus,uhd,rtl-sdr,osmo-sdr,hackrf,gqrx,bladerf,airspy
timeout = 30
cmakebuildtype = RelWithDebInfo
builddocs = OFF
cc = gcc
cxx = g++
makewidth = 4
static = False
__________
please help somone!

ptrkrysik, 2015-05-04:
@Rajnish357  gitcache option is strange for me. If also don't know what it is for - try removing it. 
This is content of my pybombs config.dat:
```
[config]
gituser = piotr
prefix = /usr/local
satisfy_order = deb,src
forcepkgs =
forcebuild = gnuradio,uhd,gr-air-modes,gr-osmosdr,gr-iqbal,gr-fcdproplus,uhd,rtl-sdr,osmo-sdr,hackrf,gqrx,bladerf,airspy
timeout = 30
cmakebuildtype = RelWithDebInfo
builddocs = OFF
cc = gcc
cxx = g++
makewidth = 4
gitcache =
gitoptions =
```

Rajnish357, 2015-05-05:
Thanks  alot @ptrkrysik Your soulution worked and solved the problem.

closed 2015-05-05 by Rajnish357