ptrkrysik/gr-gsm #48: After installing gr-gsm can't start it, ImportError libgnuradio-osmosdr

Rajnish357, 2015-05-05:
Hello everyone,
Even though this was my first time install with pybombs I somehow completed with the help of @ptrkrysik but after successful install I couldn't run gr-gsm with command:
`airprobe_rtlsdr.py`

it is giving this error output:
```
root@iHackU:~# airprobe_rtlsdr.py
Traceback (most recent call last):
  File "/usr/local/bin/airprobe_rtlsdr.py", line 19, in <module>
    import osmosdr
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/__init__.py", line 26, in <module>
    from osmosdr_swig import *
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 26, in <module>
    _osmosdr_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 22, in swig_import_helper
    _mod = imp.load_module('_osmosdr_swig', fp, pathname, description)
ImportError: /usr/local/lib/libgnuradio-osmosdr-0.1.5git.so.0.0.0: undefined symbol: _ZN3uhd6device4findERKNS_13device_addr_tENS0_15device_filter_tE
```

Please someone tell me how can I solve this issue ? I have already tried removing osmosdr package and reinstalling it.

ptrkrysik, 2015-05-05:
Try to run:
```
sudo ldconfig
```
maybe it will help.

Rajnish357, 2015-05-05:
Sorry @ptrkrysik it didn't helped, still same error

ptrkrysik, 2015-05-06:
What operating system/distribution do you use?
The problem is strange for me because it is caused by some issue with osmosdr - which is not my code.

Rajnish357, 2015-05-06:
Oh I didn't knew about that, by the way I am using I'm using Kali Linux 1.1.0a Latest release 32 Bit Version.
Any suggestion, where should I look for help regarding this ?

NinjaComics, 2015-05-08:
ping the gnuradio devs on freenode @Rajnish357

ptrkrysik, 2015-05-08:
**Do you remember if you did you run**:
```
./pybombs config forcebuild ' '
```
command? 

**You shouldn't** - the command enables pybombs to look for the packages in the distribution's repository. Kali linux has too old gnuradio in repository. Current version of GNU Radio should be built from source.

Rajnish357, 2015-05-12:
@ptrkrysik @NinjaComics Nothing helped still same error
So, i decided to do it on fresh kali install but during install stuck at:
```
Installing from source: qt4
--2015-05-13 03:47:59--  ftp://ftp.qt.nokia.com/qt/source/qt-everywhere-opensource-src-4.7.2.tar.gz
           => `qt-everywhere-opensource-src-4.7.2.tar.gz'
Resolving ftp.qt.nokia.com (ftp.qt.nokia.com)... 87.238.53.166
Connecting to ftp.qt.nokia.com (ftp.qt.nokia.com)|87.238.53.166|:21... failed: Connection timed out.
Retrying.
```

ptrkrysik, 2015-05-14:
The qt4 recipe in pybombs directs to not existing nokia ftp.
Change pybombs/recipes/qt4.lwr, line:
```
source: wget://ftp://ftp.qt.nokia.com/qt/source/qt-everywhere-opensource-src-4.6.2.tar.gz
```
to
```
source: wget://https://download.qt.io/archive/qt/4.6/qt-everywhere-opensource-src-4.6.2.tar.gz
```
and try again.

Rajnish357, 2015-05-14:
@ptrkrysik 
```
Extract qt-everywhere-opensource-src-4.6.2.tar.gz
 
    ./configure --prefix=/usr/local/ -opensource --confirm-license -L/usr/local//lib -L/usr/local//lib64 CC=gcc CXX=g++

Configuring: (100%) [==========================================================]
Configuration failed. Re-trying with higher verbosity.
make: *** No targets specified and no makefile found.  Stop.
Build failed. See output above for error messages.
```

ptrkrysik, 2015-05-15:
This is issue with pybombs. I've filled issue on their github page. Workaraound would be probably to compile qt4 manually:
cd pybombs/src/qt4
./configure --prefix=/usr/local/ -opensource --confirm-license -L/usr/local//lib -L/usr/local//lib64 && make && make install

Then add qt4 to in pybombs/config.dat to forcepkgs:
forcepkgs = qt4

and then do:
./pybombs install gr-gsm

closed 2015-06-20 by ptrkrysik