ptrkrysik/gr-gsm #50: Proposal: TCH/F decoder

romankh, 2015-05-11:
Hi @ptrkrysik,

today i did a quick&dirty integration of airprobe's approach to TCH decoding.
Did a quick test with a GSM-FR capture sample, did work well. Voice was very well recognizable, but not very good quality. Probably because it is just FR.

So, here my thoughts about the decoder:

* Feature set: 
    * one port in with bursts from demapper
    * one port out, if we handle the whole FACCH in the TCH-decoder, as in airprobe
    * select file for speech output
    * select coding
* reuse existing code
    * should we use the more recent code of OpenBTS, or the one from airprobe ? The one from airprobe is already reduced to what we need. According to the manual, they still only support GSM-FR. I tend more to use the airprobe version.
    * should we refactor the code we reuse, to fit more to gr-gsm, or try to integrate the code without modification for maintenance reasons (update source from OpenBTS). I would rather refactor, what do you think ?

ptrkrysik, 2015-05-12:
I will write regarding parts where I have comments. With the rest I agree without questions.
Regarding feature set:
-"one port out, if we handle the whole FACCH in the TCH-decoder, as in airprobe"
I'm not sure if integration of FACCH decoding in the same decoder is required. If strict synchronization between FACCH and TCH is not needed I would opt to keep them separate.

There is something in gnuradio called GSM Full Rate Audio decoder - maybe in the future the decoder could have output compatible with this block.

Regarding  general approach to code reuse - I would propose to use whatever approach that will be easier/faster for you. Working voice decoding will get gr-gsm to a new level anyway. However the code will look at the beginning it will be enclosed in a separate gnuradio block that can be quite easily improved.

romankh, 2015-05-13:
I think there is no need for us to keep them synchronized, but as FACCH is just a stolen TCH message they come together. So we could handle them together. 

You mentioned a change of the interfaces in #49, using 2 data blocks instead of each burst. But i am not sure, how this would help. My problem with FACCH is, that a message is transported in 8 bursts instead of 4, and there are differences in the processing. This is what the control channel decoder has to know, but i don't see a way the CCH decoder can find it out by itself. Or maybe i am missing something?

If we want FACCH to be decoded by the control channel decoder, we could add a parameter, where the user chooses the mode. Then we would keep the demapping of TCH as it is in #49. 

Another option would be separate decoder. This doesn't need to be configured, just drag & drop. While i generally like the approach to have as much as possible configurable, i think it would be more convenient from user perspective. I was thinking about writing a wrapper for the control channel demapper (for combined/non-combined CCH), because of situations where i just wanted to click a flow graph fast, and found myself configuring demappers. 

Third option would be making the Traffic channel decoder deal with the FACCH messages, as they are just stolen TCH messages, and they belong together, as they are used for call management only. We won't have an additional decoder, and no config in the control channel decoder. 

Regarding GSM Full Rate Audio decoder, this is great, i will take a closer look.

ptrkrysik, 2015-05-13:
> I think there is no need for us to keep them synchronized, but as FACCH
> is just a stolen TCH message they come together. So we could handle them
> together.
>
> You mentioned a change of the interfaces [...]

A given FACCH message is carried in 8 consecutive bursts. In each  burst
one of the halves contains the message - because of interleaving that is
used. Other control channels are carried by 4 consecutive bursts in both
halves (there is no block interleaving). In both cases there are in
total 8 blocks carrying a message - so there is possibility to have one
control channels decoder expecting at the input 8 blocks.

Strategically it will be good to keep number of GNU Radio blocks as low
as possible and to have blocks that (as much as possible) don't overlap
in provided functionalities.

However it doesn't have to be this way from the beginning. My opinion is
that it is better to have the project going forward even if it doesn't
keep to my subjective ideal.

> If we want FACCH to be decoded by the control channel decoder, we could
> add a parameter, where the user chooses the mode. Then we would keep the
> demapping of TCH as it is in #49
> <https://github.com/ptrkrysik/gr-gsm/pull/49>.

This might be good solution - maybe even better than integrating FACCH
decoder with TCH decoder.

> Another option would be separate decoder. This doesn't need to be
> configured, just drag & drop. While i generally like the approach to
> have as much as possible configurable, i think it would be more [...]

Separate decoder for FACCH will be almost copy of the control channels
decoder that is already there so one block with configuration option is
in my opinion better.

The idea of wrapper for the demapper is good. Generally idea to civilize
it's configuration is noble - the univeral demapper requires manual
configuration and it lets a lot of configurations that don't make sense
in reality. In reality there are only some predefined combinations of
channels that can be transmitted by a BTS - configured by few parameters
that are broadcasted in System Information messages.

> Third option would be making the Traffic channel decoder deal with the
> FACCH messages, as they are just stolen TCH messages, and they belong
> together, as they are used for call management only. We won't have an
> additional decoder, and no config in the control channel decoder.

This makes sense.

In sum I'm opting for a separate decoder for control channels and a
separate one for traffic channels - as they use different coding
schemes. Integration of TCH and FACCH decoding in one block is also
acceptable for me.

romankh, 2015-05-14:
> A given FACCH message is carried in 8 consecutive bursts. In each  burst
> one of the halves contains the message - because of interleaving that is
> used. Other control channels are carried by 4 [...]

Hmm, my understanding of the specification of FACCH was like: 
if the stealing bits are set, take the even bits of the first 4 bursts, and the odd bits of the last 4 bursts.
I got this from GSM 05.03, in 4.2.5. 

Anyway, i'll investigate issue, and then decide between the configurable control channel decoder and the integration with TCH decoder.

ptrkrysik, 2015-05-14:
Your understanding is good - mine was wrong (in the part how block diagonal interleaving works).
From GSM 05.03 3.1.3:
i(B,j) = c(n,k),
for k = 0,1,...,455
n = 0,1,...,N,N+1,...
B = B0 + 4n + (k mod 8)
j = 2((49k) mod 57) + ((k mod 8) div 4)

The "((k mod 8) div 4)" is responsible for what you are talking - it is 0 for the first four bursts and 1 for the last four bursts. In light of this single block for TCH and FACCH decoding seems to be good idea.

akibsayyed, 2015-05-18:
Guys I can provide you people with EFR and FR decode code based on libosmocore

romankh, 2015-05-18:
Hi @akibsayyed, 

thanks, but i have FR and EFR mostly done. 
But if you like, you can add more AMR bit rates to decoding, currently I have AMR 12.2 untested, would be nice to have decoding of lower bit rates as well.

romankh, 2015-05-22:
I close this issue, as basic implementation is done. 
Will open a new issue for improvement of the decoder, e.g. the implementation of (more) AMR-codings.

closed 2015-05-22 by romankh

ptrkrysik, 2015-05-22:
It's great work @romankh. I'm able to decode my old test files again :). Feels like when for the when I heard GSM voice data decoded with my gsm-receiver (which I'm not able to compile on my current system) for the first time.

romankh, 2015-05-23:
Thank you. I was quite happy when I got it working.