ptrkrysik/gr-gsm #52: Unable to decode DCS or PCS

akibsayyed, 2015-05-18:
Dear Piotr 

I am testing your implementation on DCS, PCS it used to work before but latest build is not working

ptrkrysik, 2015-05-19:
@akibsayyed this is strange as DCS and PCS are exactly like GSM900 - just on different carrier frequency. Are you able to point the commit number or at least quite precise date when it worked?

akibsayyed, 2015-05-19:
I tried it approx in august 2014 it was working. I will send you cfile for
same today.

ptrkrysik, 2015-05-19:
This is problem with DC offset of your hardware.  For example it can be tackled by inserting DC blocker block between signal source and the "GSM input adaptor".

ptrkrysik, 2015-05-22:
There might be something wrong with frequency correction. In the file that I downloaded from you (the link is in #53) I found out that I get much better results when I put central frequency 931MHz and not 1931MHz that it was really recorded on. I will look into this issue in the free time.

ptrkrysik, 2015-05-30:
Hi @akibsayyed,

You were right - there was an error. I think I corrected it in this commit:
https://github.com/ptrkrysik/gr-gsm/commit/18f769026cdda9e1bbacd772c5afb277cf56da00

ptrkrysik, 2015-05-30:
I'm closing it for now. If you still have problem I will reopen it.

closed 2015-05-30 by ptrkrysik