ptrkrysik/gr-gsm #55: Proposal: Wrapping universal control channel demapper

romankh, 2015-05-22:
Hi @ptrkrysik, 

I'll keep spamming you with issues for a while  :)
We already talked about wrapping the universal control channel demapper.
The issue is straight-forward, I just created 3 Blocks, where I hide the params starts_fn_mod51 and channel_types, and set them to values depending on their usage. The params just go to universal_ctrl_chans_demapper as before. 

I have made 3 of them:
gsm_bcch_ccch_demapper (BCCH + CCCH demapper) 
gsm_bcch_ccch_sdcch4_demapper (BCCH + CCCH + SDCCH/4 demapper)
gsm_sdcch8_demapper (SDCCH/8 demapper)

What do you think about the naming ?

ptrkrysik, 2015-05-22:
I agree that such blocks would be very helpful. First I considered implementing one block for the first timeslot of C0 that would take BS-AG-BLKS-RES, CCCH-CONF and BS-PA-MFRMS (3gpp 04.08 10.5.2.11 Control Channel Description) as parameters and configure itself accordingly. Your approach is simpler and probably it will serve the purpose better.

ptrkrysik, 2015-05-30:
I've merged the blocks implemented by you with the mainline. In a free moment I will add them to the current apps instead of Universal demapper.

closed 2015-05-30 by ptrkrysik