ptrkrysik/gr-gsm #56: Scanning a frequency range

romankh, 2015-05-22:
I would like to easily scan a frequency range for C0s, and extract the most interesting parts of the system information messages

I am aware of the system information extractor, which I probably would extend to extract more information. But what is the Gnuradio-way to iterate through the channels ?
Before I start reinventing the wheel, can this already be done using some default block ? Scanning a frequency range with given steps sounds pretty common to me

rpp0, 2015-05-22:
Not sure whether this is the best way to do it, but you could use the Function Probe block. This block will periodically call a method and assign the return value to a specified variable. In your case you could create a small Python block that increments the frequency variable every time the method is called. This is how I made a spectrum scanner with my SDR some time ago.

romankh, 2015-05-26:
Thanks, I will try that out, but it's what I was hoping to find.
My ideal block would take start freq, end freq, and channel distance, and on execution, it sets the frequency and in the best case the cell allocation, such that I have the arfcn in system info extractor (because it uses arfcn as id for the entries). When it reaches the end freq, execution is terminated.
I will try to find out, if and how that can be achieved.

ptrkrysik, 2015-05-26:
The UHD block (block for controlling USRPs) has message input for commands like "change central frequency". It also emits stream tag when the device switch frequency - although from what I have read on gnuradio mailing list it migh come before PLL on the device finish locking onto the frequency. Maybe with this infrastructure what you want to do would be easier to implement.

romankh, 2015-05-28:
Thanks.
So maybe the easier way is to do it in python.

closed 2015-05-28 by romankh

ptrkrysik, 2015-05-29:
Implementing it in Python is good if you can accept that "Run to completion" won't work for your flowgraph (at least until GNU Radio guys fix this issue).