ptrkrysik/gr-gsm #57: AMR Decoding

akibsayyed, 2015-05-23:
@romankh  @ptrkrysik 

Here is samples with AMR HR 
matrixshell.com/cfile/HR_AMR/HR_AMR_.cfile
matrixshell.com/cfile/HR_AMR/HR_AMR_abis.pcapng
matrixshell.com/cfile/HR_AMR/HR_AMR_s2.pcapng

akibsayyed, 2015-05-23:
@romankh @ptrkrysik 
Here is samples for AMR FR
matrixshell.com/cfile/amr_call_data/Call_AMR.cfile
matrixshell.com/cfile/amr_call_data/amr.pcap.pcapng

romankh, 2015-05-23:
Thanks a lot @akibsayyed, this is gonna be really helpful.

Regarding HR: I will keep going after FR first, as HR would need more work on demapping and decoding, and I am not sure, if HR still plays a major role on modern networks. 

The AMR bitrate in the capture you provided is been set to 5.90 kbit/s. 
Is it possible to explicitly configure the AMR bitrate in your environment ? If so, it would be great to have captures for bitrates 10.2, 7.95, 7.40, 6.70, 5.15, 4.75 also. 
I know that would be some effort, I don't need it immediately.

akibsayyed, 2015-05-23:
Dont worry I will do it in my free time.

ptrkrysik, 2015-05-25:
Half Rate is still a bit mysterious for me as I haven't had opportunity to see it in the wild. The literature says that one burst is shared by two different subscribers. For me this is strange as it would mean they both have to transmit burst midamble in the same time. Thanks @akibsayyed for the captures. I will have a look.

romankh, 2015-06-03:
@akibsayyed, I am currently working on the integration of AMR-decoding, I came very far, however I am stuck with a problem. Would it be possible for you to record and provide a capture of TCH-F with AMR 12.2 codec for debugging ? That would make it easier for me to track the problem down.

akibsayyed, 2015-06-03:
I will check if i can use 12.2 codec  i cannot guarantee that i can give
you samples. but i have working demo of amr5.9 codec decode that i can
share. its complete C code.

akibsayyed, 2015-06-03:
Please check code attached in mail

akibsayyed, 2015-06-03:
here is link for same
www.matrixshell.com/cfile/amr5.9.rar

romankh, 2015-06-03:
Thanks for the quick response, can you provide me the password for archive ?

akibsayyed, 2015-06-03:
i created other file with same contents.
hxxp://matrixshell.com/cfile/amr5.9.tar

romankh, 2015-06-03:
Thanks a lot.

romankh, 2015-06-07:
@akibsayyed,

your decoder was really helpful, finally I found that bug.
I have created pull request #63, if you would like to test

romankh, 2015-07-21:
Update: I currently have no access to an device that is able to capture the GSM-1800 band, at the moment I can capture only GSM-900 with my hardware.  Now having A5/3 decryption, I tried to capture an AMR traffic channel to my phone on the GSM-900 band, but as soon as the TCH channel is established, my phone receives a handover command to an other BTS on the GSM-1800 band 

So I will have to wait until I get access again to an USRP to test the other AMR rates.

ptrkrysik, 2015-08-15:
Hi Roman,

What is status of this issue? I know you already integrated A5/3 but I haven't tested AMR support myself.

romankh, 2015-08-15:
Hi Piotr, 

haven't been successful capturing AMR TCH speech channels for testing so far.
I have to share access to a N210 with another guy, I believe I won't get the N210 in the next two weeks. That was limiting me to the GSM 900 band with RTL-SDR.

Besides the BTS the handed me over to a 1800 ARFCN right after call setup, there was only one more BTS without hopping TCH chans. I successfully captured TCH just to find out that it was TCH/H. A little bit frustrating, but I take it as a motivation to aim for an TCH/H demapper/decoder in the future.

I  ordered an RTL-SDR receiver with an E-4000 chip, received it on Thursday, but didn't have much opportunity to go after TCH captures yet.

So our status is still the same, AMR 12.2 and 5.9 are tested and should work, the other bit rates are untested. But I think we can close this issue, and if we discover a bug with some bit rate, I can open open a new one for the particular bug. I expect that it is going to take some time until we get test samples for all bit rates, as I don't know any method to enforce the usage of a specific AMR rate.

mamoonali, 2016-05-13:
hi,
1. how to determine whether a network is using fr,efr ,amr?

2. is it possble to decode tch h-half rate channel?

3.  i have a sample in which in can get the calling subscrier number . But unable to decode the voice.

4. does gr-gsm works fine for amr codec?

5.grgsm works fine for open BTS as i have intercepted voice calls in FR mode.

     regards..

ptrkrysik, 2016-05-13:
Hi,

Responses below.

W dniu 13.05.2016 o 16:37, mamoonali pisze:
> hi,
> 1. how to determine whether a network is using fr,efr ,amr?
> 
This can change from call to call. I think it is transmitted in
Assignment Complete message in Channel Mode element, but I'm not 100% sure.
> 
>     is it possble to decode tch h-half rate channel?
> 
No - and I haven't seen such channel yet. If you have some test
recording with half rate channel - I'm interested to see this.

>  2.
> 
>     i have a sample in which in can get the calling subscrier number .
>     But unable to decode the voice.
Can you try to identify codec version that was used?

>  3.
> 
>     does gr-gsm works fine for amr codec?

AMR has multiple variants. Not all of them were tested.

> 5.grgsm works fine for open BTS as i have intercepted voice calls in FR
> mode.

Good to hear this :).

> |regards.. |
> 

Best Regards,
Piotr Krysik

mamoonali, 2016-05-14:
sir,
    this is the captured file..i will b obliged if you go through it and
tell me how to decode the voice from it.
    FREQUENCY of capture= 939.5MHz, GSM 900..   My project on it needs to
be submitted next week and i am stuck up here.
    I guess it is TCH half rate and codec --i am not able to get. Require
your much needed help.
         Best Regards...​
 capture88.cfile
<https://drive.google.com/file/d/0B0m86_vKuaSEczZsYW9STVBnUEU/view?usp=drive_web>
​

mamoonali, 2016-05-14:
sir,
 in the capture 88 .cfile i used RTL-SDR and kali rolling 2016

    best regards
    thank you

ptrkrysik, 2016-05-15:
Mamoon,

If your task is to decode half rate traffic channels I wont be able to
help you with finishing your project. Speaking realistically this might
be too much work to do this in one week. Especially that I have other
tasks currently that have higher priority (job and other things related
to my free software projects). I would like to add implementation of
half rate channels (it is very interesting for me how the bursts are
transmitted in half rate uplink) - but this moment to do this is not too
convenient for me.

Maybe try to adapt your expectations so probability of successful ending
of your project will be more probable (i.e. if you need to have any
voice channel decoded stick to Full Rate codec - which is already
supported and quite well tested).

Best Regards,
Piotr Krysik

W dniu 14.05.2016 o 08:17, mamoonali pisze:
> sir,
> this is the captured file..i will b obliged if you go through it and
> tell me how to decode the voice from it.
> FREQUENCY of capture= 939.5MHz, GSM 900.. My project on it needs to
> be submitted [...]

mamoonali, 2016-05-15:
dear sir,

1.  did u see the capture88.cfile i sent you. It is not be half rate.
2.  can u have a look at it please.and tell me how to decode voice..i will
be obliged.

     best regards...

ptrkrysik, 2016-05-17:
Hi Mamoon,

Can you contact me by e-mail? I think I have good news for you.

prkrmx, 2016-08-21:
Hi Piotr and all!

>      is it possble to decode tch h-half rate channel?
> No - and I haven't seen such channel yet. If you have some test
> recording with half rate channel - I'm interested to see this.

i want to share couple cfiles with half rate channel. hope that is help you finish your great work)
https://mega.nz/#F!vwlkFSxY!AaS1uWRJyMGg-P7jhVwIwg

Best Regards 
Max Parker

ptrkrysik, 2016-08-25:
W dniu 21.08.2016 o 08:49, Max Parker pisze:
> Hi Piotr and all!
>
>     |is it possble to decode tch h-half rate channel? |
>
>     No - and I haven't seen such channel yet. If you have some test
>     recording with half rate channel - I'm interested [...]
Thanks Max,

I've downloaded the files. Soon there should be support of TCH/H
channels decoding in libosmocore so it should be much simpler to add
their support in gr-gsm. Your data will be very precious for testing it.

-- 
Best Regards,
Piotr Krysik

pwr-krn, 2016-09-22:
Hello Piotr!

I can also provide some half rate channel cfiles from another network.
If you need, tell me, then i will upload some.
Bye!

ptrkrysik, 2016-10-01:
@pwr-krn , I have some half rate captures but in case they don't work I will try to contact you here. Half rate decoding should soon land in libosmocore so adding decoding will be a matter of making a proper demapper for half rate channels.

ptrkrysik, 2016-10-01:
I'm closing the issue here as AMR support is already there. I will open another issue for adding tests for testing AMR support (among other things).

closed 2016-10-01 by ptrkrysik