ptrkrysik/gr-gsm #61: Make errors when installing with Gqrx PPA install

dsound, 2015-06-01:
Hi All,
I'm rather new to the Linux environment so bare with me. I decided to go the Gqrx PPa route to install gnuradio. When I try to make gr-gsm, I get several errors indicated that the cmake checklist can't find gnuradio and dependencies. Could someone give me exact instructions on how to set the path for gnuradio and cmake modules? Here's the error output:

-- Build type not specified: defaulting to release.
-- Boost version: 1.55.0
-- Found the following Boost libraries:
--   filesystem
--   system
CMake Warning at CMakeLists.txt:92 (find_package):
  By not providing "FindGnuradio.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Gnuradio",
  but CMake did not find one.

  Could not find a package configuration file provided by "Gnuradio" with any
  of the following names:

    GnuradioConfig.cmake
    gnuradio-config.cmake

  Add the installation prefix of "Gnuradio" to CMAKE_PREFIX_PATH or set
  "Gnuradio_DIR" to a directory containing one of the above files.  If
  "Gnuradio" provides a separate development package or SDK, be sure it has
  been installed.

-- checking for module 'volk'
--   package 'volk' not found
-- Could NOT find VOLK (missing:  VOLK_LIBRARIES VOLK_INCLUDE_DIRS) 
CMake Error at CMakeLists.txt:98 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!

ptrkrysik, 2015-06-20:
Does gr-gsm installs when you don't have gqrx?

ptrkrysik, 2015-07-02:
This is most probably not related to an error in gr-gsm but problem with installation of other required component. In case of updates in this story please continue the discussion on gr-gsm mailin list (https://github.com/ptrkrysik/gr-gsm/#mailing-list) - just add link pointing to this issue in the first post.

closed 2015-07-02 by ptrkrysik