ptrkrysik/gr-gsm #65: gr-gsm make fails on RPi2

wesleygoatley, 2015-06-12:
Hi, all dependencies are installed, along with gnuradio 3.7.5, though I'm getting this error when trying to make:

 $ sudo make
[  2%] Building CXX object lib/CMakeFiles/gnuradio-gsm.dir/decoding/tch_f_decoder_impl.cc.o
In file included from /home/pi/gr-gsm/lib/decoding/tch_f_decoder_impl.cc:30:0:
/home/pi/gr-gsm/lib/decoding/tch_f_decoder_impl.h:56:55: error: a brace-enclosed initializer is not allowed here before ‘{’ token
/home/pi/gr-gsm/lib/decoding/tch_f_decoder_impl.h:56:92: error: ISO C++ forbids initialization of member ‘amr_nb_magic’ [-fpermissive]
/home/pi/gr-gsm/lib/decoding/tch_f_decoder_impl.h:56:92: error: making ‘amr_nb_magic’ static [-fpermissive]
/home/pi/gr-gsm/lib/decoding/tch_f_decoder_impl.h:56:92: error: invalid in-class initialization of static data member of non-integral type ‘const unsigned char [6]’
/home/pi/gr-gsm/lib/decoding/tch_f_decoder_impl.cc: In constructor ‘gr::gsm::tch_f_decoder_impl::tch_f_decoder_impl(gr::gsm::tch_mode, const string&)’:
/home/pi/gr-gsm/lib/decoding/tch_f_decoder_impl.cc:62:20: error: ‘amr_nb_magic’ was not declared in this scope
lib/CMakeFiles/gnuradio-gsm.dir/build.make:353: recipe for target 'lib/CMakeFiles/gnuradio-gsm.dir/decoding/tch_f_decoder_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-gsm.dir/decoding/tch_f_decoder_impl.cc.o] Error 1
CMakeFiles/Makefile2:208: recipe for target 'lib/CMakeFiles/gnuradio-gsm.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-gsm.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Any thoughts on how this could be fixed?

romankh, 2015-06-12:
Hi, 

looks like the Raspi compiler is not capable of C++ 11. 
For now you can remove `const unsigned char amr_nb_magic[6] = { 0x23, 0x21, 0x41, 0x4d, 0x52, 0x0a };` from header file lib/decoding/tch_f_decoder_impl.h, and declare the array in the constructor of lib/decoding/tch_f_decoder_impl.cc, somewhere before line 60.

wesleygoatley, 2015-06-12:
Many thanks - would you mind clarifying what the addition to lib/decoding/tch_f_decoder_impl.cc needs to be?

romankh, 2015-06-12:
`const unsigned char amr_nb_magic[6] = { 0x23, 0x21, 0x41, 0x4d, 0x52, 0x0a };` does the job

wesleygoatley, 2015-06-12:
Ah, thanks for that - that did the trick, and it appeared to install successfully.  However, when running airprobe_rtlsdr.py, I get this error response:
Traceback (most recent call last):
  File "/usr/local/bin/airprobe_rtlsdr.py", line 19, in <module>
    import grgsm
  File "/usr/local/lib/python2.7/dist-packages/grgsm/__init__.py", line 45, in <module>
    from grgsm_swig import *
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 28, in <module>
    _grgsm_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_grgsm_swig', fp, pathname, description)
ImportError: libgnuradio-gsm.so: cannot open shared object file: No such file or directory

Any thoughts?

romankh, 2015-06-15:
You may have to run `sudo ldconfig`.
If that doesn't work, you may have to add the path to libgnuradio-gsm.so to LD_LIBRARY_PATH

ptrkrysik, 2015-06-20:
@wesleygoatley can you check and report if gr-gsm works after doing what @romankh has written?

wesleygoatley, 2015-06-20:
Will do - am out of the city right now, looking forward to checking this fix when I return.

closed 2015-06-23 by wesleygoatley