ptrkrysik/gr-gsm #67: [Failed] sudo ./pybombs install gr-gsm : Configuration failed

aymeric75, 2015-06-16:
Hi everyone,

I'm trying hard to make gr-gsm work, but i have an issue when executing the command: sudo ./pybombs install gr-gsm.

Here are the last line of the shell :

Paramétrage de python-cairo-dev (1.8.8-1) ...
installation ok via: deb
Installing from source: gnuradio
remote: Counting objects: 5324, done.
remote: Total 5324 (delta 0), reused 0 (delta 0), pack-reused 5324
Receiving objects: 100% (5324/5324), 1.52 MiB | 93 KiB/s, done.
Resolving deltas: 100% (3844/3844), done.
 
    CC=gcc CXX=g++ cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=/usr/local/  -DENABLE_GR_FCD=ON -DENABLE_DOXYGEN=OFF -DENABLE_GR_LOG=OFF -DENABLE_PERFORMANCE_COUNTERS=ON -DCMAKE_CXX_FLAGS=-fpermissive 

Configuring: (100%) [==========================================================]
Configuration failed. Re-trying with higher verbosity.
make: *** Pas de cibles spécifiées et aucun makefile n'a été trouvé. Arrêt.

Thanks

Aymeric

ptrkrysik, 2015-06-20:
Please tell what system do you use and attach output of:
sudo ./pybombs -v install gr-gsm

ptrkrysik, 2015-07-02:
This doesn't have continuation so it's hard to tell if it's problem with gr-gsm. In case of updates in this story please continue the discussion on gr-gsm mailin list (https://github.com/ptrkrysik/gr-gsm/#mailing-list) - just add link pointing to this issue in the first post.

closed 2015-07-02 by ptrkrysik