ptrkrysik/gr-gsm #68: Dependency LIBUSB_FOUND = FALSE

aymeric75, 2015-06-17:
Hi,
I've the following error when running this command: sudo ./pybombs install -v gr-gsm

--   Dependency LIBUSB_FOUND = FALSE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gr-fcd but configuration checked failed
Call Stack (most recent call first):
  gr-fcd/CMakeLists.txt:35 (GR_REGISTER_COMPONENT)

I checked if libsub was installed on my computer and i can see some files like "libusb-0.1-4_2%3a0.1.12-25_amd64.deb" in /var/cache/apt/archives, so i guess libusb is installed.

Also i added "libusb" to gnuradio.lwr depends...I'm stuck now. Your help would be great.
Thanks
Aymeric

aymeric75, 2015-06-17:
Actually, i worked, i reinstalled libusb as following:

wget http://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.9/libusb-1.0.9.tar.bz2
tar xjf libusb-1.0.9.tar.bz2
cd libusb-1.0.9
./configure
make
sudo make install

ptrkrysik, 2015-06-20:
This looks more like problem with gnuradio or gr-fcd installation. Can you tell something more about system that you use?

aymeric75, 2015-06-20:
To correct this error i had to install a version of libusb manually: libusb-1.0-0-dev_1.0.11-1_amd64

Date: Fri, 19 Jun 2015 22:31:41 -0700

ptrkrysik, 2015-07-02:
This is most probably not related to an error in gr-gsm but problem with installation of other required component. In case of updates in this story please continue the discussion on gr-gsm mailin list (https://github.com/ptrkrysik/gr-gsm/#mailing-list) - just add link pointing to this issue in the first post.

closed 2015-07-02 by ptrkrysik