ptrkrysik/gr-gsm #69: [error] from Cheetah import Template ImportError: No module named Cheetah

aymeric75, 2015-06-19:
Hi,

   I'm still trying to install gr-gsm and i've another trouble with the " sudo ./pybombs install -v gr-gsm" command. 
Indeed there is an error concerning cheetah (which i installed, and for which i changed the PYTHONPATH).

I don't know what to do now, any idea ?

Thx 

Aymeric

ptrkrysik, 2015-06-19:
Can you give some details of:
-system that you use,
-error messages that you get (if you don't get any - try to run pybombs
with "-v" option)?

Without this it's hard to tell what is wrong and where the installation
stopped.

aymeric75, 2015-06-20:
Hi,

Thank you very much for your help.

 So now i just have the problem of the "Cheetah import" going on.
I'm using Kali Linux 1.1 and Cheetah-2.4.4, i have several python version on my computer and i think the problem is about the Cheetah that i installed which works only with a specific version of python, but i don't know how to install Cheetah for a specific version.

Here is the detailed error message:

Generating osmosdr_source.xml
[GCC 4.9.2]
Traceback (most recent call last):
  File "/home/aymeric/pybombs/src/gr-osmosdr/grc/gen_osmosdr_blocks.py", line 480, in <module>
    params = ''.join([parse_tmpl(PARAMS_TMPL, n=n, sourk=sourk) for n in range(max_num_channels)])
  File "/home/aymeric/pybombs/src/gr-osmosdr/grc/gen_osmosdr_blocks.py", line 451, in parse_tmpl
    from Cheetah import Template
ImportError: No module named Cheetah
make[2]: *** [grc/rtlsdr_source.xml] Error 1
make[2]: *** Waiting for unfinished jobs....

Cheers 

Aymeric

Date: Fri, 19 Jun 2015 12:25:54 -0700

aymeric75, 2015-06-21:
Hi,

Here are more details about my Cheetah import problem.

I think that when the script call the file called "gen_osmosdr_blocks.py", it's creating a new process which is made in a way that Python does not find the repertory of Cheetah.

Have a good sunday

Cheers

Aymeric

Date: Fri, 19 Jun 2015 12:25:54 -0700

aymeric75, 2015-06-21:
A little more thing to add, when i type "python" in the terminal (the default version is Python2.7.9 and do "import Cheetah" i have no error. But when i type python2.7 (same version displayed) and i type "import Cheetah", i have the error "No module named Cheetah".

So i think it's really a problem of Python version and Cheetah.

Thx

Date: Fri, 19 Jun 2015 12:25:54 -0700

aymeric75, 2015-06-22:
I found the error finally, i copy pasted a python2.7 repertory in another one already existing, so i supressed it and it works (pretty silly i know).

Now i have another error when i want to try the "airprobe_file.py", it send me this error:

  File "/usr/local/bin/airprobe_file.py", line 8, in <module>
    from gnuradio import blocks

Date: Fri, 19 Jun 2015 12:25:54 -0700

ptrkrysik, 2015-06-25:
@aymeric75 - the problem with Cheetah was specific for your configuration and everything should work ok in general situation when someone installs gr-gsm on a clean system?

aymeric75, 2015-06-25:
@ptrkrysik yes it was specific, actually i installed severa time python with several version, and i don't know why but it happened that i put a Python2.7 directory in the original Python2.7 directory, and that created a conflict with module importation (not only with Cheetah).

ptrkrysik, 2015-06-25:
I'm closing the issue then

closed 2015-06-25 by ptrkrysik