ptrkrysik/gr-gsm #70: Code refactoring from other projects (Osmocom, OpenBTS...)

ptrkrysik, 2015-06-20:
There are a lot of functionalities that are already implemented in other projects on different layers of GSM protocols. Most advanced among them are Osmocom and OpenBTS. I want to start the discussion here to create wish-list with functionalities that are implemented in them that should be imported to gr-gsm and how to do such imports.

romankh, 2015-06-21:
From my perspective the most desirable feature would be receiving and decoding uplink channels. You once mentioned it may be possible with moderate effort. 

Besides that, GPRS would also be a nice-to-have for me, as I still seeing it being used regularly in networks. EDGE seems also to be an extension to GSM, however I didn't research anything in this direction, also I believe unlike GPRS, it is not implemented in OpenBTS or Osmocom.

On a long-term view I could imagine extending gr-gsm to UMTS could also be interesting, and OpenBTS has a branch that supports UMTS. 

Regarding refactoring code in general: there is some code like the Viterbi-decoders that control channel decoder and traffic channel decoder have in common. I am planning to reassess this code when finalizing AMR decoder. However, I believe this would be a good time to add some unit testing to the project. This would make it much easier to ensure the functionality of code while developing. Currently I am short of time, for at least the next two weeks, but I will start over after that.

ptrkrysik, 2015-06-22:
Regarding the uplink - there is burst detection implemented in OsmoTRX.
It might be better than the method used currently in gr-gsm.

I have seen this message from Harald Welte suggesting that their BTS
support EDGE: "The BTS will now happily transmit EDGE/8PSK data."
(http://code.metager.de/source/history/osmocom/osmo-bts/src/common/oml.c)
I don't know if they implemented the physical layer with 8PSK modulation
and how advanced is their implementation in general.

I don't know much about UMTS currently - does it have common parts with
GSM like coding schemes, frames or protocols? If not we could start
something like gr-umts in parallel.

In the medium/long term it would be good to have GSM transmitting path
which could be taken from OsmoTRX or OpenBTS. This would make testing
easier - especially it would enable optimization of physical layer
performance with use of simulations.

Unit testing is needed indeed as well as documentation and a tutorial
for users - so they would know how to use the blocks to build their own
applications and not rely on us to build one application that will
support every situation (it might take some time before we will have
something like that).

romankh, 2015-06-22:
Yes, transmitting would be great also. 

Regarding UMTS: I don't have much knowledge as well, but I don't think that UMTS and GSM have much in common. Will keep this in mind and make a brief research. Anyway, it has not such a high priority for me, just an idea for the case, that there are no more features we can implement in gr-gsm, and we are at risk of suffering on boredom.

I also agree to the tutorial. Would be helpful for new users. I didn't think about one big application for every use case, not sure if it necessary at all. I believe the gnuradio-companion is great tool, everyone can easily create applications as needed. So I would rather focus the tutorial more on gnuradio-companion usage.

ptrkrysik, 2015-07-07:
I've looked a little bit into libosmocore - this are functions/structures common to different osmo projects. There are some things that would be very useful for gr-gsm (whole src/gsm directory) and some things that are specific to osmo projects and aren't applicable in gr-gsm. I find quite hard to read that code and search for what I need in it. It might be that I don't know some core concept that if I knew would simplify understanding how to use this implementation.

Also wireshark code has quite good example how to do interpretation of GSM messages. The functions can't be directly taken from there but it's good reference implementation which is relatively clean and readable.

ptrkrysik, 2015-07-07:
The example for wireshark:
https://github.com/wireshark/wireshark/blob/master/epan/dissectors/packet-gsm_a_rr.c#L1494

romankh, 2015-07-08:
libosmocore is definitely a good source for implementation, but I also agree with you that it is very focussed on their tasks. I ended up doing a lot of grep'ing to find the things I was looking for. 
That's why I preferred the OpenBTS code for the voice decoding.

ptrkrysik, 2015-08-15:
I'm closing the issue for now - if new ideas will emerge let's continue this matter on the mailing list.

closed 2015-08-15 by ptrkrysik

alphafox02, 2021-10-20:
I know this was closed and I also saw mention of it in another ticket, but was any umts support ever pursued?