ptrkrysik/gr-gsm #71: Using Volk machine: sse4_2_64 input.cfile: No such file or directory

aymeric75, 2015-06-22:
Hi everybody,

   I finnaly installed gr-gsm, but now i have an issue when executing the "airprobe_file.py", i type this command:
"airprobe_file.py --samp-rate=1M --fc=940M capture.cfileC

Here is the error log:

Using Volk machine: sse4_2_64
input.cfile: No such file or directory
Traceback (most recent call last):
  File "/usr/local/bin/airprobe_file.py", line 142, in <module>
    tb = airprobe_file(input_file_name=options.input_file_name, fc=options.fc, samp_rate=options.samp_rate)
  File "/usr/local/bin/airprobe_file.py", line 55, in __init__
    self.blocks_file_source_0 = blocks.file_source(gr.sizeof_gr_complex*1, input_file_name, False)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig0.py", line 436, in make
    return _blocks_swig0.file_source_make(*args, **kwargs)
RuntimeError: can't open file

p.s:    The capture.cileC is a gsm sample i got here https://svn.berlin.ccc.de/projects/airprobe/wiki/WorkingWithTheUSRP

Cheers

romankh, 2015-06-22:
I don't have access to my installation right now, but I think the command should be `airprobe_file.py --samp-rate=1M --fc=940M -i capture.cfile`, thats also what the README says. 

Also, you probably have to correct your samp-rate, as the stated capture has a decimation of 112, the samp-rate is 64e6/112 or similar, where 64e6 is depending on the device that was used for capture.

aymeric75, 2015-06-22:
Yes sorry, it was just a typo, so no error now but it seems that the execution is very long, is it normal ?

aymeric75, 2015-06-22:
I don't know anythin about gsm capturing, how should i correct the sample-rate and why ?
Thx

romankh, 2015-06-22:
You need to provide the gsm-receiver the same sample rate as it was used for recording. Some devices like USRP utilize decimation, therefore you have to take that information into consideration. 
Decimation in general is described at wikipedia: https://en.wikipedia.org/wiki/Decimation_%28signal_processing%29

Normally it shouldn't take long, it depends on the length of the capture. The mentioned file isn't that long. However, I am not sure if the airprobe_file.py you are using is set to run to completion or waits for input to end. 

I would strongly suggest you start with gnuradio-companion, and take a look at the grc files in the repository. Gnuradio-companion is a graphical editor, it allows the creation of flowgraphs, which then can be used to create a python app. grc files stores the flowgraph. This will help you to understand what is happening in the airprobe_file.py.

aymeric75, 2015-06-23:
Thank you for your help, now it's not loading for ever as yesterday but i still don't know what it has done... Maybe it wrote something in another file somewhere else on my computer?

ptrkrysik, 2015-07-02:
This is most probably not related to an error in gr-gsm but it is problem with getting it running. In case of updates in this story please continue the discussion on gr-gsm mailin list (https://github.com/ptrkrysik/gr-gsm/#mailing-list) - just add link pointing to this issue at the beginning. I'm closing the issue.

closed 2015-07-02 by ptrkrysik