ptrkrysik/gr-gsm #72: I can't get the paquets in wireshark

aymeric75, 2015-06-23:
Hi everyone,

 I installed gr-gsm, started wireshark with this command : wireshark -k -f udp -Y gsmtap -i lo  (apparentlty it's better to not use the sudo mode), and start this other command in another terminal: ./airprobe_rtlsdr.py   

The graphflows appears, no problem, but nothing in wireshark... I don't understand what is wrong

Thanks 

Aymeric

aymeric75, 2015-06-23:
ok, i just had to settle the frequency of few units...

romankh, 2015-06-23:
Is it working now ? If yes, consider closing the issues that were solved.

wesleygoatley, 2015-06-23:
I'm having similar issues - checked my ppm offset with kalibrate, and have what looks like activity in the freq/power graph, but it's just 0 being continuously printed to terminal and no packets in wireshark.  Any thoughts on this?

romankh, 2015-06-23:
These are probably "O"s, this indicates that rtl-sdr experiencing a buffer overrun.
Check your sample rate.

wesleygoatley, 2015-06-23:
Just did, changing the sample rate anywhere down to 1M/up to 3M makes no difference, what do you suggest?

romankh, 2015-06-24:
You will need the right sample rate, this is the sample rate that was used for recording. It depends on the device used for recording. If a decimation is provided with the recording, the sample rate = samp_rate_device / decimation.

ptrkrysik, 2015-07-02:
This is most probably not related to an error in gr-gsm but it is problem with getting it running. In case of updates in this story please continue the discussion on gr-gsm mailin list (https://github.com/ptrkrysik/gr-gsm/#mailing-list) - just add link pointing to this issue at the beginning. I'm closing the issue.

closed 2015-07-02 by ptrkrysik