ptrkrysik/gr-gsm #74: Where to find A5/2 encrypted data ?

aymeric75, 2015-06-24:
Hi everyone,

   I've made a program to crack A5/2 (i know it's not used anymore...) and i'm looking for a A5/2 encrypted sample of GSM datas. Do you have any idea where i can find it ?
Also i would like to know if it's possible to settle my phone so that it communicates using A5/2 algorithm to ecrypt data.

Thanks

Aymeric

romankh, 2015-06-24:
I haven't seen any captures with A5/2 encryption so far on the net. I you find any, would be nice if you drop me note. 

You normally can't set the encryption algorithm to use in the phone. The phone tells the network a list of encryptions it is capable of, and then the network tells the phone which one to use. 
However, you probably can modify OsmocomBB to do so. Still you would need a network operator that still supports A5/2. Maybe you can setup OpenBSC to do so.

Also, A5/2 was withdrawn in 2006, means no more phones are to be produced that support this encryption since the withdrawal. Here you find more detailed information http://security.osmocom.org/trac/wiki/A52_Withdrawal

If you just want to verify that your implementation works, you may want to take a look at http://www.npag.fr/project-a52hacktool . This is also an implementation of an attack on A5/2, and it supports encoding and decoding as well, so you could create some test data.

aymeric75, 2015-06-24:
Thanks, actually i do this for an internship in my university (even if it has no utility now). My goal is to create a communication using A5/2 between two cellphones and then to be abble to sniff datas and decrypt it. 
As you said, i need to dig more in OsmocomBB direction (i will have to buy compatible phones thought, see this link: http://bb.osmocom.org/trac/ )

ptrkrysik, 2015-07-02:
I'm closing the discussion now. In case there will be an update I advise to post it on the gr-gsm mailing list:
https://github.com/ptrkrysik/gr-gsm/#mailing-list

closed 2015-07-02 by ptrkrysik