ptrkrysik/gr-gsm #75: Possible to settle the RTL-SDR dongle as a BSC or a BTS ?

aymeric75, 2015-06-25:
I'm trying to recreate a mini GSM network using Osmocom Suite, the problem is that it require a BTS (see requirements part: http://openbsc.osmocom.org/trac/wiki/OpenBSC#WhatisOpenBSC ), but i can't find one yet, so i'm wondering if i can use my RTL-SDR as a BTS....

Cheers

Aymeric

NinjaComics, 2015-06-25:
@aymeric75 RTL-SDR dongle is receive only. So technically it's not possible to setup a BSC with RTL-SDR.

aymeric75, 2015-06-25:
@NinjaComics Thank you, do you know a cheap way to make a GSM network (between two phones directly or a phone and a BTS) ? 

From what i've read for now, one need a physical BTS but maybe there is a "simplier" way...

NinjaComics, 2015-06-25:
There is Calypso BTS (to make the calypso based baseband work as frontend for your BTS), if you prefer cheap GSM network. But that setup requires Calypso based phones, which are quite extinct to buy these days.

aymeric75, 2015-06-25:
Thank you, do you think this model is okay ?

http://www.ebay.fr/itm/British-Telecom-Calypso-120-Single-Line-Cordless-Phone-/291496032171?pt=LH_DefaultDomain_3&hash=item43de8477ab

NinjaComics, 2015-06-25:
@aymeric75 No. Please check this [osmocomBB wiki page](http://bb.osmocom.org/trac/wiki/Hardware/Phones) for the list of phones with Calypso baseband.

NinjaComics, 2015-06-25:
@ptrkrysik this issue can be closed

closed 2015-06-25 by ptrkrysik