ptrkrysik/gr-gsm #76: Request for a ML/Google group

NinjaComics, 2015-06-25:
@ptrkrysik Now that gr-gsm is gaining some momentum, can I suggest that we have a mailing list/google group where issues like #75 and problems concerning installation can be tackled on the list and have gr-gsm-development only discussions on Github ? 

P.S - I haven't contributed a single line of code to gr-gsm, but I am following it and have used it previously. That said, I am looking forward to contributing a bit from my end as well.

akibsayyed, 2015-06-25:
I can provide hosting for mailing list

ptrkrysik, 2015-06-25:
I don't have too much experience in setting up and administering mailing list. I agree that there is need for place for helping new users and general discussions so I decided to start google group, as it can be configured immediately. This group is:

gr-gsm@googlegroups.com

I've set up the list with most liberal settings I could set. To join the group with any e-mail addres (google account is not required) use this link:
https://groups.google.com/forum/#!forum/gr-gsm/join

ptrkrysik, 2015-06-26:
I invite everyone who has any questions regarding gr-gsm or gsm in general to join the mailing list and start discussions.

closed 2015-08-15 by ptrkrysik