ptrkrysik/gr-gsm #77: gain not working as it should

tdbone1, 2015-07-01:
running ubuntu 15.04 amd 64 desktop
E4000 tuner
airprobe_rtlsdr.py gains do not stick if you try and do it from command line or from the gui.
anything below 25 (in gui) if you press small arrow down it will change but go right back to auto setting
if you start at 0 and go up (small increments) real fast you can see gain moving signal correctly but over 25 it has no affect
if under 25 when you stop moving the button it will default to auto.
i looked at script but i not good enough to figure out the setting.
i think its a small setting as the buttons do work below 25 it just will not stick
ex: if i adjust from 30 to 0 in small increments i can signal go lower but when i stop adjusting it will default back to 30 even if i have the gain showing 0
thanks for your hard work

NinjaComics, 2015-07-02:
@tdbone1 Can you please guide us to reproduce the problem or provide with some trace ? With your description, it's hard to tell what might be causing the problem.

ptrkrysik, 2015-07-02:
I observed also strange behaviour of gain setting. This is more general
thing and has more to do with RTL-SDR receiver and GNU Radio RTL-SDR
source than with gr-gsm. One thing that might be wrong in
airprobe-rtlsdr is that gain mode is set to automatic.

tdbone1, 2015-07-03:
@NinjaComics 

did you even read my first post?
it tells you how to reproduce its step by step

ptrkrysik, 2015-07-06:
Hi @tdbone1,

You are right that gain slider don't make too much sense in current form - where it is overrided by automatic gain control.  However automatic gain control seems to be working so it migh be valuable to leave it as checkbox that is enabled by default.

ptrkrysik, 2015-08-15:
I decided that it won't be changed as we at this version as we will probably drop current grc based apps.

closed 2015-08-15 by ptrkrysik