ptrkrysik/gr-gsm #79: Python crash or missing module on OSX 10.10.4

dslokva, 2015-07-02:
Hi!!!

In my case (updated mac os 10.10, macports) - I can do cmake, make, make install with another warnings without errors. When I try to start the program - I got python crash.
```
MacBook-Pro:build dimas$ cmake ..
-- Build type not specified: defaulting to release.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: PMT
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
GNURADIO_PMT_FOUND = TRUE
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.5.
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   gnuradio-grgsm

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/dimas/gr-gsm/build
```

make output:

```
Scanning dependencies of target gnuradio-grgsm
[  2%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/receiver/receiver_impl.cc.o
[  4%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/receiver/clock_offset_control_impl.cc.o
/Users/dimas/gr-gsm/lib/receiver/clock_offset_control_impl.cc:106:16: warning: using integer absolute value function 'abs' when argument is of floating
      point type [-Wabsolute-value]
            if(abs(ppm) > 100) //safeguard against flawed measurements
               ^
/Users/dimas/gr-gsm/lib/receiver/clock_offset_control_impl.cc:106:16: note: use function 'std::abs' instead
            if(abs(ppm) > 100) //safeguard against flawed measurements
               ^~~
               std::abs
/Users/dimas/gr-gsm/lib/receiver/clock_offset_control_impl.cc:135:24: warning: using integer absolute value function 'abs' when argument is of floating
      point type [-Wabsolute-value]
                    if(abs(d_last_ppm_estimate-d_ppm_estimate) > 0.1)
                       ^
/Users/dimas/gr-gsm/lib/receiver/clock_offset_control_impl.cc:135:24: note: use function 'std::abs' instead
                    if(abs(d_last_ppm_estimate-d_ppm_estimate) > 0.1)
                       ^~~
                       std::abs
2 warnings generated.
[  6%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/receiver/cx_channel_hopper_impl.cc.o
[  9%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/misc_utils/bursts_printer_impl.cc.o
[ 11%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/misc_utils/extract_system_info_impl.cc.o
[ 13%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/demapping/universal_ctrl_chans_demapper_impl.cc.o
[ 16%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/demapping/tch_f_chans_demapper_impl.cc.o
[ 18%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/decoding/control_channels_decoder_impl.cc.o
[ 20%] Building CXX object lib/CMakeFiles/gnuradio-grgsm.dir/decoding/tch_f_decoder_impl.cc.o
In file included from /Users/dimas/gr-gsm/lib/decoding/tch_f_decoder_impl.cc:30:
/Users/dimas/gr-gsm/lib/decoding/tch_f_decoder_impl.h:76:53: warning: in-class initialization of non-static data member is a C++11 extension
[... 676 lines cut ...]
===========                      =======
CoreServices                       2516K
Dispatch continuations             16.0M
Kernel Alloc Once                     8K
MALLOC                            154.9M
MALLOC (admin)                       32K
STACK GUARD                        56.0M
Stack                              10.1M
VM_ALLOCATE                         348K
__DATA                             29.2M
__IMAGE                             528K
__LINKEDIT                        124.0M
__TEXT                            304.6M
__UNICODE                           552K
mapped file                        86.7M
shared memory                         4K
===========                      =======
TOTAL                             785.6M

```

All packages except PyQt4 was installed via macports. PyQt4 I installed from this instruction:
http://www.noktec.be/python/how-to-install-pyqt4-on-osx

Any ideas to make it working?

iZsh, 2015-08-08:
I had the same crash: you have the make sure cmake configures SWIG to use the exact same pythonlib than the one you are actually using (usually the version from /opt/local/bin if you are using macports)

By default cmake finds the system python:
-- Checking for module SWIG
-- Found SWIG version 3.0.7.
-- Found SWIG: /opt/local/bin/swig
-- Found PythonLibs: /usr/lib/libpython2.7.dylib (found suitable version "2.7.6", minimum required is "2")
-- Found PythonInterp: /opt/local/bin/python2.7 (found suitable version "2.7.10", minimum required is "2")

running cmake with the following commands:
% mkdir build && cd build
% cmake -DCMAKE_INSTALL_PREFIX:PATH=/opt/local -DPYTHON_LIBRARY=/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib -DPYTHON_INCLUDE_DIR=/opt/local/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7/ ..

gives:
-- Checking for module SWIG
-- Found SWIG version 3.0.7.
-- Found SWIG: /opt/local/bin/swig
-- Found PythonLibs: /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib (Required is at least version "2")
-- Found PythonInterp: /opt/local/bin/python (found suitable version "2.7.10", minimum required is "2")

however that was not enough. It fails while linking the swig stuff.

To solve this, I had to add the following code in swig/CMakeList.txt:
IF(${CMAKE_SYSTEM_NAME} MATCHES "Darwin")
  # Mac OS X specific code
  # Without this flag it does not find some of the libraries
  SET(CMAKE_MODULE_LINKER_FLAGS "-undefined dynamic_lookup")
ENDIF(${CMAKE_SYSTEM_NAME} MATCHES "Darwin")

once compiled and installed:
% which python
/opt/local/bin/python
% python
Python 2.7.10 (default, May 25 2015, 13:06:17)
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.56)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import grgsm
>>>

no more crash \o/

ptrkrysik, 2015-08-15:
I'm closing the issue. I can't provide support for Mac as I don't have any Apple computer and I can't test gr-gsm on MacOSX.

closed 2015-08-15 by ptrkrysik

dslokva, 2015-08-16:
Hi!

It works, thanks a lot!

mckinnley, 2016-12-30:
Hey guys! Just followed what happened here and see the update in my clones copy but and having the issue where when I `import grgsm` my MAC crashes (V 10.12.2). Any chance anyone can help me troubleshoot the issue? 

**Output of sudo make**
```Scanning dependencies of target grgsm
[  1%] Building CXX object lib/CMakeFiles/grgsm.dir/receiver/receiver_impl.cc.o
[  2%] Building CXX object lib/CMakeFiles/grgsm.dir/receiver/receiver_config.cc.o
[  4%] Building CXX object lib/CMakeFiles/grgsm.dir/receiver/viterbi_detector.cc.o
[  5%] Building C object lib/CMakeFiles/grgsm.dir/receiver/sch.c.o
[  7%] Building CXX object lib/CMakeFiles/grgsm.dir/receiver/clock_offset_control_impl.cc.o
[  8%] Building CXX object lib/CMakeFiles/grgsm.dir/receiver/cx_channel_hopper_impl.cc.o
[  9%] Building CXX object lib/CMakeFiles/grgsm.dir/demapping/universal_ctrl_chans_demapper_impl.cc.o
[ 11%] Building CXX object lib/CMakeFiles/grgsm.dir/demapping/tch_f_chans_demapper_impl.cc.o
[ 12%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/control_channels_decoder_impl.cc.o
[ 14%] Building C object lib/CMakeFiles/grgsm.dir/decoding/cch.c.o
[ 15%] Building C object lib/CMakeFiles/grgsm.dir/decoding/fire_crc.c.o
[ 16%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/tch_f_decoder_impl.cc.o
[ 18%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/AmrCoder.cpp.o
[ 19%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/BitVector.cpp.o
[ 21%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/GSM610Tables.cpp.o
[ 22%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/GSM660Tables.cpp.o
[ 23%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/GSM503Tables.cpp.o
[ 25%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/ViterbiR204.cpp.o
[ 26%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/burst_timeslot_splitter_impl.cc.o
[ 28%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/burst_sdcch_subslot_splitter_impl.cc.o
[ 29%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/burst_timeslot_filter_impl.cc.o
[ 30%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/burst_sdcch_subslot_filter_impl.cc.o
[ 32%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/burst_fnr_filter_impl.cc.o
[ 33%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/dummy_burst_filter_impl.cc.o
[ 35%] Building CXX object lib/CMakeFiles/grgsm.dir/flow_control/uplink_downlink_splitter_impl.cc.o
[ 36%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/controlled_rotator_cc_impl.cc.o
[ 38%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/controlled_fractional_resampler_cc_impl.cc.o
[ 39%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/msg_to_tag_impl.cc.o
[ 40%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/message_printer_impl.cc.o
[ 42%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/tmsi_dumper_impl.cc.o
[ 43%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/burst_file_sink_impl.cc.o
[ 45%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/burst_file_source_impl.cc.o
[ 46%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/message_file_sink_impl.cc.o
[ 47%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/message_file_source_impl.cc.o
[ 49%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/bursts_printer_impl.cc.o
[ 50%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/extract_system_info_impl.cc.o
[ 52%] Building CXX object lib/CMakeFiles/grgsm.dir/misc_utils/extract_immediate_assignment_impl.cc.o
[ 53%] Building CXX object lib/CMakeFiles/grgsm.dir/qa_utils/burst_sink_impl.cc.o
[ 54%] Building CXX object lib/CMakeFiles/grgsm.dir/qa_utils/burst_source_impl.cc.o
[ 56%] Building CXX object lib/CMakeFiles/grgsm.dir/qa_utils/message_source_impl.cc.o
[ 57%] Building CXX object lib/CMakeFiles/grgsm.dir/qa_utils/message_sink_impl.cc.o
[ 59%] Building CXX object lib/CMakeFiles/grgsm.dir/decryption/decryption_impl.cc.o
[ 60%] Linking CXX shared library libgrgsm.dylib
[ 60%] Built target grgsm
Scanning dependencies of target _grgsm_swig_doc_tag
[ 61%] Building CXX object swig/CMakeFiles/_grgsm_swig_doc_tag.dir/_grgsm_swig_doc_tag.cpp.o
[ 63%] Linking CXX executable _grgsm_swig_doc_tag
[ 63%] Built target _grgsm_swig_doc_tag
Scanning dependencies of target grgsm_swig_swig_doc
[ 64%] Generating doxygen xml for grgsm_swig_doc docs
warning: Tag `SYMBOL_CACHE_SIZE' at line 289 of file `/Users/elias/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag `SHOW_DIRECTORIES' at line 477 of file `/Users/elias/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag `HTML_ALIGN_MEMBERS' at line 803 of file `/Users/elias/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag `USE_INLINE_TREES' at line 956 of file `/Users/elias/gr-gsm/build/swig/grgsm_swig_doc_swig_docs/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
[... cut ...]
    0x7fffc86f4000 -     0x7fffc871dff7  libxpc.dylib (972.30.7) <65E41BB6-EBD5-3D93-B0BE-B190CEE4DD93> /usr/lib/system/libxpc.dylib

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 5018
    thread_create: 0
    thread_set_state: 0
```