ptrkrysik/gr-gsm #81: Install issues - Kali Linux

nomadcelt, 2015-07-09:
Good morning,

I am running in to all sorts of problems installing gr-gsm at this time in Kali.
On sudo ./pybombs -v install gr-gsm, I received the message that I had the wrong version of Automake. I've now installed 1.13 but on attempting again, while I get a lot further, I am now presented with:

checking for ranlib... (cached) ranlib
checking whether g++ supports C++11 features by default... no
checking whether g++ supports C++11 features with -std=c++11... yes
./configure: line 17375: syntax error near unexpected token `QT,'
./configure: line 17375: `    PKG_CHECK_MODULES(QT, QtCore >= 4.3, QtNetwork >= 4.3, have_qt=yes, have_qt=no)'
Configuration failed. Re-trying with higher verbosity.
make: *** No targets specified and no makefile found.  Stop.
Build failed. See output above for error messages.

I've checked QtCore and QtNetwork; I believe I am higher than the versions required but could be wrong.

Would you be able to help?

with thanks,

NC

romankh, 2015-07-10:
Hi, 

which version of automake was installed previously ? 
Can you install and try with automake >= 1.14 ?

nomadcelt, 2015-07-10:
Thanks for the advice. I was on 1.13. I upgraded to 1.14 but it still hangs on the same screen. I tried Automake 1.15 and nothing is different.

Any ideas?

romankh, 2015-07-11:
Hmm, did pybombs install the other packages that gr-gsm is depending on ? Like Gnuradio, rtl-sdr and so on ? You can verify this using `./pybombs list`. If Gnuradio was installed, please look whether it was installed from source or from deb packages.

nomadcelt, 2015-07-11:
According to ./pybombs list, not installed but accordingy to apt-get, it is.

romankh, 2015-07-11:
Which version is installed ?
gr-gsm relies on gnuradio >= 3.7
Maybe you have to uninstall your current version, and modify the pybombs config, such that at least gnuradio is built from sources rather installed from distro packages (option forcebuild) 
I would suggest this for most of the dependencies for gr-gsm. It takes more time, but it ensures that you use the recent versions that are used in developing.

nomadcelt, 2015-07-11:
I will:
Uninstall Gnu radio
Locate latest source and install
Reattempt pybombs install of gr-gsm.
Thanks for the advice, I'll report back shortly.

romankh, 2015-07-11:
you don't need to locate any of the sources, pybombs will do that for you, it will clone the corresponding repos and do the build and installation. 
You only have to uninstall the packages, and change your pybombs config, specifically the forcebuild option. By default, it is a list of packages, that will be built from sources, no matter if there are packages available. I believe you removed at least parts of the list, such that for example gnuradio is not built from source anymore, but installed via packages. This works fine and is faster, as long as your OS has the right versions in the repo. If Kali only provides a package for gnuradio in version 3.6, this won't work, and you have to keep gnuradio in the forcebuild list.

nomadcelt, 2015-07-11:
I could find no way of forcing gnuradio to the latest version. I am hit every time with the same errors.
I've now sourced the latest source code and am in the midst of compiling it. It is taking quite significant amounts of time but I hope to force the install thereafter. More to follow.

ptrkrysik, 2015-07-11:
W dniu 11.07.2015 o 23:34, nomadcelt pisze:
> I could find no way of forcing gnuradio to the latest version. I am hit
> every time with the same errors.
> I've now sourced the latest source code and am in the midst of compiling
> it. It is taking quite significant amounts of time but I hope to force
> the install thereafter. More to follow.
> 
Can you show contents of your pybombs/config.dat file?

-- 
Piotr Krysik

nomadcelt, 2015-07-11:
Sure, thank you:

[config]
gituser = root
gitcache =
gitoptions =
prefix = /usr/local
satisfy_order = deb,src
forcepkgs =
forcebuild = gnuradio,uhd,gr-air-modes,gr-osmosdr,gr-iqbal,gr-fcdproplus,uhd,rt$
timeout = 30
cmakebuildtype = RelWithDebInfo
builddocs = OFF
cc = gcc
cxx = g++
makewidth = 4

ptrkrysik, 2015-07-11:
W dniu 11.07.2015 o 23:58, nomadcelt pisze:
> Sure, thank you:
> 
> [config]
> gituser = root
> gitcache =
> gitoptions =
> prefix = /usr/local
> satisfy_order = deb,src
> forcepkgs =
> forcebuild = gnuradio,uhd,gr-air-modes,gr-osmosdr,gr-iqbal,gr-fcdproplus,uhd,rt$

Does this has this content or it was just cut at 80'th character when
you were copying it?

> timeout = 30
> cmakebuildtype = RelWithDebInfo
> builddocs = OFF
> cc = gcc
> cxx = g++
> makewidth = 4
> 

Can you send me address of iso of your distributin so I could test the
installation in virtual machine? (There are many people who want to use
gr-gsm on Kali and I would like to resolve issues with installation
under this distribution.)

-- 
Piotr Krysik

nomadcelt, 2015-07-11:
I believe it was cut but will check in the morning.

The kali distro is the latest 64-bit distro from the Kali site.

Thank you

Sent by Outlook for Android

> Sure, thank you:
>
> [config]
> gituser = root
> gitcache =
> gitoptions =
> prefix = /usr/local
> satisfy_order = deb,src
> forcepkgs =
> forcebuild = gnuradio,uhd,gr-air-modes,gr-osmosdr,gr-iqbal,gr-fcdproplus,uhd,rt$

Does this has this content or it was just cut at 80'th character when
you were copying it?

> timeout = 30
> cmakebuildtype = RelWithDebInfo
> builddocs = OFF
> cc = gcc
> cxx = g++
> makewidth = 4

Can you send me address of iso of your distributin so I could test the
installation in virtual machine? (There are many people who want to use
gr-gsm on Kali and I would like to resolve issues with installation
under this distribution.)

nomadcelt, 2015-07-12:
The full copy is:

[config]
gituser = root
gitcache =
gitoptions =
prefix = /usr/local
satisfy_order = deb,src
forcepkgs =
forcebuild = gnuradio,uhd,gr-air-modes,gr-osmosdr,gr-iqbal,gr-fcdproplus,uhd,rtl-sdr,osmo-sdr,hackrf,gqrx,bladerf,airspy
timeout = 30
cmakebuildtype = RelWithDebInfo
builddocs = OFF
cc = gcc
cxx = g++
makewidth = 4

nomadcelt, 2015-07-12:
The full output of: ./pybombs -v install gnuradio

sudo ./pybombs -v install gnuradio
Settled on prefix: /home/john
Initializing environmental variables...
Loading recipes ...
Checking if gnuradio is installed: No
/home/john/
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of make (3.81) is >= than 3.75
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of libgsl0-dev (1.15) is >= than 1.13
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of libgsl0ldbl (1.15) is >= than 1.13
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of python-cheetah (2.4.4) is >= than 2.0
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of python-numpy (1.6.2) is >= than 1.5
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of python-lxml (3.4.0) is >= than 2.3.2
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of python-gtk2 (2.24.0) is >= than 2.17
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of python-cairo-dev (1.8.8) is >= than 1.8
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of cmake (2.8.9) is >= than 2.8.3
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of cmake-data (2.8.9) is >= than 2.8.3
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of python-qt4 (4.9.3) is >= than 4.6.2
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of python-qwt5-qt4 (5.2.1) is >= than 5.2
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of bison (2.5) is >= than 2.3
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of flex (2.5.35) is >= than 2.5.35
Installing packages:
* apache-thrift
* gnuradio
Installing from source: apache-thrift

    ./bootstrap.sh && ./configure --prefix=/home/john/  \
    --enable-tests --disable-tutorial --with-cpp --with-python \
    --with-c_glib --with-java --without-csharp --without-php \
    --without-ruby --without-lua --without-nodejs --without-erlang \
    --without-go --without-haskell --with-libevent --without-zlib \
    CC=gcc CXX=g++ PY_PREFIX=/home/john/

configure.ac:89: installing './config.guess'
configure.ac:89: installing './config.sub'
configure.ac:26: installing './install-sh'
configure.ac:26: installing './missing'
compiler/cpp/Makefile.am:102: warning: source file 'src/thrifty.yy' is in a subdirectory,
compiler/cpp/Makefile.am:102: but option 'subdir-objects' is disabled
automake: warning: possible forward-incompatibility.
automake: At least a source file is in a subdirectory, but the 'subdir-objects'
automake: automake option hasn't been enabled.  For now, the corresponding output
automake: object file(s) will be placed in the top-level directory.  However,
automake: this behaviour will change in future Automake versions: they will
automake: unconditionally cause object files to be placed in the same subdirectory
automake: of the corresponding sources.
automake: You are advised to start using 'subdir-objects' option throughout your
automake: project, to avoid future incompatibilities.
compiler/cpp/Makefile.am:102: warning: source file 'src/thriftl.ll' is in a subdirectory,
compiler/cpp/Makefile.am:102: but option 'subdir-objects' is disabled
compiler/cpp/Makefile.am:34: warning: source file 'src/main.cc' is in a subdirectory,
compiler/cpp/Makefile.am:34: but option 'subdir-objects' is disabled
compiler/cpp/Makefile.am:34: warning: source file 'src/md5.c' is in a subdirectory,
compiler/cpp/Makefile.am:34: but option 'subdir-objects' is disabled
compiler/cpp/Makefile.am:34: warning: source file 'src/generate/t_generator.cc' is in a subdirectory,
compiler/cpp/Makefile.am:34: but option 'subdir-objects' is disabled
compiler/cpp/Makefile.am:34: warning: source file 'src/parse/t_typedef.cc' is in a subdirectory,
compiler/cpp/Makefile.am:34: but option 'subdir-objects' is disabled
compiler/cpp/Makefile.am:34: warning: source file 'src/parse/parse.cc' is in a subdirectory,
compiler/cpp/Makefile.am:34: but option 'subdir-objects' is disabled
compiler/cpp/Makefile.am:34: warning: source file 'src/generate/t_c_glib_generator.cc' is in a subdirectory,
compiler/cpp/Makefile.am:34: but option 'subdir-objects' is disabled
compiler/cpp/Makefile.am:34: warning: source file 'src/generate/t_cpp_generator.cc' is in a subdirectory,
compiler/cpp/Makefile.am:34: but option 'subdir-objects' is disabled
compiler/cpp/Makefile.am:34: warning: source file 'src/generate/t_java_generator.cc' is in a subdirectory,
compiler/cpp/Makefile.am:34: but option 'subdir-objects' is disabled
compiler/cpp/Makefile.am:34: warning: source file 'src/generate/t_json_generator.cc' is in a subdirectory,
compiler/cpp/Makefile.am:34: but option 'subdir-objects' is disabled
compiler/cpp/Makefile.am:34: warning: source file 'src/generate/t_as3_generator.cc' is in a subdirectory,
[... 389 lines cut ...]
                 airprobe                    None                      
                   fecapi                    None                      
                      kal                    None                      
                  libbtbb                    None                      
              libosmo-dsp                    None                      
              libosmocore        fetch  inventory                      
[... cut ...]
                 osmo-sdr    installed  inventory effcaa75bfd43262f621184c28be2935149965b9 git://git://git.osmocom.org/osmo-sdr.git
                  rtl-sdr    installed  inventory e3c03f738f5aef4dc51e2b741fbdb542b9cc1bb1 git://git://git.osmocom.org/rtl-sdr
                      uhd    installed  inventory d9656de88f7af77b39bfe9985f0ac7c623932d71 git://https://github.com/EttusResearch/uhd.git
Category: pseudo
                      all                    None 

I have attempted to install libosmocore throught pybombs but it doesn't complete.

ptrkrysik, 2015-07-12:
W dniu 12.07.2015 o 11:08, nomadcelt pisze:
> The full output of: ./pybombs -v install gnuradio
>
> sudo ./pybombs -v install gnuradio
> Settled on prefix: /home/john
> Initializing environmental variables...
> Loading recipes ...
> Checking if gnuradio [...]

Hi,

GNU Radio developers decided to include Control Port by default in
installation through pybombs. In my opinion this decision was done too
early. There are many problems with installation of apache-thrift that
is included because it is required by control port.

As control port is not required by gr-gsm the easy solution is to remove
apache-thrift from list of dependencies. I've attached a pybombs recipe
that has it removed. Copy it to pybombs/recipes and run installation again.

-- 
Best Regards,
Piotr Krysik

#
# This file is part of PyBOMBS
#
# PyBOMBS is free software; you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation; either version 3, or (at your option)
# any later version.
#
# PyBOMBS is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with PyBOMBS; see the file COPYING.  If not, write to
# the Free Software Foundation, Inc., 51 Franklin Street,
# Boston, MA 02110-1301, USA.
#

depends: make boost fftw cppunit swig gsl uhd alsa git python cheetah wxpython numpy lxml pygtk pycairo cmake pyqt4 pyqwt5 gcc
category: common
satisfy_deb: gnuradio-dev
source: git://http://www.gnuradio.org/git/gnuradio.git
gitbranch: master
gitargs: --recursive
var config_opt = " -DENABLE_GR_FCD=ON -DENABLE_DOXYGEN=$builddocs -DENABLE_GR_LOG=OFF -DENABLE_PERFORMANCE_COUNTERS=ON -DCMAKE_CXX_FLAGS=-fpermissive "
inherit: cmake
configure_static {
    cmake .. -DCMAKE_BUILD_TYPE=$cmakebuildtype -DCMAKE_INSTALL_PREFIX=$prefix -DENABLE_STATIC_LIBS=True $config_opt
}

install {
    make install
    make install
}

nomadcelt, 2015-07-12:
replace gnuradio.lwr with this content?

ptrkrysik, 2015-07-12:
W dniu 12.07.2015 o 13:05, nomadcelt pisze:
> replace gnuradio.lwr with this content?
> 
Yes,

To make things simpler (also for me to understand what is happening) do
everything from the beginning:

git clone https://github.com/pybombs/pybombs.git

(copy gnuradio.lwr into pybombs/recipes/)

then configure pybombs by changing only install directory - the rest
leave as default, and finally:

cd pybombs
sudo ./pybombs install gr-gsm

-- 
Best Regards,
Piotr Krysik

nomadcelt, 2015-07-12:
After all that, ./pybombs list nor reports GNUradio Make Inventory.

ptrkrysik, 2015-07-12:
W dniu 12.07.2015 o 16:07, nomadcelt pisze:
> After all that, ./pybombs list nor reports GNUradio Make Inventory.
> 
Nomadcelt,

I don't understand what this means exactly. Can you explain it more
clearly what is the result?

nomadcelt, 2015-07-12:
I've added more detail to that message now. 

Nomadcelt,

I don't understand what this means exactly. Can you explain it more
clearly what is the result?

nomadcelt, 2015-07-12:
Piotr,

There were pages and pages of "installing", with no errors. The final lines were:

-- Installing: /home/john/include/gnuradio/swig/wxgui_swig.i
-- Installing: /home/john/include/gnuradio/swig/wxgui_swig_doc.i
make: *** No rule to make target `make'. Stop.

NC

nomadcelt, 2015-07-12:
I have conducted a : sudo ./pybombs install gnuradio

It installed with no errors, apparently.

.pybombs list reports "Make inventory" next to gnuradio.

Do I need to installed libosmocore? That won't install at this stage either.

nomadcelt, 2015-07-12:
Wow....well, everything appears to be installing fine (Apart from libosmocore).

I now can't find ~/.gnuradio/config.conf to change the entry.

ptrkrysik, 2015-07-13:
libosmocore isn't gr-gsm's dependency (or dependency of any other
program that gr-gsm depends on). It shouldn't be installed at all when
you install gr-gsm.

If you compiled gnuradio from source you don't need to edit that file -
gr-gsm should work without this change.

nomadcelt, 2015-07-13:
Piotr,
Thank you for all your help. When I get home tonight, I shall look at it again. I am trying to execute airprobe_rtlsdr.py but it can't call gnuradio. I assume, therefore, that I must amend sonething else. I'll post error messages later.
Regarding executing gnuradio, airprobe, etc, must I do that from inside the usr/local/ folder or can they be called from elsewhere?

ptrkrysik, 2015-07-13:
I didn't try to install gnuradio and gr-gsm in other folders than
/usr/local. It can be done for sure but I don't know what difficulties
it might create.

nomadcelt, 2015-07-13:
Piotr,

when I try to run ./airprobe_rtlsdr.py, the following errors occur:

Traceback (most recent call last):
  File "./airprobe_rtlsdr.py", line 10, in <module>
    from gnuradio import blocks
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/blocks/__init__.py", line 32, in <module>
    from blocks_swig import *
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig.py", line 22, in <module>
    from blocks_swig0 import *
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig0.py", line 26, in <module>
    _blocks_swig0 = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig0.py", line 22, in swig_import_helper
    _mod = imp.load_module('_blocks_swig0', fp, pathname, description)
ImportError: libgnuradio-blocks-3.7.8git.so.0.0.0: cannot open shared object file: No such file or directory

nomadcelt, 2015-07-13:
ok, getting somewhere, I think.

Trying to install gr-gsm now but am stumbling over libosmo-dsp installation:

./configure: line 3754: syntax error near unexpected token `pic-only'
./configure: line 3754: `LT_INIT(pic-only)'

I've been in to configure.ac and it looks as fine as I think it should regarding formatting / parameters.

Any ideas?

ptrkrysik, 2015-07-14:
W dniu 13.07.2015 o 23:22, nomadcelt pisze:
> ok, getting somewhere, I think.
>
> Trying to install gr-gsm now but am stumbling over libosmo-dsp installation:
>
> ./configure: line 3754: syntax error near unexpected token |pic-only'
> ./configure: line [...]

Hi,

I tried to install gr-gsm on Kali linux.

Kali linux has (cerrect me if I'm wrong) very small repository of it's
own packages and pybombs tries to compile a LOT of packages. That
considerably increases possibility that eventually something will go
wrong. In my case compilation fails at the very beginning. I don't have
solution for this yet and currently Kali Linux should be considered
unsupported.

Also I don't have much time to solve this problem - if you can research
this and post solution I would be very grateful.

-- 
Best Regards,
Piotr Krysik

nomadcelt, 2015-07-14:
Piotr,
I'll gladly force develop a solution for this and will let you know.
Thanks,
John

Hi,

I tried to install gr-gsm on Kali linux.

Kali linux has (cerrect me if I'm wrong) very small repository of it's
own packages and pybombs tries to compile a LOT of packages. That
considerably increases possibility that eventually something will go
wrong. In my case compilation fails at the very beginning. I don't have
solution for this yet and currently Kali Linux should be considered
unsupported.

Also I don't have much time to solve this problem - if you can research
this and post solution I would be very grateful.

closed 2015-08-06 by ptrkrysik

blackdog07, 2015-11-21:
So I just tried to install gr-gsm on Kali 2.0. Ran into problems and googled and found this. So I got it to work I believe by doing this as user root.

apt-get install cmake libboost-all-dev gnuradio-dev libvolk-dev libcppunit-dev libosmocore doxygen swig
apt-get install build-essential libtool shtool autoconf automake git-core pkg-config make gcc liblog4cpp5-dev libpcsclite-dev

git clone git://git.osmocom.org/libosmocore.git
cd libosmocore/
autoreconf -i
./configure
make
sudo make install
sudo ldconfig -i
cd ..

git clone https://github.com/ptrkrysik/gr-gsm.git
cd gr-gsm
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

cd ~
mkdir .gnuradio
echo "[grc]
local_blocks_path=/usr/local/share/gnuradio/grc/blocks
" > ~/.gnuradio/config.conf

I have only tested it by running gnuradio-companion airprobe_rtlsdr,grc and I am able to capture gsm traffic. Your mileage may vary. Hope this is helpful for someone!

blackdog07, 2015-11-21:
Follow-up
Ok I was able to get gr-gsm to work but I then tried to install Kalibrate-hackrf and it failed to build
I searched the Internet and found a blog that said that the guy installed gr-gsm with pybombs and kalibrate-hackrf would build ok but if not it wouldn't build. So I just did a pybombs install of gr-gsm on top of the install described above and I could then build kalibrate-hackrf.
The problem I ran into after the pybombs install was that gqrx would no longer run....arg...still not too clean.

ptrkrysik, 2015-11-22:
Thank you for sharing the steps needed to run gr-gsm on Kali Linux.
The libosmocore is first installed from distribution's repository and
then built manually.

Did you try to compile gr-gsm without manual building of libosmocore but
only installing with apt-get?

blackdog07, 2015-11-23:
I first installed libosmocore via apt-get and tried to compile gr-gsm and had errors. I then did apt-get remove libosomocore and then built from source. That's what lead me to building it from source. Sorry I wasn't too clear on that.

tweekspeak, 2016-01-17:
Hi, maybe u can help me out. I have a marquee and  installed Linux deploy an vcn, but during install it said it skips mnt partitions, /proc/sys/dev/tty, an more. Dns fail, mtab fail, ssh fail vcn fail. Went into vcn viewer an got "port on which tje compter is listening could mot be contacted. Do you think y could help me along?

ptrkrysik, 2016-01-18:
@tweekspeak , I don't think this is the right address to ask this question