ptrkrysik/gr-gsm #84: Extractor for Immediate Assignment info

romankh, 2015-07-12:
Hi, 

some time ago I found that a block for extracting the assigned information from immediate assignments is necessary for me, so I started coding that block. I think it might be useful for other users too, see branch ia-collector in my repo. 

There is an option to enable printing the infos to output.
I implemented the getter functions like they are in the system info extractor. 
Just wondering if there is a convenient way to have just one function that returns a vector with the data class. I did some research but didn't find such a method.

I am using a map for holding the data, as I believe I will implement some option that handles duplicate assignments for the same request reference. 

If no objections, I will create a pull request for that.

romankh, 2015-07-13:
I added two options now:
* ignore GPRS Immediate assignments: those are not always from interest, specially as we have no gprs support yet
* unique request references: ignores duplicate assignments for the same request reference, as most of the time it is not necessary to have the same information twice (or more)

ptrkrysik, 2015-07-13:
Hi @romankh,

There is no easy way to pass c++ map to python through swig (without
writing magic swig config). Passing c++ vectors is straightforward,
however. So what I used is rewriting map to a vector and returning a vector.

Look here:
https://github.com/ptrkrysik/gr-gsm/blob/master/lib/misc_utils/extract_system_info_impl.cc#L300

Is this what you need?

romankh, 2015-07-13:
Yes, thank you, just wanted to be sure that I haven't overseen something.
 In the end, I copied your approach in immediate assignment extractor, so we have some functions returning a vector with the corresponding values.

romankh, 2015-07-19:
Closing this issue as the feature is implemented, and merged in commit 8b60acab97231916d786ae482b892e3425f80ad0

closed 2015-07-19 by romankh