ptrkrysik/gr-gsm #85: Add A5/3 decryption

romankh, 2015-07-13:
Hi all,

I need to capture some AMR encoded speech data, but the providers I could use have all A5/3 in place, and I don't have a phone that isn't support A5/3 yet.

Therefore I would like to propose the implementation of A5/3 decryption into the current decrypter block. The Kc can be retrieved from some phones, and then be used to decrypt the captured data.

I think it can be done with moderate effort, as there is an extension/reimplementation of A5 including version 3 in the liba53 library used in OpenBTS, see https://github.com/RangeNetworks/liba53

Basically I would add an dropdown to the block to choose the right algorithm, and replace the current a5 source by liba53.

ptrkrysik, 2015-07-13:
Hi @romankh,

This is great idea - especially if you have some test data for it A5/3.

Your proposition to use implementation of Osmo team for all ciphers is
also very good. Current implementation of A5/1 in gr-gsm was taken from
very old demonstration code that probably isn't too fast nor has good
interface to switch between A5/1 and A5/2.

Generally whole liba53 looks like a good library as it has clearly
defined purpose. If it can be easily installed - maybe let's add recipe
for it in pybombs without adding the code directly to gr-gsm. What you
think about this?

romankh, 2015-07-13:
Hi @ptrkrysik,

I think test data isn't a problem. I can do some captures, and my phone supports Kc retrieval. 

Regarding the recipe: you are right, would be good approach. 
If I understood it right, we would only have to contribute the recipe to pybomb's recipe repo ?
Besides the changes in the decrypter, of course.

ptrkrysik, 2015-07-13:
W dniu 13.07.2015 o 09:55, romankh pisze:
> Hi @ptrkrysik <https://github.com/ptrkrysik>,
>
> I think test data isn't a problem. I can do some captures, and my phone
> supports Kc retrieval.
>
> Regarding the recipe: you are right, would be good [...]

Yes, first we can fork pybombs, add the recipe for liba53, change gr-gsm
recipe to include liba53 dependency and make pull request to pybombs.

We will change installation procedure so it point to our fork. When
liba53 recipe and changed gr-gsm recipe will be included in pybombs we
will change the installation procedure back.

romankh, 2015-07-13:
Ok, then I will go for it that way. 
Can you create the pybombs fork, think it's better if it is together with gr-gsm

ptrkrysik, 2015-07-13:
Done.

romankh, 2015-07-13:
Sorry, but I believe we need the recipes - repo
Anyway, I created a recipe that should do the job:

```
#
# This file is part of PyBOMBS
#
# PyBOMBS is free software; you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation; either version 3, or (at your option)
# any later version.
#
# PyBOMBS is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with PyBOMBS; see the file COPYING.  If not, write to
# the Free Software Foundation, Inc., 51 Franklin Street,
# Boston, MA 02110-1301, USA.
#

category: baseline
source: git://https://github.com/RangeNetworks/liba53.git
gitbranch: master
inherit: empty

install {
    make install
    ldconfig
}

uninstall {
    rm -f /usr/lib/liba53.so.1.0 /usr/lib/liba53.so /usr/lib/liba53.so.1 /usr/include/a53.h
    ldconfig
}

verify {
    make installtest
}

```

I am doing the uninstall manually because the make file has no uninstall target.
So, I think you can just paste the code into liba53.lwr and try it out.

What also came to my mind: maybe we should wait a few days after the pull request to the origin repository. I can develop locally, maybe the gr team merges the recipe in that time. So you could save time changing instructions, and we avoid confusing users.

ptrkrysik, 2015-07-13:
Ah, last time I forked pybombs recipes and the rest were in one repository.

ptrkrysik, 2015-07-13:
> I am doing the uninstall manually because the make file has no uninstall
> target.
> So, I think you can just paste the code into liba53.lwr and try it out.
>
> What also came to my mind: maybe we should [...]

Current recipe doesn't install the files in install dir configured in
pybombs. There is DESTDIR variable in the makefile but it isn't the same
as install prefix. This might be obstacle in getting the recipe to
pybombs mainline.

The idea avoid confusing users is good. Till the recipe gets accepted we
can use forked pybombs and you can develop decryption in a separate
(your own) branch.

romankh, 2015-07-13:
Ok, I see. 
For some reason, they also only add the header file for the A5/3 implementation in the make file. 
Maybe I should forget the make file, and handle everything in the recipe.

The library itself looks good, think we will have the decryption for the other versions soon

romankh, 2015-07-14:
I found out that libosmocore already contains the a5 functions we are looking for. 
Maybe it's better to add libosmocore to gr-gsm's dependencies, as it contains many other useful code ?

I tried doing that, however I have problems in understanding what it takes to have the library included. I installed libosmocore using pybombs, and added it to gr-gsm's recipe as dependency. Do I have to edit CMakeList.txt, or should pybombs take care about the include paths for libosmocore ?
In my case, while libosmocore's header file are installed to pybombs target path, the build fails as osmocom/gsm/a5.h cannot be found.

ptrkrysik, 2015-07-16:
If gr-gsm's doesn't find the include file the you'll have to edit CMakeList.txt

romankh, 2015-07-17:
Yes, that's what I did, but I am not getting the libraries linked. 
I created a FindLibosmocore.cmake, added a find_package() and added it to target_link_libraries, but it's not getting linked, I end up with an unresolved symbol. 

I will commit it later into a branch in my fork, maybe you could take a look ?

romankh, 2015-07-17:
Just for update: 
libosmocore is C, name mangling hit me. 
It's working now.

ptrkrysik, 2015-07-17:
Ok,

I've added libosmocore to the list of gr-gsm's dependencies.

romankh, 2015-07-17:
@ptrkrysik, I pushed the current implementation to my repo, see above. 

If you have time and you like, you could do a review and test, specially the libosmocore installation / integration. 

I ended up adding all A5/[1-4], as there wasn't much to do for it. 
A5/1 is working, tested with one of your test captures. 
I will next try to capture some A5/3 encrypted SMS with my phone, and will add some unit testing for the decryption block. 
A5/2 and A5/4 will remain an untested feature, until we get some test data. I just thought they are a cheap nice-to-have feature.

ptrkrysik, 2015-07-18:
I've tested new A5/1 decryption and it work on my captures. I've made
pull request to pybombs recipes. When they accept it I will merge your
new decryption blocks with the main branch.

romankh, 2015-07-18:
Great news :)
I captured a SMS with A5/3, and it also works well. 
Will post a file with the corresponding bursts here for testing, the capture file contains some other subscriber's data with A5/1, and I don't want to expose their communication to the internet. 

Will also do the unit testing code soon

ptrkrysik, 2015-07-18:
What phone are you using during your captures of A5/3? I'm using Nokia
3310 to obtain test data as it can be set to camp on a selected channel.
It is not capable of doing A5/3 encryption however.

romankh, 2015-07-18:
I am using a Samsung S3. 
It has no such option to select a channel, but it shows the channel it is camping on with an USSD code `*#0011#`. Also it allows retrieving the Kc using a serial connection and an at-command `at+crsm=176,28448,0,0,8`
I have two pre-paid SIM, on for a operator with a cell that implements A5/1 only, and another one where both A5/1 and A5/3 are supported, the network then always chooses A5/3.

This should work with all phones that are supported by https://github.com/2b-as/xgoldmon

romankh, 2015-07-19:
You can find the demapped bursts here: 
http://www.file-upload.net/download-10777199/sms-a53.bursts.html

Just feed to a decryption block and then to control channel decoder.

Kc A5/3:             0x41,0xBC,0x19,0x30,0xB6,0x31,0x8A,0xC8

However, I found my burst source throwing an exception about the thread joining itself. 
Does this occur for you also ? I believe the change of position of the thread->join() is the reason
Maybe the burst source needs some changes.

ptrkrysik, 2015-07-20:
Mainlinee pybombs now includes libosmocore on gr-gsm's dependency list
so it is possible to add your new decryption block to main branch of gr-gsm.

romankh, 2015-07-20:
Ok, I will finish the QA test and create the pull request later today

ptrkrysik, 2015-07-20:
There is some error in pybombs that was reported by one of the users (I
can also observe it myself) - so we will have to wait a little bit till
they fix it.

romankh, 2015-07-20:
Ok, no problem, I am not done yet with that testing. 
It's more work than I thought, I ended up creating some new burst source and burst sink for testing

Burst source takes lists of frame numbers, timeslots, burst strings (from burst printer), and creates the corresponding pmt, and the sink takes the pmt's as input, and we get three vectors back, which can be compared. This blocks can be later reused for testing other blocks that take bursts as input.

romankh, 2015-07-21:
I am done with testing, have created the pull request already

romankh, 2015-07-23:
Closing issue, enhancement merged in 590d96c8874faae2ef7babd09fd5f081c5e5da8c

closed 2015-07-23 by romankh

Keskebeu, 2018-02-21:
Sorry guys to jump back into this so a5/1 a5/2 a5/3 and a5/4 are part of gr-gsm now? Meaning decrypting on a5/3 algorithm is possible? I have liba53 and also libosmocore.

ptrkrysik, 2018-02-21:
It should be possible as osmocom's A5/3 implementation is used here.
However I don't remember using it yet.

Keskebeu, 2018-02-21:
@ptrkrysik 

Thank you I started reading cmake files and yes I believe it is. I just have to try it. I appreciate your fast response.

arifkyi, 2019-03-15:
> You can find the demapped bursts here:
> http://www.file-upload.net/download-10777199/sms-a53.bursts.html
>
> Just feed to a decryption block and then to control channel decoder.
>
> Kc A5/3: [...]

Hello,  do you have the cfile to test the decryption code with that your  KC A5/3: 0x41,0xBC,0x19,0x30,0xB6,0x31,0x8A,0xC8 ?   my purpose just want to test grgsm_Decode with A5/3 ?