ptrkrysik/gr-gsm #89: Burst source thread join

romankh, 2015-07-19:
There is a bug with the join() of d_thread. 
Currently the burst source throws an exception, and therefore the flow graph stops. 
The cause is imho that 
```
        d_thread->interrupt();
        d_thread->join();
``` 
are currently executed in the child thread, and d_thread also references the child thread. 

I moved the statements back to the stop function, however then again the flow graph didn't terminate.

Finally I found the intended method to archive a proper termination in this post https://lists.gnu.org/archive/html/discuss-gnuradio/2015-05/msg00287.html

So I added `post(pmt::mp("system"), pmt::cons(pmt::mp("done"), pmt::from_long(1)));` to the end of the run() function. 

Pull request follows.

ptrkrysik, 2015-07-20:
I have messed something with multithreading when I tried to make this
block exit when all input bursts are processed. Merging your change.

romankh, 2015-07-20:
Closing this issue, should be fixed with the above merge.

closed 2015-07-20 by romankh