ptrkrysik/gr-gsm #91: error " sudo ./pybombs config "

lamledinh, 2015-07-20:
I can't config pybombs. I don't know why. Can you help me, please !

lamledinh@lamledinh-PC:~/pybombs$ sudo ./pybombs config
[sudo] password for lamledinh: 
Initializing config file...
Username for GIT access
gituser [root]:
Directory of git cache repository
gitcache []:
Options to always pass to git clones (--depth=1 or --single-branch will speed up clones)
gitoptions []:
Install Prefix
Traceback (most recent call last):
  File "./pybombs", line 25, in <module>
    from mod_pybombs import verbosity as v
  File "/home/lamledinh/pybombs/mod_pybombs/__init__.py", line 22, in <module>
    from globals import *;
  File "/home/lamledinh/pybombs/mod_pybombs/globals.py", line 48, in <module>
    config_init(config);
  File "/home/lamledinh/pybombs/mod_pybombs/cfg.py", line 70, in config_init
    if os.path.basename(pwd)=="pybombs":
  File "/usr/lib/python2.7/posixpath.py", line 121, in basename
    i = p.rfind('/') + 1
AttributeError: 'NoneType' object has no attribute 'rfind'

romankh, 2015-07-20:
@lamledinh, 

while gr-gsm uses pybombs as installation option, pybombs is not part of this project, and this issue tracker is intended for issues regarding gr-gsm only. 

But you can write to the project's mailinglist gr-gsm@googlegroups.com (see Readme).
Please also provide more information, like your OS and how to reproduce the error (like in this case: was pybombs configured before, or is this the initial config ? What did you enter as Install Prefix ?)

ptrkrysik, 2015-07-20:
I can also observe this problem - something is messed in current pybombs. Solution for now is to checkout older revision:
cd pybombs
git checkout b0f9b6f89f7a74b5baec46b061b23c34ce7fde82

I've filled the bug report on pybombs github page and I'm closing the issue here.

closed 2015-07-20 by ptrkrysik