ptrkrysik/gr-gsm #94: uplink-decoding

gtr5, 2015-07-21:
Gr-gsm project has now many useful functions for monitoring GSM signals. I think the next major step would be to have it able to decode uplink channels. I saw some people saying that they were able to implement such ability with airprobe but nothing was made available to the public. I don't think this would be an easy task but with some effort we should be able to realise it.

ptrkrysik, 2015-07-21:
W dniu 21.07.2015 o 20:38, gtr5 pisze:
> Gr-gsm project has now many useful functions for monitoring GSM signals.
> I think the next major step would be to have it able to decode uplink
> channels. I saw some people saying that they were able [...]
Hi,

Reception of uplink is similar to reception of hopping channels - in
both cases bursts detection and ability to keep synchronization are very
helpful. Gr-gsm's receiver currently supports hopping channels, although
this support is not finished as we are not able yet to decode whole calls.

I was able to decode some of uplink messages by doing very simple hack
in current code for receiving hopping channels. To get started with
decoding of something from uplink will be relatively easy. What might be
harder is finding weak points (in bursts detection, demodulation,
decoding implementation) and improve them in order to have stable stream
of decoded messages in the output.

-- 
Best Regards,
Piotr Krysik

gtr5, 2015-07-22:
1) You mean that you are not able to decode the uplink part of the call or that even the downlink decoding is still missing parts.

2) Is decoding of uplink for the beacon channel ( non-hopping ) possible with the current tools ?

ptrkrysik, 2015-07-22:
W dniu 22.07.2015 o 09:26, gtr5 pisze:
> 1) You mean that you are not able to decode the uplink part of the call
> or that even the downlink decoding is still missing parts.
> 
> 2) Is decoding of uplink for the beacon channel ( non-hopping ) possible
> with the current tools ?
> 

Support for uplink can be relatively simply added to current gr-gsm. It
is a matter of:
-adding some form of configuration to grc/receiver/gsm_receiver.xml - my
proposition is to add a switch if user wants to process uplink streams
also or not, if yes then additional inputs with names like up0,up1, ...,
up${num_streams} would appear,
-adding parameter (true if user wants to process uplink) to the
receiver's parameters,
-processing additional input streams using detection of bursts (as in
normal_or_noise here:
https://github.com/ptrkrysik/gr-gsm/blob/master/lib/receiver/receiver_impl.cc#L303),
-assigning uplink bursts numbers that are delayed by 3 timeslots in
relation to current frame/timeslots counter. This is the main difference
in relation to normal_or_noise processing, it can be simply done by
checking if stream is uplink stream - if yes call something like:
	send_burst(d_burst_nr-3, output_binary, GSMTAP_BURST_NORMAL, input_nr);
instead of
	send_burst(d_burst_nr, output_binary, GSMTAP_BURST_NORMAL,
input_nr);
}
-adding support for uplink in demapping blocks (uplink and downlink
bursts should be processed separately).

And that's probably all to get started with uplink decoding.
Then what will be required will be improving things here and there in
order to increase efficiency of the reception/decoding.

I can implement this myself but it will take me some time before I can
work on this issue, as I don't have much of free time currently. If
there is someone willing to take this task - I can give my advices in
order to complete it.

-- 
Best Regards,
Piotr Krysik

ptrkrysik, 2015-08-03:
Hi all,

I'm assigning myself to this issue as I already made some moves toward implementing uplink reception and decoding.

ptrkrysik, 2015-08-08:
I've made step forward in uplink implementation. You can check it in uplink-decoding branch in which I've added a receiver block that receives both uplink and downlink. I have also changed control channels demappers in order to support uplink decoding. The code was tested on a capture where phone was located close (few meters) to the receiver.

gtr5, 2015-08-11:
What SDR are you using for capturing both downlink and uplink ? Also can uplink be decoded without synchronization with the downlink signal.

ptrkrysik, 2015-08-11:
W dniu 11.08.2015 o 10:36, gtr5 pisze:
> What SDR are you using for capturing both downlink and uplink ? Also can
> uplink be decoded without synchronization with the downlink signal.
> 
Hi,

I'm using two USRPs N2x0 or one USRP X310 in order to have two separate
synchronized channels. I'm aware that it might be quite expensive setup
for normal user but we have to start from something and I've chosen
hardware that won't create any additional problems.

We might in the future implement two channel coherent receiver
consisting of two RTL-SDR devices or try to work without synchronization
with downlink. But this will be considered in the next step - not the first.

Currently uplink support for control channels decoding seems to be
working. It was tested for a phone that was few meters from the
receiver, so we need to have more test data for a phone located in
larger distance and then probably some issues will come out.

-- 
Best Regards,
Piotr Krysik

bluelightvision, 2015-08-12:
Hi,

Concerning the SDR to be used, the UmTRX could probably be a suitable choice ( didn't test it yet ) since it has two channels, its cheaper than the usrp and its designed to be used for GSM.

ptrkrysik, 2015-08-12:
W dniu 12.08.2015 o 18:20, bluelightvision pisze:
> Hi,
> 
> Concerning the SDR to be used, the UmTRX could probably be a suitable
> choice ( didn't test it yet ) since it has two channels, its cheaper
> than the usrp and its designed to be used for GSM.
Hi,

Probably any device that has two independent synchronous RX channels
should be good for uplink reception.

UmTRX also has advantage of having sampling rate equal to 26MHz - which
is integer multiply of of GSM symbol rate. The result is that you need
only decimation in order to obtain GSM signal with proper sampling rate.

-- 
Best Regards,
Piotr Krysik

bluelightvision, 2015-08-12:
Concerning the decoding of uplink control channels, shouldn't you get data flowing just by monitoring the correct frequency as in downlink control channels. Does higher distances from the target phone have other effects than resulting in a weaker signal?

ptrkrysik, 2015-08-12:
W dniu 12.08.2015 o 18:58, bluelightvision pisze:
> Concerning the decoding of uplink control channels, shouldn't you get
> data flowing just by monitoring the correct frequency as in downlink
> control channels. Does higher distances from the target phone have other
> effects than resulting in a weaker signal?
> 

Weaker signal and higher inter-symbol interferences due to multipath
propagation.

In my test data I get some unencrypted traffic on uplink in SDCCH
channel. I didn't make successful test where data from other phones was
decoded. I didn't make too many tests yet however.

-- 
Best Regards,
Piotr Krysik

sparks57, 2015-08-13:
>  try to work without synchronization with downlink  But this will be considered in the next step - not the first.

I am also interested in catching some of the uplink packets with just a single RTLSDR but I understand that doesn't work yet.  I did try to build it anyway and got TypeError: Required argument 'uplink_starts_fn_mod51' (pos 4) not found when trying to run the airprobe_rtlsdr.py file.

I also have a bladerf; would that work with this code somehow?

ptrkrysik, 2015-08-14:
W dniu 14.08.2015 o 00:22, sparks57 pisze:
> try to work without synchronization with downlink But this will be
>     considered in the next step - not the first.
>
> I am also interested in catching some of the uplink packets with just a
> single [...]

I didn't check if new code doesn't brake current apps. I'm using
modified flowgraph where I have UHD source receiving two channels and
then version of GSM receiver that processes uplink.

Regarding bladerf - I've looked into it's specification and it has
single with bandwidth up to 40MHz. Acording to this it won't be possible
to do this with single device.

-- 
Best Regards,
Piotr Krysik

sdrUser, 2015-08-28:
I have a question about uplink decoding.

I have a single channel radio only. I am recording the uplink channel side in a building where I am the only phone on the network.  After that, I manually go through the data and extract the groups of 4 normal "uplink" bursts and try to run them through a small program that calls your downlink decoder functions using simple CPP code from a command line environment.

I have successfully done the same operation on a group of 4 manually extracted downlink bursts and seen complete paging messages. Then tried to move over to the uplink and do the same. So... I know that my test setup is correct, except for the part of not understanding why the decoder did not work on the uplink data. Or... perhaps 4 consecutive uplink bursts cannot be trusted to be a single message?

Is there a difference for uplink and downlink decode, and if so, what is the difference between the decode of the uplink and down link once you have the 4 bursts? 

I have the data and graphs that show our data if you need my examples to better explain?

ptrkrysik, 2015-08-29:
W dniu 28.08.2015 o 23:26, sdrUser pisze:
> I have a question about uplink decoding.
>
> I have a single channel radio only. I am recording the uplink channel
> side in a building where I am the only phone on the network. After that,
> I manually go [...]

Hi,

When you say "4 consecutive bursts" you mean "bursts in the same
timeslot in 4 consecutive frames" (probably yes if decoding worked for
downlink but I want to be sure)? If you see only them inside of multiple
frame periods - then most probably they are all part of the same message.

Regarding the differences - the main difference from downlink is that
frames are delayed by three timeslots periods in uplink. But this is
probably not what is affecting you. Another difference that I can think
of is much lower signal power (and maybe more limited ability to improve
channel impulse response by changing position of rx antenna). I suppose
that you have good SNR if you see the bursts on some plot.

There is no difference in coding on downlink and uplink.

What is most probably affecting you is that there is much lower
percentage of unencrypted traffic on uplink. Usually there is some on
AGCH channel and at the beginning of communication on SDCCH channel.
Your 4 bursts might be encrypted and this is why you can't decode them.

Generally I'm interested in any joint efforts to create free software
implementation(s) of GSM uplink receptions.

-- 
Best Regards,
Piotr Krysik

sdrUser, 2015-09-01:
Thank you sir for the information. You did answer my main question of... Are the bursts different when it comes to the decoding? They are the same.

To answer your questions. Yes, these are all consecutive frames/bursts on the same timeslot. I have also taken into account the 3 timeslot periods for the uplink, since there is only one phone/sets of bursts showing up in our graphs. As for unencrypted traffic, I can see the RACH burst in my data and am looking directly at the next burst(s) for the Paging Response (CM Service Request). This message should not be encrypted. 
What I did to get it working.... Since I am in a controlled environment and the MS signal is right on top of the antenna, the signal was so strong that it was not able to decode the messages. By turning the gain of the radio down to almost nothing, the signal was very clean and it worked. To much signal in this case was a bad thing.
Your decoder functions are working fine for the uplink decode as long as there is only a single phone and you can start the decode in the correct spot. Now it is time to get the downlink timing working so it automates finding the correct spot to start the decode and pick a MS out from all of the other handsets. Once I get the alignment, I will come back to you and if it will help, send you what I have for timing alignment. Other than that, you have it all done it seems. 

Thank you very much for the help.

   

Hi,

When you say "4 consecutive bursts" you mean "bursts in the same
timeslot in 4 consecutive frames" (probably yes if decoding worked for
downlink but I want to be sure)? If you see only them inside of multiple
frame periods - then most probably they are all part of the same message.

Regarding the differences - the main difference from downlink is that
frames are delayed by three timeslots periods in uplink. But this is
probably not what is affecting you. Another difference that I can think
of is much lower signal power (and maybe more limited ability to improve
channel impulse response by changing position of rx antenna). I suppose
that you have good SNR if you see the bursts on some plot.

There is no difference in coding on downlink and uplink.

What is most probably affecting you is that there is much lower
percentage of unencrypted traffic on uplink. Usually there is some on
AGCH channel and at the beginning of communication on SDCCH channel.
Your 4 bursts might be encrypted and this is why you can't decode them.

Generally I'm interested in any joint efforts to create free software
implementation(s) of GSM uplink receptions.

ptrkrysik, 2015-09-01:
For example how to get the alignment - look at this paper:
http://boris.unibe.ch/40204/1/islam2.pdf

ptrkrysik, 2015-09-01:
The discussion doesn't concern issues/bugs in gr-gsm so if you itend to
continue it (I hope you do) please write to the mailing list:
https://github.com/ptrkrysik/gr-gsm#mailing-list

You can add short introduction with pointing to this thread and that the
discussion is a continuation of it.

gtr5, 2015-12-15:
I tested the uplink decoder and it works fine!
I think you should include it in the master branch.

ptrkrysik, 2015-12-18:
@gtr, it's great to hear that you tested the uplink code and it works
for you! Can you tell something more about the tests that you performed
and what were the results?

gtr5, 2015-12-19:
Hi, 
The hardware that I used is the UmTRX, I've done two different tests:
- First I tried to capture and decode the downlink/uplink signals from OpenBTS which was running on another SDR. (I didn't try to record a call yet, hopefully I'll try it soon and provide some feedback).
- The second test was on a real BTS (I tried several BTS where some of them were a few hundred meters away).
 I was able to obtain the "service request", "location updating request", "classmark change" and "paging response" from my MS in the first case and from other MS (maybe for people in the street) in the second case. I was able to obtain uplink signals every few seconds which seems good.

I must also note that in some cases I do not receive any uplink signal, I'm still not sure if this is due to the antennas used or that the "uplink receiver" still requires some enhancement.

palachzzz, 2015-12-25:
Hi,
I tryed to decode uplink and I had good result when it SDCCH. 
But when I tryed to decode voice, I saw double uplink frames, and no downlink frames, so speech.gsm not playable (I think there duplicated frames). 
[uplink_call.zip](https://github.com/ptrkrysik/gr-gsm/files/72184/uplink_call.zip)
For tests I used OpenBTS on UmTRX, and another UmTRX for gr-gsm, my flowgraph:
[image]

ptrkrysik, 2015-12-26:
@gtr5, thank you for the effort to test the uplink. I will merge the
uplink-decoding to master branch with the change proposed by you. It
will require some time in order to test how existing application work
after this change.

khawatkom, 2016-05-02:
Hello,

I have received USRP B200mini (wide bandwidth that should support 45MHz)
Do you have a grc file to test the uplink receiver using one device?
Where can I find the py or grc files to test the uplink receiver using two USRPs?

Looking for your reply.
Regards,

ptrkrysik, 2016-06-09:
Uplink works currently. It can be of course improved - but it works.

closed 2016-06-09 by ptrkrysik

Ho3einAmini, 2016-09-21:
Hi,
I have two USRP(N210 & B210) that one for receive down-link and other for up-link.
I try to receive up-link by using flowgraph that same flowgraph of palachzzz. when I run, I only receive down-link. Can you upload grc or python file of your work?

engrmasood2002, 2017-07-14:
Hi Piotr
I want to receive uplink mobile messages but was not able to find your uplink-decoding  branch you mentioned earlier in this post. can you provide me the link to download it and instructions on using it.
Thanks

ptrkrysik, 2017-07-24:
Uplink-decoding branch was removed as it was merged with master branch.
The uplink decoding is implemented in the main branch now.

I provided some example here how it can be used:
http://ptrkrysik.github.io/

2014siavash, 2018-04-30:
Dear ptrkrysik,

According to the following link, there are different training sequences for different frequencies in order to get the uplink bursts. I want to know if this fact reflects in your codes and blocks?

https://patents.google.com/patent/US7068631

More precisely, our main problem is in getting uplink bursts with frequency hopping. In fact, I cannot decode any uplink packet with frequency hopping while downlink packets with hopping can be decoded and seen in the wireshark, correctly. Can you guess what is the problem? (I can provide any information you want)

Thanks

velichkov, 2018-04-30:
Hi @2014siavash,

> According to the following link, there are different training sequences for different frequencies in order to get the uplink bursts. 
> https://patents.google.com/patent/US7068631

I don't know whether this patent is actually used in GSM or not. Could you find a reference about this in the 3GPP specifications?

>  I want to know if this fact reflects in your codes and blocks?

If the training sequences are not explicitly specified the `GSM Receiver` block tries to detect them by trying each one and choose the best.

https://github.com/ptrkrysik/gr-gsm/blob/2471052cfe724699c8bd2d4fec20d7f194cf6bb9/lib/receiver/receiver_impl.cc#L450-L473

but if you know the training sequences you could also specify them, right click on the `GSM Receiver` block and in `TSEQ numbers` add `[2, 3, 4]` for example.

> More precisely, our main problem is in getting uplink bursts with frequency hopping.

Are you able to decode the uplink bursts when there is no frequency hopping?

2014siavash, 2018-04-30:
Thanks for your answer.

About your request for the reference in 3gpp, I could not find it soon, but I will provide if there is something! When we encounter to the problem of getting gsm uplink signals in hopping mode, we guess it as the reason, absolutely not sure!

However, as I can see in your code, there is only one final training sequence for a hopping channel and so ,if each frequency has different training sequence, then your code maybe get not true. But I am not sure and I have to more study to be sure on it.

We are trying to get uplink packets with frequency hopping. To do this, we use four synchronized (coherent) rtlsdr2832 connected together and also use the block of multi-rtlsdr for running them with gnuradio, simultaneously. In this senario, we had no problem to get uplink bursts (and so packets) for non-hopping situation.

But in hopping situation, there is a problem for CX output. More precisely, the bursts which come from CX output of your block are not the true bursts and we cannot see the training sequence in the middle of them. However, the output for C0 has no problem.

One thing we guessed was for ppm parameter of these rtlsdrs and so, we try to arrange them in order to get the correct results. It seems the ppm parameter can effect on the output bursts and some times we could get some bursts with some shifts(!), but finally understand that they were not true bursts. Hence, the problem was not solved. It is interesting that we cannot also get Downlink Bursts in hopping mode! Therefore, we thought that maybe the problem is raised from different Tseq for different frequencies, but we are not sure! We are working on it, but till now, we cannot get the true result.

Can you help us to solve this important issue? Do you have any sample for getting uplink signals in hopping mode? Do you work with very low cost rtlsdr2832 to get these signals and know the issues with these SDRs (because one guess is the problem raise from random delays of these SDRs)?

We dont know how these 4 SDRs together works! We only used an opamp to amplify the signal of one sdr oscillator (master) and removed the oscillator of the other three (slaves) and connect them to the master rtlsdr2832. The interesting thing is it works(!) and SDRs are successfully known by the driver in computer. Also we test them and sure that they are worked correctly. But in the case of hopping, we dont know how they arranged to work. because we connect all these 4 SDRs with a hob to the computer. So they cannot send their sample together and gnuradio consider some (perhaps random) delays for each of them to send their data. In this situation, we dont know that if they are sample the signals from air, simultaneously or not (there is some time shift?)?! And we dont know that if these delays may made some effects on the results or not?!

Can you explain for us how these 4 connected and coherent SDRs (which all are connected to one hob) work? Any information can be provided if needed.

Thanks again for your attention.

tom-2015, 2019-09-16:
I see in GNU radio there is an uplink/downlink splitter block. Can I use this to capture the uplink signal with the USRP B200 (when capturing a wide band of 50MHz) or is there a way to capture uplink with the B210?

I can't find much documentation or example so I don't know how to connect everything can anyone help with this?

Thanks!

velichkov, 2019-09-16:
Hi @tom-2015,

> I see in GNU radio there is an uplink/downlink splitter block. Can I use this to capture the uplink signal with the USRP B200

This block is used to split downlink from uplink bursts after they have been decoded and it's not used when capturing.

> (when capturing a wide band of 50MHz) or is there a way to capture uplink with the B210?
> 
> I can't find much documentation or example so I don't know how to connect everything can anyone help with this?

I don't have USRP B2xx board and don't have experience with them but in theory it it should be possible to capture both uplink and downlink using USRP B200, you need to capture with big enough sampling rate (50 MS/s should be enough for non hopping channels, try with 46 MS/s as well). You will get one big file (cfile) so you need a fast enough hard drive as you should not get "O" characters in the output while capturing ("O" indicates buffer overflow) and make sure the device is connected on a USB3 port.

Then you need to channelize the output cfile to get 2 cfiles - the center frequency of the first needs to be the downlink frequency while the center frequency of the second needs to be the uplink frequency. For channelization you could use `grgsm_channelize` script but you need to modify it a bit as right now the arfcn list specifies only downlink frequencies. 

https://github.com/ptrkrysik/gr-gsm/blob/fa184a9447a90aefde2ca0dea1347b702551015d/apps/helpers/grgsm_channelize#L92-L93

And then you could use uplink_decoding_multi_rtl.grc flowgraph from the [examples repository](https://github.com/ptrkrysik/examples/tree/master/uplink_decoding_demo) as a start and again you need to modify it to suite your needs and match how the BTS is configured.

P.S.
Please continue the discussion in our [mailing list](https://github.com/ptrkrysik/gr-gsm/#mailing-list)