ptrkrysik/gr-gsm #100: flow control

romankh, 2015-08-03:
As discussed in the mailing list (see https://groups.google.com/forum/#!topic/gr-gsm/LLjv9GKjsf8), we will add some blocks for controlling the burst flow (and later message flow):

* New GRC category "Flow control"
* Add block "Timeslot splitter"
* Add block "Sub-slot splitter" with configuration for SDCCH/4 and SDCCH/8
* Add block "Framenumber filter" with configuration for "less or equal", "greater or equal"
* Add block "Dummy burst filter"

romankh, 2015-08-06:
I will add two more blocks, as I found that timeslot splitter and sub-slot splitter have a drawback in some use cases. For example, I want to create a flow graph of capable extracting a given SDCCH subslot, where I can set params timeslot, sub-slot and SDCCH config. For that two filter blocks for timeslot and sub-slot with one port in and one port out would be more efficient:

ptrkrysik, 2015-08-06:
As I'm working currently on uplink support I will add later a filter block that passes downlink and uplink bursts based on user selection.

ptrkrysik, 2016-02-28:
I'm closing the issue as @romankh created the blocks and the tests of these blocks.

closed 2016-02-28 by ptrkrysik