ptrkrysik/gr-gsm #104: Packaging gr-gsm

ptrkrysik, 2015-08-06:
Hi all,

We are looking for someone who has experience in creating deb and rpm packages (or someone who wants to learn how to do this) who would like to modify gr-gsm's cmake based buildsystem in order to create packages for Debian/Ubuntu and Fedora.

ashmastaflash, 2016-01-24:
I'm open to helping out with this.  What are your contribution guidelines?

ashmastaflash, 2016-01-24:
What's more important to you, Debian, Ubuntu, or Fedora?

ptrkrysik, 2016-01-24:
Hi Ash, I've replied on the gr-gsm's mailing-list.

DanielMasonNZ, 2016-02-17:
A Debian package would be great :+1:

bdheeman, 2017-03-25:
I usually package each and every project/software before I install and, or test it on/for ArchLinux/ArchLinux-ARM Linux.

Please provide guide lines where in this should we/I add the ArchLinux packaging, I would love to share the PKGBUILD script for gr-gsm :-)

ptrkrysik, 2017-03-30:
W dniu 25.03.2017 o 19:09, Balwinder S Dheeman pisze:
> I usually package each and every project/software before I install and,
> or test it on/for ArchLinux/ArchLinux-ARM Linux.
> 
> Please provide guide lines where in this should we/I add the ArchLinux
> packaging, I would love to share the PKGBUILD script for gr-gsm :-)
> 
What information do you need to prepare the script?

Currently I know something about Debian packaging (probably I forgot a
lot after few months of not looking into this topic), but I don't know
what is needed for ArchLinux.

petterreinholdtsen, 2017-08-29:
I'm working on packaging gr-gsm for Debian under the debian-hams umbrella.  You can track the status on https://bugs.debian.org/871055 .

petterreinholdtsen, 2017-09-02:
The gr-gsm package entered Debian today, see https://tracker.debian.org/pkg/gr-gsm for its status.  Time to close this issue?

closed 2018-06-03 by ptrkrysik