ptrkrysik/gr-gsm #105: Channel hopping improvements

rpp0, 2015-08-09:
As mentioned in https://github.com/ptrkrysik/gr-gsm/issues/51 the following improvements can be made in context of channel hopping: 
- Implementing channelization. GNU Radio's channelization block is not ideal for GSM in terms of performance, so a new block must be created.
- Adding documentation / example app.
- Adding a tutorial.
- Automatic detection of which channels to hop over.

I have self-assigned this enhancement. The first three features shouldn't take too much time to complete. The last one is a bit harder, but the new filtering blocks could be used as a starting point to extract the hopping ARFCN list.

alex0500, 2016-03-06:
any one cam share cfile capture for hopping testing with tchf traffic?

danyhm, 2016-08-14:
Hi 
So I'm new to both git hub and grgsm but i have some programming skills :)
i read the #51 issue and my operator is also using channel hopping.

If I'm not wrong there's a main C0 ARFCN that has a lot of traffic and is the broadcast channel and assigns MS's to other channels for the purposes like SMS and voice. Also i noticed that the bandwidth of all ARFCNs related to a single operator doesn't exceed 3 - 5 Mhz in my case.

now first of all I didn't find a compete tutorial on how to decode encrypted text with hopping channels. Could someone please explain this a little bit more? also how should i capture a wideband cfile ? the grgsm_capture tool only takes the frequency parameter and nothing about the bandwidth.

I can provide cfile captures for hopping channels if anyone's interested.

doraexploradora2014, 2016-08-15:
Hi,
I cannot capture/decode a second arfcn where is hopping the arfcn 0(Sys Info Type 1.List ARFCN and the Inmediate assign indicate this). I'm capture through wideband and after use channelize, but nothing get when decoding.  Also I try capture only that ARFCN 2 but nothing

Attach image like I see signal. I think that problem is why burst irregular but i dont know how catching those.

(Sorry my horrible english :o )

Dora
[image]

ptrkrysik, 2016-08-16:
Guys, I will try to answer your requests by writing a channel hopping
tutorial. I don't know yet when I will do that though - for sure not
before Saturday ;).

doraexploradora2014, 2016-08-17:
Hi, thanks for your reply.
I add something, I create the following GRC.

[image]

I take a capture of a hopping ARFCN with next command:
grgsm_capture.py 892.2e6 -f -c -s 1e6 movi_a243_sr1M --band GSM850 -T 60

When I run the GRC, I do not get anything on the console output. Is this right ?. 

Thanks in advanced.

ptrkrysik, 2016-08-25:
W dniu 17.08.2016 o 17:19, doraexploradora2014 pisze:
> Hi, thanks for your reply.
> I add something, I create the following GRC.
> 
> hopping_channel
> <https://cloud.githubusercontent.com/assets/21040507/17741759/ceb03ebe-6473-11e6-86fc-74326cfda49a.png>
> 
> I take a capture of a hopping ARFCN with next command:
> grgsm_capture.py 892.2e6 -f -c -s 1e6 movi_a243_sr1M --band GSM850 -T 60
> 
> When I run the GRC, I do not get anything on the console output. Is this
> right ?.
> 
> Thanks in advanced.

You are running the command incorrectly. After -f there should be
central frequency and after -c name of the output capture file. You
don't need to specify band if you are not specifying a channel instead
of carrier frequency.

-- 
Best Regards,
Piotr Krysik

doraexploradora2014, 2016-08-29:
uh sorry, I mistake when I post the question.

I wrote flowgraph with hopping, but I dont have Lapdm, only BCCH and the follow:

	55.346144400	127.0.0.1	127.0.0.1	GSMTAP	81	(CCCH) (GTTP) 

I config param of hopping block with the inmmediate assigment.
Thanks!
Dora

lunaalvarez, 2017-04-03:
Hi, 
Can anyone share a simple hopping capture file sample? I can't correctly decode my capture data with the steps shown in these issues, so i was wondering maybe there is an issue with my capture file and want to verify my steps. 
Thanks! 
Gozde

dvolvox, 2018-01-07:
HI @ptrkrysik , sorry to bother you - any updates on that tutorial? Having trouble decoding the channels.

Cheers!

closed 2026-11-09 by ptrkrysik