ptrkrysik/gr-gsm #107: Noise at the beginning of file produced by full rate decoder

ptrkrysik, 2015-08-10:
There are some noises at the beginning of files produced by full rate decoder of TCF/F decoder block.

romankh, 2015-08-11:
I implemented an option for a boundary check for voice traffic, addressing issue #107:

if this option is enabled (which it is by default), the decoder 
* ignores any TCH bursts prior receiving either a "Connect" or "Connect Acknowledge" message
* ignores any TCH bursts after receiving a "Release" message

This eliminates almost all of the noise that is caused by those bursts. 
In some cases there is still a little bit on the beginning and the end of speech file, however after a brief search I haven't found a hint on how to detect the real boundaries of speech transmissions. 

If anyone has an idea on how this can be detected more precisely, drop me a note, otherwise I think we can close this issue for now.

I also fixed the missing gsm-fr header for the audio frames, which caused `toast -d speechfile.au.gsm` to fail, so this is working now too. However toast seems to expect the file extension `.au.gsm` to properly decode the speech file, but as using toast isn't necessary at all, I'll not enforce file extensions in the decoder for now.

ptrkrysik, 2015-08-12:
Not having to use toast to convert the file is definitely an improvement :)

closed 2015-08-12 by ptrkrysik