ptrkrysik/gr-gsm #109: Build with pybombs - error - can you help?

Test5467, 2015-08-13:
Built target pygen_swig_d325e
>>> Error: Block key "gsm_clock_offset_control" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_input" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_message_printer" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_receiver" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_universal_ctrl_chans_demapper" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_clock_offset_control" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
[... 75 lines cut ...]
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_universal_ctrl_chans_demapper" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_universal_ctrl_chans_demapper" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_universal_ctrl_chans_demapper" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_universal_ctrl_chans_demapper" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/home/mac/target/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
Validation failed:

Block - blocks_file_source_0 - File Source(blocks_file_source):
	Source - out(0):

ptrkrysik, 2015-08-13:
Hi, issues on github are currently used for problems that are better defined. Please ask the same question on the list. Issue will be filled only if we figure out that something is really wrong with gr-gsm. Also - please provide more details, for example:
-what system/distribution do you use,
-what version of gnuradio do you use and how you installed it

ptrkrysik, 2015-08-13:
mailing-list: https://github.com/ptrkrysik/gr-gsm/#mailing-list

closed 2015-08-13 by ptrkrysik