ptrkrysik/gr-gsm #111: Unable to complete installation

Lethaha, 2015-08-15:
I followed the following steps in installing gr-gsm on my HP Probook laptop:

sudo apt-get install git
git clone https://github.com/pybombs/pybombs.git
cd pybombs
./pybombs config
sudo ./pybombs install gr-gsm

I'm running ubuntu 14.10 and everything installs perfectly until throwing the error below:

[100%] Built target pygen_swig_9c20d
>>> Error: Block key "gsm_clock_offset_control" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/usr/local/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/usr/local/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/usr/local/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/usr/local/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/usr/local/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/usr/local/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/usr/local/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/usr/local/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/usr/local/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_decryption" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/usr/local/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_decryption" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/usr/local/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_decryption" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/usr/local/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
[... 100 lines cut ...]
ImportError: No module named grc.python.Platform
>>> Error: Block key "gsm_universal_ctrl_chans_demapper" not found in Platform - grc(GNU Radio Companion)
Traceback (most recent call last):
  File "/usr/local/bin/grcc", line 30, in <module>
    from grc.python.Platform import Platform
ImportError: No module named grc.python.Platform
Validation failed:

Block - blocks_file_source_0 - File Source(blocks_file_source):
	Source - out(0):
		Port is not connected.
Error during file compilation.
apps/CMakeFiles/airprobe_file.py.dir/build.make:52: recipe for target 'airprobe_file.py' failed
make[2]: *** [airprobe_file.py] Error 1
CMakeFiles/Makefile2:662: recipe for target 'apps/CMakeFiles/airprobe_file.py.dir/all' failed
make[1]: *** [apps/CMakeFiles/airprobe_file.py.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.009.git-329-g5c9fbf9d

[100%] Built target airprobe_rtlsdr.py
Linking CXX shared module _grgsm_swig.so
[100%] Built target _grgsm_swig
Makefile:123: recipe for target 'all' failed
make: *** [all] Error 2
Build failed. See output above for error messages.

ptrkrysik, 2015-08-15:
I think that I've corrected this issue. Please pull new version and check now if you can.

ijarraki, 2015-08-16:
Yes, build errors have been resolved but issue #110 persists. Last working commit is 57711f6694d1f477342cc8fc46fdd64017bf11b7

ptrkrysik, 2015-08-16:
If it persist - how it can be distinguished from error #111? I thought it is one and the same problem with building of gr-gsm failing during compilation of grc files. 

There might be different problem though - I observe segmentation faults of grcc that happen occasionally during installation.

ijarraki, 2015-08-16:
I no longer get fatal errors during "make", but like with issue #110 airprobe_rtlsdr.py and airprobe_file.py are not placed in PATH by "make install".

ptrkrysik, 2015-08-17:
This is different thing than what was described in #110. I didn't notice
this - thank you for the report.

ijarraki, 2015-08-17:
Confirmed, commit 20d2de40b95fcc846a989022396e1851cc9264b3 has resolved this issue.

closed 2015-08-18 by ptrkrysik