ptrkrysik/gr-gsm #112: demapping/decoder blocks should not mutate the GSMTAP's header but copy it

iZsh, 2015-08-15:
currently the demapping blocks (e.g. the universal ctrl chans demapper or the tch one) change the GSMTAP's header in place:

gsmtap_hdr * header = (gsmtap_hdr *)pmt::blob_data(header_plus_burst);
[...]
header->sub_type = ch_type;

the problem is that if we connect the GSM receiver block to the socket UDP block to broadcast the initial GSMTAP_TYPE_UM_BURST packet, we get a race condition where the message being UDP-sent (and therefore processed by a third party application, but the same would happen if we connect the GSM receiver's output to multiple other blocks) can have invalid values (e.g. the sub_type will have a type only valid for a GSMTAP_TYPE_UM packet).

While I understand the necessary evil of changing the sub_type in the demapper without changing the burst type as an intermediate stage, the demapper should generate a new message instead of mutating it in-place to prevent from the race condition described above.

NB: the decoders seem to be doing the same, it modifies the header *before* copying the data to a new message.

romankh, 2015-08-16:
First of all, please use the mailing list https://groups.google.com/forum/#!forum/gr-gsm for general questions and discussions, as we try to use the issue tracker for development and bug fixing, e.g. issues that will have code changes as outcome. 

I will answer your question here for now, if you have more questions about it, please use the mailing list.

The demapper blocks are intended to be used in combination with the corresponding decoders. They take the bursts incoming from some source and group them together as the decoder needs them for further processing. The reason why the decoder generates a new message is that the decoder takes some bursts (4 or 8), decodes them and packs them in one decoded gsm message. 

If you just want to send the unprocessed bursts via udp and you want to have only bursts for a specific timeslot you may want to use some blocks from flow control, like the timeslot filter.

iZsh, 2015-08-16:
unless i misunderstand what you just said, I don't think I was clear about what the bug is. I'm not asking how GSM or gr-gsm is working. I'm reporting a bug: a race condition.

Currently the demapper is NOT generating a new message, and that's the problem: it is modifying its input message in-place. You are getting a pointer from it, and then modifying the INPUT message using that pointer instead of copying it before making any modifications to it.

For instance, let's say you have something like that

receiver -------> demapper ----> [some further processing/blocks]
|
|__________---> udp

for the sake of the example, let's say we are receiving a burst from SACCH/4 channel.
what will happen is:
1 - the receiver generates a GSMTAP_TYPE_UM_BURST with sub-type GSMTAP_BURST_NORMAL (= 6)
2 - the demapper will modify (NB: modify, not create a new message) this gsmtap message and change it's sub-type to GSMTAP_CHANNEL_SDCCH4 | GSMTAP_CHANNEL_ACCH (= 135)

so two things can happen:
- the message generated by (1) is sent through UDP before (2) has the time to process it (and modifies its header). We therefore have a UDP gsmtap packet with type GSMTAP_TYPE_UM_BURST and subtype GSMTAP_BURST_NORMAL. This is the correct behaviour: when we connect a UDP socket to the receiver we really expect the UDP packet to be exactly what the receiver is outputting.

- OR because of some processing delay (race condition), it can happen that the message generated by (1) is processed by (2) before the UDP socket has the time to process and send it. Because (2) modifies the message in-place instead of making a copy, the UDP packet sent now has a type GSMTAP_TYPE_UM_BURST but a subtype 135 instead of the original 6.
But we connected the UDP to the receiver, not the demapper, this is therefore not the UDP packet we are expecting to send. Because (2) modifies the input message instead of generating a new modified one, when this race condition happens it is as if we had connected the UDP socket to the demapper.

And this race condition is the bug I'm reporting.

romankh, 2015-08-16:
Sorry, misunderstanding was from my side, now I got it. 
You are right, this is a bug.

iZsh, 2015-08-16:
I'm trying to fix it in my sandbox, with the following patch:

diff --git a/lib/demapping/universal_ctrl_chans_demapper_impl.cc b/lib/demapping/universal_ctrl_chans_demapper_impl.cc
index 8a6bfb5..d0fc3f1 100644
--- a/lib/demapping/universal_ctrl_chans_demapper_impl.cc
+++ b/lib/demapping/universal_ctrl_chans_demapper_impl.cc
@@ -140,7 +140,7 @@ namespace gr {
     {
         pmt::pmt_t header_plus_burst = pmt::cdr(msg);
         gsmtap_hdr * header = (gsmtap_hdr *)pmt::blob_data(header_plus_burst);
-
+
         uint32_t frame_nr = be32toh(header->frame_number);
         uint32_t fn_mod51 = frame_nr % 51;
         uint32_t fn51_start = d_starts_fn_mod51[fn_mod51];
@@ -148,14 +148,22 @@ namespace gr {
         uint32_t ch_type = d_channel_types[fn_mod51];

         if(header->timeslot==d_timeslot){
-      header->sub_type = ch_type;
-      header->sub_slot = d_subslots[fn_mod51 + (51 * (frame_nr % 2))];
+            gsmtap_hdr new_hdr;
+            memcpy(&new_hdr, header, sizeof(gsmtap_hdr));
+            new_hdr.sub_type = ch_type;
+            new_hdr.sub_slot = d_subslots[fn_mod51 + (51 * (frame_nr % 2))];
+            pmt::pmt_t msg_binary_blob = pmt::make_blob(&new_hdr,sizeof(gsmtap_hdr));
+            pmt::pmt_t msg_out = pmt::cons(pmt::PMT_NIL, msg_binary_blob);
+
+            // header->sub_type = ch_type;
+            // header->sub_slot = d_subslots[fn_mod51 + (51 * (frame_nr % 2))];

             if(fn_mod51>=fn51_start && fn_mod51<=fn51_stop)
             {
                 uint32_t ii = fn_mod51 - fn51_start;
                 d_frame_numbers[ii] = frame_nr;
-                d_bursts[ii] = msg;
+                d_bursts[ii] = msg_out;
+                // d_bursts[ii] = msg;
             }

             if(fn_mod51==fn51_stop)

but although I can see the bursts are now correct in wireshark, the decoder stops working with this patch, I'm not sure why (it's been a while since I've done any gnuradio coding, I'm not fully familiar with the new messaging system). I'm investigating...

By the way, of course other blocks need to be fixed, e.g.:

diff --git a/lib/decoding/control_channels_decoder_impl.cc b/lib/decoding/control_channels_decoder_impl.cc
index 3ea3fd9..a79c9e0 100644
--- a/lib/decoding/control_channels_decoder_impl.cc
+++ b/lib/decoding/control_channels_decoder_impl.cc
@@ -137,13 +137,13 @@ namespace gr {
            //send message with header of the first burst
             pmt::pmt_t first_header_plus_burst = pmt::cdr(d_bursts[0]);
             gsmtap_hdr * header = (gsmtap_hdr *)pmt::blob_data(first_header_plus_burst);
-            header->type = GSMTAP_TYPE_UM;

             int8_t * header_content = (int8_t *)header;

             int8_t header_plus_data[sizeof(gsmtap_hdr)+DATA_BYTES];
             memcpy(header_plus_data, header_content, sizeof(gsmtap_hdr));
             memcpy(header_plus_data+sizeof(gsmtap_hdr), outmsg, DATA_BYTES);
+            ((gsmtap_hdr*)header_plus_data)->type = GSMTAP_TYPE_UM;

             pmt::pmt_t msg_binary_blob = pmt::make_blob(header_plus_data,DATA_BYTES+sizeof(gsmtap_hdr));
             pmt::pmt_t msg_out = pmt::cons(pmt::PMT_NIL, msg_binary_blob);

(and other blocks as well)

if you know why my first patch breaks the decoder later down the processing, feel free to tip me in, that would save me some time ^^

iZsh, 2015-08-16:
Oh I think I forgot to copy the data of the message :-)
sorry, me being stupid

romankh, 2015-08-16:
Wanted to write about the payload right now :)
Thanks a lot for fixing the issue

iZsh, 2015-08-16:
I'll send you a patch in a minute

romankh, 2015-08-16:
Thanks, but no hurry, I don't do the merges, @ptrkrysik does.

iZsh, 2015-08-16:
here is the patch for it (sorry, I'm being lazy because i'm in the middle of something, so I didn't branch it and did a pull request)

    diff --git a/lib/decoding/control_channels_decoder_impl.cc b/lib/decoding/control_channels_decoder_impl.cc
    index 3ea3fd9..1d25664 100644
    --- a/lib/decoding/control_channels_decoder_impl.cc
    +++ b/lib/decoding/control_channels_decoder_impl.cc
    @@ -137,13 +137,10 @@ namespace gr {
                //send message with header of the first burst
                 pmt::pmt_t first_header_plus_burst = pmt::cdr(d_bursts[0]);
                 gsmtap_hdr * header = (gsmtap_hdr *)pmt::blob_data(first_header_plus_burst);
    -            header->type = GSMTAP_TYPE_UM;
    -            
    -            int8_t * header_content = (int8_t *)header;
    -            
                 int8_t header_plus_data[sizeof(gsmtap_hdr)+DATA_BYTES];
    -            memcpy(header_plus_data, header_content, sizeof(gsmtap_hdr));
    +            memcpy(header_plus_data, header, sizeof(gsmtap_hdr));
                 memcpy(header_plus_data+sizeof(gsmtap_hdr), outmsg, DATA_BYTES);
    +            ((gsmtap_hdr*)header_plus_data)->type = GSMTAP_TYPE_UM;
                 
                 pmt::pmt_t msg_binary_blob = pmt::make_blob(header_plus_data,DATA_BYTES+sizeof(gsmtap_hdr));
                 pmt::pmt_t msg_out = pmt::cons(pmt::PMT_NIL, msg_binary_blob);

    diff --git a/lib/decoding/tch_f_decoder_impl.cc b/lib/decoding/tch_f_decoder_impl.cc
    index a3b09fb..32e05e5 100644
    --- a/lib/decoding/tch_f_decoder_impl.cc
    +++ b/lib/decoding/tch_f_decoder_impl.cc
    @@ -153,11 +153,10 @@ namespace gr {
     
                         pmt::pmt_t first_header_plus_burst = pmt::cdr(d_bursts[0]);
                         gsmtap_hdr * header = (gsmtap_hdr *)pmt::blob_data(first_header_plus_burst);
    -                    header->type = GSMTAP_TYPE_UM;
    -                    int8_t * header_content = (int8_t *)header;
                         int8_t header_plus_data[sizeof(gsmtap_hdr)+DATA_BYTES];
    -                    memcpy(header_plus_data, header_content, sizeof(gsmtap_hdr));
    +                    memcpy(header_plus_data, header, sizeof(gsmtap_hdr));
                         memcpy(header_plus_data+sizeof(gsmtap_hdr), outmsg, DATA_BYTES);
    +                    ((gsmtap_hdr*)header_plus_data)->type = GSMTAP_TYPE_UM;
     
                         pmt::pmt_t msg_binary_blob = pmt::make_blob(header_plus_data,DATA_BYTES+sizeof(gsmtap_hdr));
                         pmt::pmt_t msg_out = pmt::cons(pmt::PMT_NIL, msg_binary_blob);
    

    diff --git a/lib/demapping/tch_f_chans_demapper_impl.cc b/lib/demapping/tch_f_chans_demapper_impl.cc
    index cc24b9e..a54ed3f 100644
    --- a/lib/demapping/tch_f_chans_demapper_impl.cc
    +++ b/lib/demapping/tch_f_chans_demapper_impl.cc
    @@ -29,6 +29,8 @@
     #include <grgsm/endian.h>
     #include <grgsm/gsmtap.h>
     
    +#define BURST_SIZE 148
    +
     namespace gr {
       namespace gsm {
     
    @@ -80,12 +82,20 @@ namespace gr {
             int8_t * burst_bits = (int8_t *)(pmt::blob_data(header_plus_burst))+sizeof(gsmtap_hdr);
     
             if(header->timeslot == d_timeslot){
    -            header->sub_type = GSMTAP_CHANNEL_TCH_F;
    +            int8_t new_msg[sizeof(gsmtap_hdr)+BURST_SIZE];
    +            gsmtap_hdr * new_hdr = (gsmtap_hdr*)new_msg;
    +            memcpy(new_msg, header, sizeof(gsmtap_hdr)+BURST_SIZE);
     
    +            new_hdr->sub_type = GSMTAP_CHANNEL_TCH_F;
                 if (fn_mod13 == 12)
    -            {
                     header->sub_type = GSMTAP_CHANNEL_ACCH|GSMTAP_CHANNEL_TCH_F;
     
[... 72 lines cut ...]
             uint32_t fn_mod51 = frame_nr % 51;
             uint32_t fn51_start = d_starts_fn_mod51[fn_mod51];
    @@ -148,14 +150,19 @@ namespace gr {
             uint32_t ch_type = d_channel_types[fn_mod51];
             
             if(header->timeslot==d_timeslot){
    -       header->sub_type = ch_type;
    -       header->sub_slot = d_subslots[fn_mod51 + (51 * (frame_nr % 2))];
    +            int8_t new_msg[sizeof(gsmtap_hdr)+BURST_SIZE];
    +            gsmtap_hdr * new_hdr = (gsmtap_hdr*)new_msg;
    +            memcpy(new_msg, header, sizeof(gsmtap_hdr)+BURST_SIZE);
    +            new_hdr->sub_type = ch_type;
    +            new_hdr->sub_slot = d_subslots[fn_mod51 + (51 * (frame_nr % 2))];
    +            pmt::pmt_t msg_binary_blob = pmt::make_blob(new_msg,sizeof(gsmtap_hdr)+BURST_SIZE);
    +            pmt::pmt_t msg_out = pmt::cons(pmt::PMT_NIL, msg_binary_blob);
            
                 if(fn_mod51>=fn51_start && fn_mod51<=fn51_stop)
                 {
                     uint32_t ii = fn_mod51 - fn51_start;
                     d_frame_numbers[ii] = frame_nr;
    -                d_bursts[ii] = msg;
    +                d_bursts[ii] = msg_out;
                 }
                 
                 if(fn_mod51==fn51_stop)

iZsh, 2015-08-16:
I feel bad about this, so I'm preparing a real pull request. Should be done in a minute

closed 2015-08-16 by iZsh