ptrkrysik/gr-gsm #114: Recording traffic data

ijarraki, 2015-08-16:
Now, I am just getting into this field so please excuse me if I am asking a dumb question here. Would it be possible to modify airprobe_rtlsdr.py with a file sink (or something) so that the raw GSM data (post fine-tuning through UI) can be stored to a capture file as well as being parsed to Wireshark? That way we can fine-tune into and look at traffic as well as being able to replay the same traffic later with airprobe_file.py. Having these capture files be compatible with legacy airprobe/go.sh would be a must.

Are there other means of recording traffic data directly into capture files compatible with gr-gsm/airprobe? I find myself unable to produce compatible files with GQRX and some of the old GRCs I have been using to convert rtl_sdr and hackrf_transfer captures are not working with GnuRadio 3.7.

ijarraki, 2015-08-16:
I just found https://groups.google.com/forum/?_escaped_fragment_=topic/gr-gsm/1C7LQCOgLEY# in which ptrkrysik provides a means of generating a compatible cfile directly from rtl-sdr source.

Having a -o/--output-file option in airprobe_rtlsdr.py would be much better since we can then fine-tune to the frequency we want to see and record. Any way to modify airprobe_rtlsdr.grc so that traffic is stored to /path/to/cfile until the .py gets an output option?

romankh, 2015-08-16:
As mentioned in the previous issue, we are going to provide some python apps to improve usability.

ijarraki, 2015-08-16:
Great, but can anyone tell me if it is possible and then where to add a file sink to the current distributions GRCs so that the tuned GSM data does get dumped to a file?

ptrkrysik, 2015-08-16:
You can connect the file_sink directly after rtl-sdr source in airprobe_rtlsdr.grc (using gnuradio-companion of course) so exactly the same signal as you receive will be stored in a file. Also - for discussions regarding things other than bugs we use now the mailing list.

ijarraki, 2015-08-16:
It is not my intention to ignore the mailing list, I just figured that we should resolve this "issue" here and then I will of course post any future questions to the mailing list. 

I have inserted a file sink into airprobe_rtlsdr.grc (https://pastebin.com/APqZPBz9) and used the resulting airprobe_rtlsdr.py to record five minutes of traffic on a non-hopping ARFCN.  Replaying said capture with airprobe_file.py gives no errors and 24947 packets in Wireshark; so far so good.

Now, to look at different timeslots and frame numbers I will have to replay the same cfile through legacy airprobe/go.sh (./go.sh /srv/capture/capture.cfile 64 (It seems that I have to specify 64 as decimation rate, otherwise I get nothing but errors)).

For some reason, legacy airprobe is only able to decode a few packets (1696 in Wireshark) and I get the motherload of these errors:
sch.c:260 ERR: conv_decode 4
sch.c:260 ERR: conv_decode 7
sch.c:260 ERR: conv_decode 9
sch.c:260 ERR: conv_decode 6
sch.c:260 ERR: conv_decode 11
sch.c:260 ERR: conv_decode 6
sch.c:260 ERR: conv_decode 7
sch.c:260 ERR: conv_decode 10
sch.c:260 ERR: conv_decode 10
sch.c:260 ERR: conv_decode 8
sch.c:260 ERR: conv_decode 6
sch.c:260 ERR: conv_decode 6
sch.c:260 ERR: conv_decode 9
sch.c:260 ERR: conv_decode 8
sch.c:260 ERR: conv_decode 9
sch.c:260 ERR: conv_decode 6
sch.c:260 ERR: conv_decode 5
sch.c:260 ERR: conv_decode 11
sch.c:260 ERR: conv_decode 3
sch.c:260 ERR: conv_decode 5
sch.c:260 ERR: conv_decode 10
sch.c:260 ERR: conv_decode 9

Any insights?

ijarraki, 2015-08-16:
It is not my intention to ignore the mailing list; my thinking is that we should resolve this "issue" here and then I will of course ask any future questions through the ml.

I have added a file sink to airprobe_rtlsdr.grc (http://pastebin.com/CKUrhVUB) and used the resulting airprobe_rtlsdr.py to record five minutes of traffic on a non-hopping ARFCN. Replaying the capture file through airprobe_file.py gives no errors and 24947 packets are parsed to Wireshark; so far so good.

Now, to look at frame numbers and different timeslots I will have to run the same capture file through legacy airprobe (./go.sh /path/to/capture.cfile 64 (it seems I have to specify 64 as decimation rate, otherwise I get nothing but errors)). Problem is, only 1696 packets are successfully decoded and parsed to Wireshark and I get the motherload of errors like this:
sch.c:260 ERR: conv_decode 5
sch.c:260 ERR: conv_decode 10
sch.c:260 ERR: conv_decode 9

Any insights?

ptrkrysik, 2015-08-17:
Regarding legacy airprobe - we don't provide support for it. Everything
that was possible with it is now possible with gr-gsm. There are no
equivalent commandline parameters for conffiguration of timeslots in the
gr-gsm apps yet that were available in the old airprobe but everything
can be configured by modification of the flowgraph (in
airprobe_file.grc). If you tell me some examples of what you want to
achieve I will tell you how it can be done currently.

P.S. regarding decimation rate 64 for the old airprobe - it is strange
that something was decoded, because this decimation rate gives
64e6/64=1e6 [Hz] (64MHz was sampling rate of USRP1), but default
configuration of rtl-sdr source in gr-gsm is to 2 MHz.

ijarraki, 2015-08-17:
Of course I do not expect you to provide support for legacy airprobe, I'm just asking if you have any ideas. 

The capture file was generated using a sample rate of 1M (for some reason I get peaks at both 943M and 944.3M, decoded packets show that this is in fact the same BTS/ARFCN). As to the specific decimation rates of the USRP1/2 I seem to recall that these are 112/174?

Can you confirm that both RTL-SDR and the HackRF use a decimation rate of 64?

ptrkrysik, 2015-08-17:
W dniu 17.08.2015 o 19:23, ijarraki pisze:
> Of course I do not expect you to provide support for legacy airprobe,
> I'm just asking if you have any ideas.
>
> The capture file was generated using a sample rate of 1M (for some
> reason I get peaks at [...]
Decimation rate without original sample rate doesn't make too much
sense. When only practically available SDR was USRP1 it could be assumed
that sample rate was 64MHz - as this was sample rate of the device. In
RTL-SDR you can set sample rate quite flexibly and it doesn't have to be
64MHz divided by some integer. I don't know HackRF but probably also it
doesn't use the same internal clock rate as USRP1.

There are two peaks probably due to some nonlinear distortion in the
receiving hardware.

-- 
Best Regards,
Piotr Krysik

ijarraki, 2015-08-19:
Yes, the twin peak (pun intended) went away after I moved a nearby power strip.

As to cfile by gr-gsm not working all that well with legacy airprobe: If I tune to 944M with gr-gsm I get a distinct peak about ten times higher than the surrounding/level static and no decode errors. With legacy airprobe the peak is barely noticeable and I get mostly decode errors. Could these issues be related and do you have any idea what could be causing them?

ptrkrysik, 2015-08-19:
W dniu 19.08.2015 o 12:45, ijarraki pisze:
> Yes, the twin peak (pun intended) went away after I moved a nearby power
> strip.
>
> As to cfile by gr-gsm not working all that well with legacy airprobe: If
> I tune to 944M with gr-gsm I get a distinct [...]
Hi,

If you record the signal on the same frequency outside of gr-gsm with
the same device everything is well?

Regarding the spectrums that you are comparing - are you 100% sure that
they are both computed directly at the output of the receiver with the
same reception parameters (bandwidth, gain)? I doubt it.

On the sidenote: the frequency offset correction in airprobe wasn't very
stable and robust (I know that too well - I've written it). This might
be the reason why you can't decode anything with airprobe if you i.e.
have large frequency offset.

There are also other things that were corrected in gr-gsm and that were
wrong in Airprobe. For example synchronization of normal bursts wasn't
working so positions of normal bursts were hardcoded until this will be
resolved (it was resolved finally - in gr-gsm). Airprobe didn't work for
some of base stations which transmitted normal bursts at different
positions in frames.

This is why I would advise you to use gr-gsm for what you want to do.

-- 
Best regards,
Piotr Krysik

ijarraki, 2015-08-25:
No, recording the same BTS, at the same frequency, with the same hardware mostly fails with legacy airprobe (tons of decode errors). Replaying a cfile successfully captured with gr-gsm through legacy airprobe also mostly fails, lots of those same errors (of course the cfile works just fine with gr-gsm).

"Airprobe didn't work for some of base stations which transmitted normal bursts at different positions in frames." <--- Maybe all towers at my locale do this?

You say it is possible to replay different timeslots, print frame numbers to terminal and provide KC for decryption by modifying the GRCs, care to share an example of such a GRC?

ptrkrysik, 2015-08-30:
Look at the decoding application added by @romankh - it might be what you need.

ijarraki, 2015-09-06:
Yes, the decoding application is looking good but it does not print frames/bursts to console? We need a way to replay cfiles (like what airprobe_file.py did) with frame and modified frame numbers to console.

romankh, 2015-09-07:
airprobe_decode.py should already send gsmtap messages to udp port 4729, as the airprobe_file.py did before. 

I can add an option for printing dumping bursts / messages in the next days, meanwhile you can still use airprobe_file, the flow graph is just moved to the examples directory.

For the newer apps (capture and decode), there are no grc files, they are just written manually in python.

ijarraki, 2015-09-10:
Strange, Wireshark is not getting any data from airprobe_decode (tested on the last two commits). Also, airprobe_file is nowhere to be found in the examples directory?

If/when you do add an option for printing bursts/frames to airprobe_decode please make the output look like it did in legacy airprobe (with frame numbers, C1/C0, etc) like you discussed in pull request 59 (https://github.com/ptrkrysik/gr-gsm/pull/59).

romankh, 2015-09-11:
Sorry, airprobe_file was removed recently. 
But you can re-create it using grc if necessary. 
Drop me a note on the mailing list if you need help with that, or with airprobe_decode. 

I am going to add message output to airprobe_decode this weekend, with frame number and modified frame number

ptrkrysik, 2015-09-14:
The aiprobe_file.grc was moved here:
https://github.com/ptrkrysik/examples/blob/master/get_bcch_msgs_from_cfile/get_bcch_msgs_from_cfile.grc

You can open it in gnuradio-companion, generate python file and use it
the same way as airprobe_file.py.

To download examples you can do:
git submodule init examples
git submodule update examples

in the gr-gsm's directory.

closed 2015-09-14 by ptrkrysik

Spanky1503, 2016-01-20:
gr-gsm not working 
Traceback (most recent call last):
  File "/usr/local/bin/airprobe_rtlsdr.py", line 18, in <module>
    import grgsm
  File "/usr/local/lib/python2.7/dist-packages/grgsm/__init__.py", line 45, in <module>
    from grgsm_swig import *
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 26, in <module>
    _grgsm_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 22, in swig_import_helper
    _mod = imp.load_module('_grgsm_swig', fp, pathname, description)
ImportError: libgnuradio-gsm.so: cannot open shared object file: No such file or directory

Spanky1503, 2016-01-20:
can someone help please im new thank you

ptrkrysik, 2016-01-22:
Usually sudo ldconfig solves the problem

Spanky1503, 2016-01-23:
thank you ill try it

Spanky1503, 2016-08-24:
---------- Forwarded message ----------
From: Credit One Bank <donotreply@creditonemail.com>
Date: Mon, Aug 22, 2016 at 10:31 AM
Subject: Know someone who could use a Platinum Visa?
To: peoplesvoiceoc@gmail.com

Tell a friend about Credit One Bank | View this email online
<http://creditonemail.com/a/hBXuzXHB8Zuz5B9DYR5AAgtYGVR/viewonline>

[image: Credit One Bank Logo]

[image: Tell a Friend!]

Dear John,

Thank you for being a Credit One Bank® card member. We hope you're enjoying
all the benefits your card has to offer. Do you have friends or family
members who also could use a card like this? They can get Pre-Qualified for
a Platinum Visa® card. All cards include great benefits like:

•  Unlimited 1% cash back on gas and grocery purchases
•  Free online monthly credit score
•  Automatic credit line increase reviews
•  Zero liability for fraudulent charges

Spread the word! Forward this email to friends and family. They can click
on the link below to get Pre-Qualified, today.

[image: Visit CreditOneBank.com.]
<http://creditonemail.com/a/hBXuzXHB8Zuz5B9DYR5AAgtYGVR/button>

    Follow Us
      [image: Follow Us on Facebook]
<http://creditonemail.com/a/hBXuzXHB8Zuz5B9DYR5AAgtYGVR/facebook>

------------------------------
*Important Information from Credit One Bank*

You are receiving this email because you are a Credit One Bank Customer. If
you do not wish to receive this particular email notification in the
future, please click here
<http://creditonemail.com/a/hBXuzXHB8Zuz5B9DYR5AAgtYGVR/unsub>.

Credit One Bank and its service providers are committed to providing
meaningful privacy protection for their customers. For information on our
Privacy Policy or how to contact us, please visit our website at
www.CreditOneBank.com
<http://creditonemail.com/a/hBXuzXHB8Zuz5B9DYR5AAgtYGVR/privacy>.

©2016 Credit One Bank, N.A. All rights reserved.
Credit One®, Credit One Bank® and corresponding logos are federally
registered trademarks.
P.O. Box 98873, Las Vegas, NV 89193-8873