ptrkrysik/gr-gsm #120: Multiple problems with installation

ptrkrysik, 2015-08-24:
Due to extensive development some problems with installation were introduced. 

One of them is related to adding libosmocore to gr-gsm's dependencies. Libosmocore is depends on psclite (required for simcard communication - functionality not used in gr-gsm). On ubuntu pybombs fails to install psclite as it doesn't fond recipe for libudev-dev package - user has to install libudev-dev manually. I don't know yet how exactly to fix this. The best way would be to install libosmocore without support for smartcards - as probably installation of psclite will fail in many colourful ways on different distributions.

Another problem is with cmake creating grgsm library - the reason is totally mysterious for me currently as when building gr-gsm manually there are no problems.

romankh, 2015-08-24:
For the psclite, I am the one to blame :/
I removed it from the libosmocore recipe in my pybombs installation, and later forgot to handle that somehow. 

I agree, probably installing libosmocore without pcsclite is the best option, however we again end up with a custom pybombs installer, or at least custom recipes. Or did I miss something ?

Whats the other problem with cmake ? Haven't noticed that so far...

ptrkrysik, 2015-08-25:
W dniu 24.08.2015 o 22:51, Roman Khassraf pisze:
> For the psclite, I am the one to blame :/
> I removed it from the libosmocore recipe in my pybombs installation, and
> later forgot to handle that somehow.
>
> I agree, probably installing libosmocore [...]

I think it's more of a problem of pybombs and libosmocore that hit us
because of our dependency on libosmocore. Pybombs guys might not be
aware yet that psclite in order to compile itself needs libudev-dev.
The fastest solution might be to add dependency on libudev-dev in
psclite's recipe and add recipe for libudev-dev (and hope this will be
the end :) ).

> 
> Whats the other problem with cmake ? Haven't noticed that so far...
> 

This problem is quite peculiar. I also don't get it on my system. But
during installation with pybombs on clean Ubuntu 14.04 - it is there.
When compiling gr-gsm manually - there is no problem. I don't know how
to reproduce this bug outside of pybombs yet.

But I suspect that my insisting on compiling of grc files during
building of gr-gsm might be unsuccessful - when I correct one thing
something even more unexpected breaks. This might also be the reason of
this problem. I also get from time to time Segmentation fault (:/)
during installation which is really hard to explain - maybe grcc causes
some race condition in installation. Whole experience makes me miss the
old solution where I compiled grc files manually on the oldest
compatible GNU Radio. I will go back to that and see if the problems
disappear.

-- 
Best Regards,
Piotr Krysik

romankh, 2015-08-25:
I think I will have the decoder app until tomorrow evening, so maybe we could remove the grc compilation after merging.

Will also try to create a working recipe for psclite after that, hope it will not make too much troubles.

ptrkrysik, 2015-08-25:
W dniu 25.08.2015 o 10:16, Roman Khassraf pisze:
> I think I will have the decoder app until tomorrow evening, so maybe we
> could remove the grc compilation after merging.
> 
> Will also try to create a working recipe for psclite after that, hope it
> will not make too much troubles.
> 
Ok, I've found one problem with gr-gsm's pybombs recipe. In the install
there is following command:
cd build

the pybombs is already in build directory at that moment and stops on
this command.

-- 
Best Regards,
Piotr Krysik

romankh, 2015-08-29:
I assigned this to myself now, will (try to) fix the pcsclite problem

romankh, 2015-08-29:
@Piotr (or anyone else), could you cross check my fix: https://github.com/romankh/recipes

ptrkrysik, 2015-08-29:
@romankh, I've started installation with pybombs in virtual machine with
clean Ubuntu 14.04. I will let you know of the result as soon as it ends.

ptrkrysik, 2015-08-30:
@Roman - I can confirm that after application of your fixes gr-gsm can
be installed with pybombs on a clean Ubuntu 14.04.

romankh, 2015-08-30:
Piotr, thanks for verification, I created a pull request to Gnuradio's recipe repo.

romankh, 2015-08-30:
Fix is merged into the recipe repo.
Guess we can close that issue ?

ptrkrysik, 2015-08-30:
I'm closing the issue. I will also add missing information about libosmocore in manual installation description.

closed 2015-08-30 by ptrkrysik