ptrkrysik/gr-gsm #122: Installation problem, Bison dependancy error.

orsty3001, 2015-09-03:
Trying to install gr-gsm using pybomb and I get this error. 

checking for bison... yes
checking for bison version >= 2.5... no
configure: error: Bison version 2.5 or higher must be installed on the system!
Configuration failed. Re-trying with higher verbosity.
make: *** No targets specified and no makefile found.  Stop.
Build failed. See output above for error messages.

Bison is already installed and the version is > 2.5.

u332@us2222-Inspiron-3520:~/pybombs$ bison --version
bison (GNU Bison) 3.0.2

romankh, 2015-09-04:
Hi @orsty3001, 
please open a thread in the mailing list and provide more detailed information about your installation. 
See the readme for both the mailing list and a minimum set of information regarding installation problems.

ptrkrysik, 2015-09-14:
I've installed gr-gsm using pybombs on clean Ubuntu 14.04. Closing the issue because of inability to reproduce it.

closed 2015-09-14 by ptrkrysik