ptrkrysik/gr-gsm #125: ImportError: cannot import name Range

AxisRay, 2015-09-07:
When I run **airprobe_rtlsdr.py** with gnuradio 3.7.5,there is an Error
> Traceback (most recent call last):
>   File "/usr/local/bin/airprobe_rtlsdr.py", line 26, in <module>
>     from gnuradio.qtgui import Range, RangeWidget
> ImportError: cannot import name Range

ptrkrysik, 2015-09-07:
W dniu 07.09.2015 o 10:01, Ray Lee pisze:
> When I run *airprobe_rtlsdr.py* with gnuradio 3.7.5,there is an Error
> 
>     Traceback (most recent call last):
>     File "/usr/local/bin/airprobe_rtlsdr.py", line 26, in
>     from gnuradio.qtgui import Range, RangeWidget
>     ImportError: cannot import name Range
> 

Hi,

Try to open airprobe_rtlsdr.grc in gnuradio-companion and run it there.
Then write here if it worked or not.

-- 
Best Regards,
Piotr Krysik

AxisRay, 2015-09-07:
It worked! Thank you !

closed 2015-09-07 by AxisRay

nejohnson2, 2015-09-21:
I am getting the same error as above, though when I run **airprobe_rtlsdr.grc** from **gnuradio-companion** I get the following error:

```
Traceback (most recent call last):
  File "/home/pi/workspace/gr-gsm/apps/airprobe_rtlsdr.py", line 19, in <module>
    import grgsm
  File "/usr/local/lib/python2.7/dist-packages/grgsm/__init__.py", line 45, in <module>
    from grgsm_swig import *
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 28, in <module>
    _grgsm_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_grgsm_swig', fp, pathname, description)
ImportError: libgnuradio-grgsm.so: cannot open shared object file: No such file or directory
```

I should note that I'm trying to run this from a Raspberry Pi.  Everything seemed to install ok...

Thanks!

ptrkrysik, 2015-09-21:
Hi Nicholas,

Can you try invoking:

ldconfig

with root privileges?

rojavatechops, 2017-04-11:
I think I'm having the same software compatibility issues with gnu-radio. I'm running version 3.7.5.

Can you either fix this script or tell me what version I should downgrade to? Please and thank you. 

#!/usr/bin/env python
##################################################
# Gnuradio Python Flow Graph
# Title: Gr-gsm Livemon
# Author: Piotr Krysik
# Description: Interactive monitor of a single C0 channel with analysis performed by Wireshark (command to run wireshark: sudo wireshark -k -f udp -Y gsmtap -i lo)
# Generated: Tue Apr 11 13:59:26 2017
##################################################

from PyQt4 import Qt
from PyQt4.QtCore import QObject, pyqtSlot
from gnuradio import blocks
from gnuradio import eng_notation
from gnuradio import gr
from gnuradio import qtgui
from gnuradio.eng_option import eng_option
from gnuradio.filter import firdes
from math import pi
from optparse import OptionParser
import PyQt4.Qwt5 as Qwt
import grgsm
import osmosdr
import pmt
import sip
import sys

from distutils.version import StrictVersion
class grgsm_livemon(gr.top_block, Qt.QWidget):

    def __init__(self, args="", fc=939.4e6, gain=30, osr=4, ppm=0, samp_rate=2000000.052982, shiftoff=400e3):
        gr.top_block.__init__(self, "Gr-gsm Livemon")
        Qt.QWidget.__init__(self)
        self.setWindowTitle("Gr-gsm Livemon")
        try:
             self.setWindowIcon(Qt.QIcon.fromTheme('gnuradio-grc'))
        except:
             pass
        self.top_scroll_layout = Qt.QVBoxLayout()
        self.setLayout(self.top_scroll_layout)
        self.top_scroll = Qt.QScrollArea()
        self.top_scroll.setFrameStyle(Qt.QFrame.NoFrame)
        self.top_scroll_layout.addWidget(self.top_scroll)
        self.top_scroll.setWidgetResizable(True)
        self.top_widget = Qt.QWidget()
        self.top_scroll.setWidget(self.top_widget)
        self.top_layout = Qt.QVBoxLayout(self.top_widget)
        self.top_grid_layout = Qt.QGridLayout()
        self.top_layout.addLayout(self.top_grid_layout)

        self.settings = Qt.QSettings("GNU Radio", "grgsm_livemon")
        self.restoreGeometry(self.settings.value("geometry").toByteArray())

        ##################################################
        # Parameters
        ##################################################
        self.args = args
        self.fc = fc
        self.gain = gain
        self.osr = osr
        self.ppm = ppm
        self.samp_rate = samp_rate
        self.shiftoff = shiftoff

        ##################################################
        # Variables
        ##################################################
[... 251 lines cut ...]
    parser.add_option("-f", "--fc", dest="fc", type="eng_float", default=eng_notation.num_to_str(939.4e6),
        help="Set fc [default=%default]")
    parser.add_option("-g", "--gain", dest="gain", type="eng_float", default=eng_notation.num_to_str(30),
        help="Set gain [default=%default]")
    parser.add_option("", "--osr", dest="osr", type="intx", default=4,
        help="Set OSR [default=%default]")
    parser.add_option("-p", "--ppm", dest="ppm", type="eng_float", default=eng_notation.num_to_str(0),
        help="Set ppm [default=%default]")
    parser.add_option("-s", "--samp-rate", dest="samp_rate", type="eng_float", default=eng_notation.num_to_str(2000000.052982),
        help="Set samp_rate [default=%default]")
    parser.add_option("-o", "--shiftoff", dest="shiftoff", type="eng_float", default=eng_notation.num_to_str(400e3),
        help="Set shiftoff [default=%default]")
    (options, args) = parser.parse_args()
    if(StrictVersion(Qt.qVersion()) >= StrictVersion("4.5.0")):
        Qt.QApplication.setGraphicsSystem(gr.prefs().get_string('qtgui','style','raster'))
    qapp = Qt.QApplication(sys.argv)
    tb = grgsm_livemon(args=options.args, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff)
    tb.start()
    tb.show()
    def quitting():
        tb.stop()
        tb.wait()
    qapp.connect(qapp, Qt.SIGNAL("aboutToQuit()"), quitting)
    qapp.exec_()
    tb = None #to clean up Qt widgets

vallettea, 2017-07-31:
I have the same problem as @nejohnson2 on rpi
Using `ldconfig` as suggested by @ptrkrysik did changed anything.

johnc2k, 2017-08-06:
I am also having the problem on a Raspi3 (built using the instructions on the wiki).

Executing grgsm_livemon.grc directly in gnuradio_companion results in the following error:

`pi@raspberrypi:~ $ gnuradio-companion
Warning: Block key "gsm_msg_to_tag.xml" not found when loading category tree.
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

<<< Welcome to GNU Radio Companion 3.7.5 >>>

Preferences file: /home/pi/.grc
Block paths:
        /usr/local/share/gnuradio/grc/blocks
        /usr/share/gnuradio/grc/blocks
        /home/pi/.grc_gnuradio

Showing: ""

Loading: "/home/pi/gr-gsm/apps/grgsm_livemon.grc"
>>> Done

Showing: "/home/pi/gr-gsm/apps/grgsm_livemon.grc"

Generating: "/home/pi/gr-gsm/apps/grgsm_livemon.py"

Executing: "/home/pi/gr-gsm/apps/grgsm_livemon.py"

linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
Exact sample rate is: 2000000.052982 Hz
Using Volk machine: neon_hardfp_orc
Traceback (most recent call last):
  File "/home/pi/gr-gsm/apps/grgsm_livemon.py", line 335, in <module>
    tb = grgsm_livemon(args=options.args, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff)
  File "/home/pi/gr-gsm/apps/grgsm_livemon.py", line 177, in __init__
    timeslot_nr=1,
  File "/usr/local/lib/python2.7/dist-packages/grgsm/gsm_sdcch8_demapper.py", line 39, in __init__
    self.msg_connect((self.gsm_universal_ctrl_chans_demapper_0, 'bursts'), (self, 'bursts'))
TypeError: msg_connect() takes exactly 5 arguments (3 given)

>>> Done
`